intel/compute-runtime #337: possible to use HKEY_CURRENT_USER to specify the cl_cache dir?

aisnote, 2020-07-10:
Some app will running as medium IL on windows and can not write the key under    
 HKEY_LOCAL_MACHINE\SOFTWARE\Intel\IGFX\OCL:

jhajducz, 2020-07-13:
I think it's enough to create 'cl_cache' directory in the same place where binary will be executed from.

mrsabhar, 2020-07-15:
@jhajducz  What if developer wants to create in user space and dont have permission to create key?

jhajducz, 2020-07-16:
You mean that there is no permission to create new directory ("cl_cache") in the same place where binary will be launched from? Because as far as I understand that would be all, there is no need to play with register keys then.

aisnote, 2020-07-23:
> You mean that there is no permission to create new directory ("cl_cache") in the same place where binary will be launched from? Because as far as I understand that would be all, there is no need to play with register keys then.

yes. Lost of enterprise exe is deployed via MSI and can not write the same place.

AdamCetnerowski, 2020-09-03:
Fix was released as part of 20.32.17625. If there are new concerns, please open a new issue.

closed 2020-09-03 by AdamCetnerowski