intel/compute-runtime #338: compute runtime conflicts with latest OpenCL-Headers

asavah, 2020-07-11:
This commit https://github.com/KhronosGroup/OpenCL-Headers/commit/b04034a17e214322772c19e7ccd2adf0cd306920#diff-b3f8b700e5939bbbf18229c394660489 adds intel api stuff to official Khronos opencl headers.
This causes build failute in compute-runtime atleast as of commit 146fc900c33492fe4a806cd897a8d32cee40e7b6

Reverting the commit in OpenCL-Headers leads to a successful build.

Intel, get your shit straight please.

```
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:928:7: error: conflicting declaration of C function ‘void* clHostMemAllocINTEL(cl_context, cl_mem_properties_intel*, size_t, cl_uint, cl_int*)’
  928 | void *clHostMemAllocINTEL(
      |       ^~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/dispatch.h:12,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:10,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/opt/pfx-suite/include/CL/cl_ext_intel.h:476:1: note: previous declaration ‘void* clHostMemAllocINTEL(cl_context, const cl_mem_properties_intel*, size_t, cl_uint, cl_int*)’
  476 | clHostMemAllocINTEL(
      | ^~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:935:7: error: conflicting declaration of C function ‘void* clDeviceMemAllocINTEL(cl_context, cl_device_id, cl_mem_properties_intel*, size_t, cl_uint, cl_int*)’
  935 | void *clDeviceMemAllocINTEL(
      |       ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/dispatch.h:12,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:10,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/opt/pfx-suite/include/CL/cl_ext_intel.h:492:1: note: previous declaration ‘void* clDeviceMemAllocINTEL(cl_context, cl_device_id, const cl_mem_properties_intel*, size_t, cl_uint, cl_int*)’
  492 | clDeviceMemAllocINTEL(
      | ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:943:7: error: conflicting declaration of C function ‘void* clSharedMemAllocINTEL(cl_context, cl_device_id, cl_mem_properties_intel*, size_t, cl_uint, cl_int*)’
  943 | void *clSharedMemAllocINTEL(
      |       ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/dispatch.h:12,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:10,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/opt/pfx-suite/include/CL/cl_ext_intel.h:510:1: note: previous declaration ‘void* clSharedMemAllocINTEL(cl_context, cl_device_id, const cl_mem_properties_intel*, size_t, cl_uint, cl_int*)’
  510 | clSharedMemAllocINTEL(
      | ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:951:8: error: conflicting declaration of C function ‘cl_int clMemFreeINTEL(cl_context, const void*)’
  951 | cl_int clMemFreeINTEL(
      |        ^~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/dispatch.h:12,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:10,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/opt/pfx-suite/include/CL/cl_ext_intel.h:528:1: note: previous declaration ‘cl_int clMemFreeINTEL(cl_context, void*)’
  528 | clMemFreeINTEL(
      | ^~~~~~~~~~~~~~
make[4]: *** [igdrcl_lib_release/sharings/CMakeFiles/igdrcl_lib_release_sharings_enable.dir/build.make:109: igdrcl_lib_release/sharings/CMakeFiles/igdrcl_lib_release_sharings_enable.dir/va/enable_va.cpp.o] Error 1
```

JablonskiMateusz, 2020-09-15:
Hi @asavah,
Could you retry with [20.33.17675](https://github.com/intel/compute-runtime/releases/tag/20.33.17675) or newer release?

asavah, 2020-09-18:
```In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:126:20: error: conflicting declaration of C function ‘_cl_mem* clCreateBufferWithPropertiesINTEL(cl_context, const cl_mem_properties_intel*, size_t, void*, cl_int*)’
  126 | cl_mem CL_API_CALL clCreateBufferWithPropertiesINTEL(
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/dispatch.h:12,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/api/api.h:10,
                 from /home/asavah/pfx-suite/src/compute-runtime/opencl/source/sharings/va/enable_va.cpp:14:
/opt/pfx-suite/include/CL/cl_ext_intel.h:685:1: note: previous declaration ‘_cl_mem* clCreateBufferWithPropertiesINTEL(cl_context, const cl_mem_properties_intel*, cl_mem_flags, size_t, void*, cl_int*)’
  685 | clCreateBufferWithPropertiesINTEL(
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

JablonskiMateusz, 2020-10-07:
Our build scripts are prepared to compile with CL headers from our third party directory. These headers are used by default even if there are CL headers installed on the system. The fact that you ran into the issue indicates that you have set additional environment variables to force using installed CL headers. 

Could you please retry with this commit https://github.com/intel/compute-runtime/commit/155578568b544e529265f1eb48b71d9b8718a7b4 ? I hope it should cover the recent failure.
If the issue is still visible, could you provide steps how to reproduce this issue?

JacekDanecki, 2020-11-09:
Closing due to inactivity. Feel free to open new issue.

closed 2020-11-09 by JacekDanecki