intel/compute-runtime #339: Build fails without level zero headers

eero-t, 2020-07-13:
README doesn't list level zero as dependency.  But build fails if level zero headers are missing:
```
...
-- LevelZero_INCLUDE_DIRS: 
-- Level zero headers not found
-- Configuring done
CMake Error at cmake/run_ult_target.cmake:44 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:ze_intel_gpu_core_tests>

  No target "ze_intel_gpu_core_tests"
Call Stack (most recent call first):
  target_unit_tests/gen11/ehl/CMakeLists.txt:9 (include)
...
```
This is followed with huge amount of similar errors from the tests CMakeLists.txt file(s).

jandres742, 2020-07-13:
@eero-t : Please take a look at https://github.com/intel/compute-runtime/blob/master/level_zero/doc/BUILD.md

eero-t, 2020-07-13:
It builds fine with level zero headers.  This issue is about the build either succeeding without them (like earlier releases), or showing a sensible error message and that dependency being documented.

JablonskiMateusz, 2020-07-15:
The issue should be resolved since this commit https://github.com/intel/compute-runtime/commit/edf35e14a3ab8e967338c3c00dbff0b5dd2ade60 @eero-t  could you confirm that?

eero-t, 2020-08-03:
Sorry for the late reply.

> The issue should be resolved since this commit edf35e1 @eero-t could you confirm that?

With the latest release / tag, build starts now, but due to new issue #346, I can't verify whether the build will complete without level Zero headers.

The original issue is now fixed though, so I'll close this one.

closed 2020-08-03 by eero-t

eero-t, 2020-08-03:
Was able to verify that that build succeeds without level zero installed.  Thanks!