intel/compute-runtime #340: Command queue properties: error if there is no leading zero in properties array

boxerab, 2020-07-13:
`{0, CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE,0, 0}`  works but
`{CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE,0, 0}` gives error
when calling 
`clCreateCommandQueueWithProperties`.

Reproducer can be found [here](https://github.com/GrokImageCompression/latke)

To reproduced the error, change line 150 in `debayer.cpp`

```
  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO 
  Driver Version                                  20.25.17111
  Device OpenCL C Version                         OpenCL C 2.0 
```

JablonskiMateusz, 2020-07-15:
Hi @boxerab , I think you're using the OCL API in a wrong way. The properties array passed to `clCreateCommandQueueWithProperties` requires following format `{queuePropertyName0, queuePropertyValue0, queuePropertyName1, queuePropertyValue1, ... , 0}` . If you just specify property value without specifying property name then driver tries to interpret it as a property name and therefore it won't work.
For the case with leading zero, if you specify zero at the beginning of the properties, then driver takes it as a terminating zero and doesn't parse other properties. I assume that it is not what you want to achieve.

boxerab, 2020-07-15:
Yes, that was it! The khronos documentation doesn't really make this clear, unfortunately. Works fine now - thank you!

closed 2020-07-15 by boxerab