intel/compute-runtime #343: On heavy kernels execution OpenCL hogs CPU core

llmike, 2020-07-29:
https://github.com/intel/compute-runtime/blob/b776dd7e7a0487409848f6fc5d7554450c6af429/shared/source/command_stream/command_stream_receiver.cpp#L228

yield() call just let OS to do rescheduling if other threads or higher priority threads are available for execution, if not, this call does almost nothing and immediately return control to the caller. As the result OpenCL apps hog 100% of one CPU core when executing heavy kernels. With the same effect you could just remove this call.

Please consider usage of cond vars instead. Or just nanosleep() for 1ns, but nanosleep() could introduce some undesired delays, because kernel could sleep for a kernel tick size, not only for 1ns.

Thanks.

bfliflet, 2020-07-30:
Thanks, Mike, for pointing this out.  This is typically caused when someone enqueues a kernel and performs a serializing event such as clWaitForEvents or clFinish where the host must wait for the GPU to finish the kernel or other task.  Since the driver doesn't know how long the task will take to complete, it combines some polling with an event-driven mechanism.  In an ideal world, we'd just use the event driven mechanism as the power impact would be low.  Unfortunately, the only event-driven mechanism at our disposal currently has such a high latency that it negatively impacts performance of some workloads.  Because of this, we attempt to poll for small amount of time and then switch to the event-driven model to save power.  There's no easy way to answer how much to poll versus switch to event-driven.

We originally had a _sleep_ call along with the _yield_ and it was decided that _yield_ encompassed what _sleep_ was doing and then some.  It was also expected that _pause_ would provide a low power way to waste a few CPU clocks until GPU finished what it was doing.  I'd be curious if adding a _sleep_ provided the same latency at lower power on your system.  If that turns out to be true, we'll add it back.

If this was CPU to CPU communication, I agree, something like std::condition_variable would provide the right event-driven solution.

Imagine being told you need to go to work later but you had no idea when you'd be needed.  Now imagine your only options are to call your boss periodically to ask if needed right now or have your boss walk to your house and bang on your door.  Can you watch a 2hr movie?  Can you take a nap?  How long do you sleep?  Could you set an alarm?  That's challenge of GPU -> CPU notifications typical in GPGPU workloads.  There are improvements to this model on the horizon.  Stay tuned!

llmike, 2020-07-31:
I didn't look to the code too much. But I'm familiar with Mesa3D/i965 driver and it does just blocking calls to i915/DRM. In case of using clFInish() it is the perfect solution, in case of clWaitForEvents() it would be logical to throw CL_RUNNING before submission to i915/DRM and then CL_COMPLETE right after the blocking call return. Since currently Intel's implementation of OpenCL propagates to the userland only one device and one platform, it should not be too asynchronous.

bfliflet, 2020-08-07:
Blocking on i915/DRM is the high-latency, event driven mechanism I spoke of.  We'll take a look to see if this path isn't trickling into i915.  Depending on the duration of 'completion' in a given workload, we will intentionally spike in CPU utilization to provide a low latency completion for performance.

AdamCetnerowski, 2020-09-03:
Closing as answer has been provided. If you have follow up questions, please re-open or start a new issue.

closed 2020-09-03 by AdamCetnerowski