intel/compute-runtime #346: Compute-runtime build fails claiming earlier generated file to be missing, although there was no error on its generation

eero-t, 2020-08-03:
Although generating the *.gen input file(s) shows no errors:
```
[212/1865] Generating ../../../../bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.spv, ../../../../bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.bin, ../../../../bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.gen
[213/1865] Building CXX object igdrcl_lib/CMakeFiles/igdrcl_lib.dir/__/__/shared/source/compiler_interface/compiler_options/compiler_options_base.cpp.o
[214/1865] Generating ../../../../bin/built_ins/x64/gen12lp/copy_buffer_rect_Gen12LPlp.spv, ../../../../bin/built_ins/x64/gen12lp/copy_buffer_rect_Gen12LPlp.bin, ../../../../bin/built_ins/x64/gen12lp/copy_buffer_rect_Gen12LPlp.gen
```
Later build steps trying to generate *.cpp files from them, claim them to be missing:
```
[215/1865] Generating ../../../../bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.cpp
FAILED: bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.cpp 
cd /home/nobody/source/compute-runtime/shared/source/built_ins/kernels && /home/nobody/source/compute-runtime/build/bin/cpp_generate_tool --file /home/nobody/source/compute-runtime/build/bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.gen --output /home/nobody/source/compute-runtime/build/bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.cpp --array copy_buffer_to_buffer --platform Gen12LPlp
File cannot be opened!
```

(Above error message doesn't say it, but I checked from sources that it wanted the *.gen file.)

Directory where these files are being generated, is writable, but even if it wasn't, *.gen file generation should throw an error about it (if not, that should be fixed).

PS. I don't know whether it's relevant, but I'm building IGC with VC disabled, in a clean Docker container.

eero-t, 2020-08-03:
Note: I don't seem to get this issue when building IGC with Vector Compute support.

HoppeMateusz, 2020-09-10:
Hello,

is the .gen file (/home/nobody/source/compute-runtime/build/bin/built_ins/x64/gen12lp/copy_buffer_to_buffer_Gen12LPlp.gen) actually created in the first step:   [214/1865] Generating ...  ?

Starting from: 86b133207fe8fc0f11ae7aa538ec727ef2e8c30b build log should have errors printed when generation fails.

Can you share which version of NEO and IGC are you using?

eero-t, 2020-09-17:
I was using the latest libdrm, gmmlib, media-driver, IGC and compute-runtime releases available at the time when I filed this bug.

Note: to disable VC, I had to patch IGC (I was disabling VC because I had other build failures when VC was enabled, which I was able to resolve a bit later on), but this is fixed in the latest IGC release.

With the latest releases version of the whole stack, compute-runtime *build succeeds even when VC is disabled* from IGC.  And with the error output when file generation fails, I think this can be declared fixed.  Feel free to close! :-)

closed 2020-09-17 by AdamCetnerowski