intel/compute-runtime #347: Build fails with latest level Zero (v1.0) installed

eero-t, 2020-08-03:
Building compute-runtime with the latest level Zero checkout breaks because there seems to have been radical change in the installed level zero header file names:
```
FAILED: level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/gen11/icllp/cmdqueue_icllp.cpp.o 
/usr/bin/c++    @level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/gen11/icllp/cmdqueue_icllp.cpp.o.rsp -MD -MT level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/gen11/icllp/cmdqueue_icllp.cpp.o -MF level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/gen11/icllp/cmdqueue_icllp.cpp.o.d -o level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/gen11/icllp/cmdqueue_icllp.cpp.o -c ../level_zero/core/source/gen11/icllp/cmdqueue_icllp.cpp
In file included from ../level_zero/core/source/cmdlist/cmdlist.h:13,
                 from ../level_zero/core/source/cmdqueue/cmdqueue_hw.inl:28,
                 from ../level_zero/core/source/gen11/icllp/cmdqueue_icllp.cpp:11:
../level_zero/core/source/cmdqueue/cmdqueue.h:11:10: fatal error: level_zero/ze_common.h: No such file or directory
   11 | #include <level_zero/ze_common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Latest L0 v1.0 doesn't any more install header named like that.  Build with L0 v0.91.21 works fine.

L0 check should check that the major version matches to what compute-runtime expects.

jandres742, 2020-08-03:
This is indicated in the Release Notes for each release. Support for 1.0 is not released yet.

20.30.17454

Packages included in the release
intel-gmmlib (https://github.com/intel/gmmlib)
intel-igc-core and intel-igc-opencl (https://github.com/intel/intel-graphics-compiler)
intel-opencl, intel-ocloc, intel-level-zero-gpu (https://github.com/intel/compute-runtime)

Components revisions included in the release
intel/compute-runtime@20.30.17454
intel/gmmlib@intel-gmmlib-20.2.2
intel/intel-graphics-compiler@igc-1.0.4427

Additional components revisions used in build
intel/libva@c9bb65b (Compatible with va_api_major_version = 1)
intel/llvm-patches@5005890
intel/opencl-clang@6a9cd2c
KhronosGroup/SPIRV-LLVM-Translator@424e375 (for opencl-clang)
intel/vc-intrinsics@d7c5f99
KhronosGroup/SPIRV-LLVM-Translator@e87b59a (for vector compiler)
llvm/llvm-project@llvmorg-10.0.0
**oneapi-src/level-zero@v0.91.10 (Compatible with oneAPI Level Zero API Spec v0.91)**

eero-t, 2020-08-04:
This bug is about compute-runtime build configure phase not checking that there's compatible L0 version available.

And list of binary build artifacts stating that L0 v0.91 build binary is compatible with L0 spec of same version, is not documentation stating that *compute-runtime sources* will build only with that particular version.

eero-t, 2020-08-28:
Few latest releases build fine with L0 v1.0 headers (as noted in release notes).  I didn't check whether the L0 version check is fixed though.

jandres742, 2020-08-30:
Correct. 1.0 is supported since 20.32.17625. L0 version check hasn't been implemented.

eero-t, 2020-08-31:
I added related issue for L0: https://github.com/oneapi-src/level-zero/issues/37

It providing pkg-config file would simplify checking the version.

jandres742, 2020-09-12:
L0 specification is currently in 1.0 and is expected to stay stable for quite some time. Hence, there is no urgent need of implementing the check. We could implement it in the future in the driver when more versions of the L0 headers becomes available.

Closing it.

closed 2020-09-12 by jandres742

eero-t, 2020-10-07:
There are now more (1.0.x) versions of L0, and L0 project added pkg-config file so that it's versions are trivial to check, see: oneapi-src/level-zero#37

=> I think this should be re-opened so that it can be fixed.

jandres742, 2020-10-07:
>There are now more (1.0.x) versions of L0

Those are minor revisions, which have same ABI, so there no concerns about incompatibilities. But we can consider adding it. 

> so that it can be fixed.

Could you confirm whether there something is broken that requires this change to get it fixed?

eero-t, 2020-10-08:
> Could you confirm whether there something is broken that requires this change to get it fixed?

Well, like before, user could have L0 version from before 1.0.  It doesn't provide pkg-config, so that's easy to detect.

And there are going to be further versions of L0. E.g. Sysman API has few rough edges:
- API giving number of items, but doc telling user to allocate more (API for that should be done differently so that it's reliable)
- Sampling APIs not telling at which rate data can change, so that user would know whether the provided data is random (due to too low sampling rate)
- No tracing infrastructure to provide data when sampling doesn't provide reliable data (e.g. for kernel GPU freq requests)

eero-t, 2020-10-08:
Issue stated in the current bug title is fixed as compute-runtime does support latest L0, although the root-cause for it isn't.  I can open new "detect system L0 version / compatibility" ticket if that's preferred.