intel/compute-runtime #348: undefined symbol GmmAdapterDistroy

khaled-rahman, 2020-08-04:
I'm trying to build this on a gen9 machine. When I use the command `make`, it shows me the following error message after the completion of building > 88% of the code. What could be the possible reason and how can I resolve this issue? It would be helpful.

[ 90%] Built target igdrcl_aub_tests
re-creating working directory for icllp AUBs generation...
Running AUB generation for icllp in .../compute-runtime/build/bin/icllp_aub
.../compute-runtime/build/bin/igdrcl_aub_tests: symbol lookup error: .../compute-runtime/build/bin/igdrcl_aub_tests: undefined symbol: GmmAdapterDestroy
opencl/test/unit_test/aub_tests/gen11/icllp/CMakeFiles/run_icllp_aub_tests.dir/build.make:81: recipe for target 'run_icllp_aub_tests' failed
make[2]: *** [run_icllp_aub_tests] Error 127
CMakeFiles/Makefile2:11920: recipe for target 'opencl/test/unit_test/aub_tests/gen11/icllp/CMakeFiles/run_icllp_aub_tests.dir/all' failed
make[1]: *** [opencl/test/unit_test/aub_tests/gen11/icllp/CMakeFiles/run_icllp_aub_tests.dir/all] Error 2
Makefile:171: recipe for target 'all' failed
make: *** [all] Error 2

JablonskiMateusz, 2020-08-04:
It looks like an issue with incompatible gmmlib library - there is a failure during lookup for a symbol `GmmAdapterDestroy`.

As aub tests use gmmlib library built in workspace or installed in the system, please make sure you have proper version of gmmlib - based on our release notes.

khaled-rahman, 2020-08-04:
Hi, I have tried with the debian files. It was installed but the problem is it can not access GPU.
For example, with this release installed, if I type `clinfo`, it shows me the following message: (You can see that driver version and device type)
```
Platform Name                                   Intel(R) OpenCL
Number of devices                                 1
  Device Name                                     Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 (Build 0)
  Driver Version                                  2020.10.7.0.15
  Device OpenCL C Version                         OpenCL C 2.0
  Device Type                                     CPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
```
But if I install a previous version, it can find the GPU.
```
Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO
  Driver Version                                  19.07.12410
  Device OpenCL C Version                         OpenCL C 2.0
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
```
Why this problem is seen with the latest release? Is it related to any other issue? It would be helpful to know.

JablonskiMateusz, 2020-08-04:
The driver you have installed (`19.07.12410`) is quite old. If you want to build the newer neo without building the dependencies from source or just to install our package `intel-opencl`, you should also update `intel-gmmlib`, `intel-igc-core` and `intel-igc-opencl` packages, you can find the binaries in our release's assets: https://github.com/intel/compute-runtime/releases

khaled-rahman, 2020-08-04:
I have installed all debian packages mentioned in the releases note (https://github.com/intel/compute-runtime/releases). I downloaded them as directed in the instructions (using `wget`). Now it shows `Number of platforms 0`. I am trying to install it on 18.04.4 LTS (Bionic Beaver). In the release note, I saw that it works for Ubuntu 18.04. I was wondering to know what is happening in my case.  What other information would be helpful to find out the reason?

khaled-rahman, 2020-08-05:
Hi @JablonskiMateusz, any comment on my previous issue? It would be helpful to know.

JablonskiMateusz, 2020-08-06:
Could you provide file log.txt created by command:

`strace -f -s 255 -o log.txt clinfo`
?

khaled-rahman, 2020-08-06:
I have attached the log.txt file. Please check it out.

[log.txt](https://github.com/intel/compute-runtime/files/5034759/log.txt)

JablonskiMateusz, 2020-08-06:
In the log I see `2087  openat(AT_FDCWD, "/opt/intel/mediasdk/lib64/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 4`

Similar issue was seen in https://github.com/intel/compute-runtime/issues/325

khaled-rahman, 2020-08-06:
Thanks. It worked!

closed 2020-08-06 by khaled-rahman