intel/compute-runtime #350: clinfo number of platforms 0

ye-luo, 2020-08-11:
On my laptop ubuntu 20.04
Add source `https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl`
After installing intel-opencl-icd I got
```
$ clinfo 
Number of platforms                               0
$ ls /etc/OpenCL/vendors/
intel-cpu.icd  intel.icd
```
I'm wondering why no GPU can be found.

jandres742, 2020-08-11:
Did you try with sudo? You might be missing being part of the video/render groups to get access to the GPU.

ye-luo, 2020-08-11:
Adding users to the render group solves my issue. I didn't try the video group.

MAMorgado, 2021-07-01:
Hello ye-luo. I'm having the same problem. I tried with sudo install but still I get the same problem.
Can you help me?

JacekDanecki, 2021-07-01:
Can you provide file log.txt created by command:
```
strace -f -s 255 -o log.txt clinfo
```

MAMorgado, 2021-07-02:
Hi Jacek,
I've opened another issue (https://github.com/intel/compute-runtime/issues/428) where I try to better explain my problem. Right now I don't get this error anymore but I get something else.
But I can recreate the problem again if you'd prefer it and post the log as you request. 
Let me know what do you think would be best.
Thank you very much,
MaM

MAMorgado, 2021-07-02:
Either way here is my current log.txt
[log.txt](https://github.com/intel/compute-runtime/files/6753384/log.txt)

AdamCetnerowski, 2021-07-08:
Closing this and will continue in #428

closed 2021-07-08 by AdamCetnerowski

Cherish-Gww, 2024-04-08:
> Adding users to the render group solves my issue. I didn't try the video group.

`$ usermod -aG video <user name>`