intel/compute-runtime #351: Getting debug section from Intel GPU binary

aarontcopal2, 2020-08-19:
I use the function call `clGetProgramInfo(program, CL_PROGRAM_BINARIES, device_count * sizeof(uint8_t*), binary, nullptr); `to get the GPU binary. The sections that i get inside this binary are:
1. BuildOptions
2. SPIRV Object
3. Intel(R) OpenCL Device Binary
4. .shstrtab

Is there any buildtime flags that i can pass to get the debug_info too as part of this binary?

HoppeMateusz, 2020-08-28:
Hello,

This commit :  https://github.com/intel/compute-runtime/commit/856a2f7be7fd7599f1f23e9a9d3c68faba4fafea
adds debug data to the binary queried with clGetProgramInfo(program, CL_PROGRAM_BINARIES ... when debug info exists.

You should build your program with "-g" option to have debug info available. 
Optionally, you can also use -cl-opt-disable to disable optimizations.

aarontcopal2, 2020-08-28:
thats great news! Whats the latest release that has this commit?

AdamCetnerowski, 2020-09-03:
This went in as build 17756, so it hasn't been included in a weekly release yet. Likely to show up this week.

JacekDanecki, 2020-09-11:
Today I've released [Neo 20.36.17813](https://github.com/intel/compute-runtime/releases/tag/20.36.17813). Can you check with this version?

AdamCetnerowski, 2020-10-09:
Issue has been resolved.

closed 2020-10-09 by AdamCetnerowski