intel/compute-runtime #352: reproducibility on copy_image3d_to_buffer_stateless_Gen12LPlp and scheduler_Gen12LPlp

cheeyanglee, 2020-08-26:
Facing reproducibility issue on copy_image3d_to_buffer_stateless_Gen12LPlp and scheduler_Gen12LPlp, while other build_ins can be reproduce, 

Attached diffoscope results [intel-compute-runtime-src.log](https://github.com/intel/compute-runtime/files/5128997/intel-compute-runtime-src.log)

JacekDanecki, 2020-09-10:
Can you provide steps how to reproduce this issue?

cheeyanglee, 2020-09-14:
i use yocto build system to build it. seems like copy_image3d_to_buffer_stateless_Gen12LPlp can now be reproduce after 20.33.17675, but not for scheduler_Gen12LPlp .
attached the build log and build steps from yocto, hopefully these could help.

[diffoscope-intel-compute-runtime-src-20.33.17675.log](https://github.com/intel/compute-runtime/files/5216616/diffoscope-intel-compute-runtime-src-20.33.17675.log)

[run.do_configure.3230554.txt](https://github.com/intel/compute-runtime/files/5216604/run.do_configure.3230554.txt)
[run.do_compile.3231112.txt](https://github.com/intel/compute-runtime/files/5216603/run.do_compile.3231112.txt)

[log.do_configure.3230554.txt](https://github.com/intel/compute-runtime/files/5216602/log.do_configure.3230554.txt)
[log.do_compile.3231112.txt](https://github.com/intel/compute-runtime/files/5216601/log.do_compile.3231112.txt)

AdamCetnerowski, 2020-11-03:
@cheeyanglee - I am trying to understand what you are attempting to do and what is your issue:

You are trying to compile NEO on Yocto and compilation of some built ins fail? Please elaborate.

cheeyanglee, 2020-11-04:
follow the reproducible build practice[1], i should be able to always build/produce identical binaries on different build. 
This works on most the build ins, However scheduler_Gen12LPlp is creating different binaries on different build.

i tried to look for the causes, found that it is creating different .cpp code on different build which causes the binaries to be different. 

[1] https://reproducible-builds.org/

AdamCetnerowski, 2020-12-02:
This looks like a compiler (IGC) issue. We are investigating and will update once we have more information.

JacekDanecki, 2021-01-08:
I'm transferring this issue to IGC project: https://github.com/intel/intel-graphics-compiler