intel/compute-runtime #355: how to install pre-built driver on ubuntu 16.04

guoyejun, 2020-09-11:
Hi,

I'm using ubuntu 16.04 and failed to install pre-built drivers according to https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl, i also capture the info below.
[image]

Here are my steps:
$ sudo add-apt-repository ppa:intel-opencl/intel-opencl
$ sudo apt-get update

add the following at the end of /etc/apt/sources.list
deb http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu xenial main 
deb-src http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu xenial main 

$ sudo apt-get update
$ clinfo
Number of platforms                               0
$ sudo apt-get install  libigc-dev libigdgmm-dev   # no error with this command

$ sudo apt-get install intel-opencl
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 intel-opencl : Depends: intel-igc-opencl but it is not installable
                Depends: intel-gmmlib but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

$ sudo apt-get install intel-igc-opencl
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package intel-igc-opencl is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libigc

E: Package 'intel-igc-opencl' has no installation candidate

$ sudo apt-get install libigc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libigc is already the newest version (1.0.3586-1~ppa1~xenial1).

JacekDanecki, 2020-09-11:
Neo package on launchpad is named as intel-opencl-icd , so you need to install it instead of intel-opencl. Where did you get intel-opencl package for xenial from?
```
$ apt-get install intel-opencl-icd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libigc libigdgmm11
The following NEW packages will be installed:
  intel-opencl-icd libigc libigdgmm11
0 upgraded, 3 newly installed, 0 to remove and 151 not upgraded.
Need to get 30.3 MB of archives.
After this operation, 119 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu xenial/main amd64 libigdgmm11 amd64 19.4.1-1~ppa1~xenial1 [109 kB]
Get:2 http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu xenial/main amd64 libigc amd64 1.0.3586-1~ppa1~xenial1 [29.2 MB]
Get:3 http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu xenial/main amd64 intel-opencl-icd amd64 20.12.16259-1~ppa1~xenial2 [997 kB]
```
and then
```
$ clinfo -l    
Platform #0: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) Gen9 HD Graphics NEO
```
Support for Ubuntu 16.04 was dropped on IGC side, so I can't provide newer Neo packages on launchpad for Xenial. Could you switch to Ubuntu 18.04 or 20.04?

guoyejun, 2020-09-11:
thanks, i just get the package name for the website (in the figure I posted: overview of published packages),  ok, let me switch to ubuntu 18.04

guoyejun, 2020-09-12:
hi, i upgraded to Ubuntu 18.04.5 LTS (GNU/Linux 4.15.0-117-generic x86_64), there is error when install the .deb files according to https://github.com/intel/compute-runtime/releases/tag/20.35.17767, but clinfo works, are these errors expected? thanks.

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04
Codename:       bionic

$ sha256sum *.deb
e131650d40583fdbef341dc52d84733398a68437437fd3fbcfc689d50334bc63  intel-gmmlib_20.2.4_amd64.deb
fecda8345a376e8ef26f0d581190939259c759c007e6326d7280b39b0ee61074  intel-igc-core_1.0.4756_amd64.deb
80e8096fb0471fc82bedebc8978a385369ad65e41b9d0d80d38519f8ee6eb379  intel-igc-opencl_1.0.4756_amd64.deb
8aedee4507b4d8693d9d16ef7e3b55d4a7198e40ced8f0467a8a0e18577e7615  intel-level-zero-gpu_1.0.17767_amd64.deb
de13349ccbbadd9b128697f04edb2757ec4df4e1ab9f8635f762593d14be0222  intel-ocloc_20.35.17767_amd64.deb
e734c7d033f68f9841323498b45b0171e337dc312e09133bf0529a5c6177981c  intel-opencl_20.35.17767_amd64.deb

$ sudo dpkg -i *.deb
(Reading database ... 235285 files and directories currently installed.)
Preparing to unpack intel-gmmlib_20.2.4_amd64.deb ...
Unpacking intel-gmmlib (20.2.4) over (20.2.4) ...
Preparing to unpack intel-igc-core_1.0.4756_amd64.deb ...
Unpacking intel-igc-core (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-igc-opencl_1.0.4756_amd64.deb ...
Unpacking intel-igc-opencl (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-level-zero-gpu_1.0.17767_amd64.deb ...
Unpacking intel-level-zero-gpu (1.0.17767) over (1.0.17767) ...
Preparing to unpack intel-ocloc_20.35.17767_amd64.deb ...
Unpacking intel-ocloc (20.35.17767) over (20.35.17767) ...
Preparing to unpack intel-opencl_20.35.17767_amd64.deb ...
Unpacking intel-opencl (20.35.17767) over (20.35.17767) ...
Setting up intel-gmmlib (20.2.4) ...
dpkg: dependency problems prevent configuration of intel-igc-core:
 libigc (1.0.3586-1~ppa1~xenial1) breaks intel-igc-core and is installed.

dpkg: error processing package intel-igc-core (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-igc-opencl:
 intel-igc-opencl depends on intel-igc-core (= 1.0.4756); however:
  Package intel-igc-core is not configured yet.
 libigc (1.0.3586-1~ppa1~xenial1) breaks intel-igc-opencl and is installed.

dpkg: error processing package intel-igc-opencl (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-level-zero-gpu:
 intel-level-zero-gpu depends on intel-igc-opencl (= 1.0.4756); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-level-zero-gpu (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-ocloc:
 intel-ocloc depends on intel-igc-opencl (= 1.0.4756); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-ocloc (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-opencl:
 intel-opencl depends on intel-igc-opencl (= 1.0.4756); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-opencl (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.27-3ubuntu1.2) ...
Errors were encountered while processing:
 intel-igc-core
 intel-igc-opencl
 intel-level-zero-gpu
 intel-ocloc
 intel-opencl

$ clinfo
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.1 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_il_program cl_khr_fp64 cl_khr_subgroups cl_intel_spirv_device_side_avc_motion_estimation cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory_preview cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_device_side_avc_motion_estimation cl_intel_advanced_motion_estimation cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_intel_media_block_io cl_khr_3d_image_writes cl_intel_va_api_media_sharing 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO 
  Driver Version                                  20.35.17767
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               24
  Max clock frequency                             1050MHz
...

JacekDanecki, 2020-09-14:
These errors are expected because you can't mix packages from launchpad with packages from github releases. Before installing packages from github, please remove packages installed from launchpad.
When you install packages from launchpad on system with github packages installed installer will remove github packages automatically.

guoyejun, 2020-09-15:
thanks, i tried to remove packages with:
$ sudo apt-get remove intel-gmmlib intel-igc-core intel-igc-opencl intel-level-zero-gpu intel-ocloc intel-opencl libigc-dev libigdgmm-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libigc-dev
E: Unable to locate package libigdgmm-dev

but, still meet the same issue, 
$ sudo dpkg -i *.deb
(Reading database ... 235312 files and directories currently installed.)
Preparing to unpack intel-gmmlib_20.2.4_amd64.deb ...
Unpacking intel-gmmlib (20.2.4) over (20.2.4) ...
Preparing to unpack intel-igc-core_1.0.4756_amd64.deb ...
Unpacking intel-igc-core (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-igc-opencl_1.0.4756_amd64.deb ...
Unpacking intel-igc-opencl (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-level-zero-gpu_1.0.17767_amd64.deb ...
Unpacking intel-level-zero-gpu (1.0.17767) over (1.0.17767) ...
Preparing to unpack intel-ocloc_20.35.17767_amd64.deb ...
Unpacking intel-ocloc (20.35.17767) over (20.35.17767) ...
Preparing to unpack intel-opencl_20.35.17767_amd64.deb ...
Unpacking intel-opencl (20.35.17767) over (20.35.17767) ...
Setting up intel-gmmlib (20.2.4) ...
dpkg: dependency problems prevent configuration of intel-igc-core:
 libigc (1.0.3586-1~ppa1~xenial1) breaks intel-igc-core and is installed.

dpkg: error processing package intel-igc-core (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-igc-opencl:
 intel-igc-opencl depends on intel-igc-core (= 1.0.4756); however:
  Package intel-igc-core is not configured yet.
 libigc (1.0.3586-1~ppa1~xenial1) breaks intel-igc-opencl and is installed.

dpkg: error processing package intel-igc-opencl (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-level-zero-gpu:
 intel-level-zero-gpu depends on intel-igc-opencl (= 1.0.4756); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-level-zero-gpu (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-ocloc:
 intel-ocloc depends on intel-igc-opencl (= 1.0.4756); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-ocloc (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of intel-opencl:
 intel-opencl depends on intel-igc-opencl (= 1.0.4756); however:
  Package intel-igc-opencl is not configured yet.

dpkg: error processing package intel-opencl (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.27-3ubuntu1.2) ...
Errors were encountered while processing:
 intel-igc-core
 intel-igc-opencl
 intel-level-zero-gpu
 intel-ocloc
 intel-opencl

JacekDanecki, 2020-09-15:
You did not remove libigc.

guoyejun, 2020-09-16:
hi, 

i now have a new idea to build the compute-runtime from source code on my ubuntu 18.04 system which has installed the .deb files successfully.

to make the thing simpler, i don't want to build others (such as igc) from source code, i just want to build only the compute-runtime, and i met issue to run cmake:
```
$ cmake ..             
-- branch dir list: /
-- Release build configuration
-- Computed OpenCL version major is: 20
-- Computed OpenCL version minor is: 37
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- Source Level Debugger headers dir: /work/gfx/compute-runtime/third_party/source_level_debugger
-- Aub Stream Headers dir: /work/gfx/compute-runtime/third_party/aub_stream/headers
-- Metrics Library dir: /work/gfx/compute-runtime/third_party/metrics_library
-- Metrics Discovery dir: /work/gfx/compute-runtime/third_party/metrics_discovery
-- i915 includes dir: /work/gfx/compute-runtime/third_party/uapi
-- Khronos OpenCL headers dir: /work/gfx/compute-runtime/third_party/opencl_headers
-- Khronos OpenGL headers dir: /work/gfx/compute-runtime/third_party/opengl_headers
-- Third party dir: /work/gfx/third_party
-- Checking for module 'igc-opencl'
--   No package 'igc-opencl' found
CMake Error at CMakeLists.txt:400 (message):
  Intel Graphics Compiler not found!
```

of course, i already installed intel-igc-opencl_1.0.4756_amd64.deb 
$ dpkg -L intel-igc-opencl    
/.
/usr
/usr/local
/usr/local/lib
/usr/local/lib/libigdfcl.so.1.0.4756
/usr/local/lib/libopencl-clang.so.10
/usr/local/lib/libigdfcl.so
/usr/local/lib/libigdfcl.so.1
/usr/local/lib/libopencl-clang.so

what else i need to install? thanks.

JacekDanecki, 2020-09-16:
You need to install devel packages, as described in https://github.com/intel/compute-runtime/blob/master/BUILD.md
For IGC devel package you can use  intel-igc-opencl-devel_1.0.4756_amd64.deb from https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.4756
gmmlib devel packages you can find on ~ocl-dev launchpad account: https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl/+packages?field.name_filter=&field.status_filter=&field.series_filter=bionic

guoyejun, 2020-09-17:
thanks

> You need to install devel packages, as described in https://github.com/intel/compute-runtime/blob/master/BUILD.md
> For IGC devel package you can use intel-igc-opencl-devel_1.0.4756_amd64.deb from https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.4756
ok, i downloaded intel-igc-opencl-devel_1.0.4756_amd64.deb

> gmmlib devel packages you can find on ~ocl-dev launchpad account: https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl/+packages?field.name_filter=&field.status_filter=&field.series_filter=bionic
i have had intel-gmmlib_20.2.4_amd64.deb on my system, and i could not find the exact same version for dev package, so I downloaded intel-gmmlib-dev_20.2.4-1_ppa1_bionic1_amd64.deb

then, i installed the .deb files, but there is error. Is there a single place that I can get all the necessary packages including dev packages? thanks.
```
$ sudo dpkg -i *.deb
(Reading database ... 235505 files and directories currently installed.)
Preparing to unpack intel-gmmlib_20.2.4_amd64.deb ...
Unpacking intel-gmmlib (20.2.4) over (20.2.4) ...
Preparing to unpack intel-gmmlib-dev_20.2.4-1_ppa1_bionic1_amd64.deb ...
Unpacking intel-gmmlib-dev:amd64 (20.2.4-1~ppa1~bionic1) over (20.2.4-1~ppa1~bionic1) ...
Preparing to unpack intel-igc-core_1.0.4756_amd64.deb ...
Unpacking intel-igc-core (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-igc-opencl_1.0.4756_amd64.deb ...
Unpacking intel-igc-opencl (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-igc-opencl-devel_1.0.4756_amd64.deb ...
Unpacking intel-igc-opencl-devel (1.0.4756) over (1.0.4756) ...
Preparing to unpack intel-level-zero-gpu_1.0.17767_amd64.deb ...
Unpacking intel-level-zero-gpu (1.0.17767) over (1.0.17767) ...
Preparing to unpack intel-ocloc_20.35.17767_amd64.deb ...
Unpacking intel-ocloc (20.35.17767) over (20.35.17767) ...
Preparing to unpack intel-opencl_20.35.17767_amd64.deb ...
Unpacking intel-opencl (20.35.17767) over (20.35.17767) ...
Setting up intel-gmmlib (20.2.4) ...
dpkg: dependency problems prevent configuration of intel-gmmlib-dev:amd64:
 intel-gmmlib-dev:amd64 depends on intel-gmmlib (= 20.2.4-1~ppa1~bionic1); however:
  Version of intel-gmmlib on system is 20.2.4.

dpkg: error processing package intel-gmmlib-dev:amd64 (--install):
 dependency problems - leaving unconfigured
Setting up intel-igc-core (1.0.4756) ...
Setting up intel-igc-opencl (1.0.4756) ...
Setting up intel-igc-opencl-devel (1.0.4756) ...
Setting up intel-level-zero-gpu (1.0.17767) ...
Setting up intel-ocloc (20.35.17767) ...
Setting up intel-opencl (20.35.17767) ...
Processing triggers for libc-bin (2.27-3ubuntu1.2) ...
Errors were encountered while processing:
 intel-gmmlib-dev:amd64
```

then, I tried cmake for compute runtime, should I also install level zero dev package anywhere?
```
$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- branch dir list: /
-- CMAKE_BUILD_TYPE not specified, using Release
-- Release build configuration
-- Computed OpenCL version major is: 20
-- Computed OpenCL version minor is: 38
-- GTest repeat count set to 1
[... 74 lines cut ...]
/work/gfx/compute-runtime/shared/source/gen12lp/hw_info_rkl.h:16:17: note: suggested alternative: ‘IGFX_ICELAKE’
 struct HwMapper<IGFX_ROCKETLAKE> {
                 ^~~~~~~~~~~~~~~
                 IGFX_ICELAKE
/work/gfx/compute-runtime/shared/source/gen12lp/hw_info_rkl.h:16:32: error: template argument 1 is invalid
 struct HwMapper<IGFX_ROCKETLAKE> {
                                ^
offline_compiler/source/CMakeFiles/ocloc_lib.dir/build.make:179: recipe for target 'offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/helpers/hw_info.cpp.o' failed
make[2]: *** [offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/helpers/hw_info.cpp.o] Error 1
CMakeFiles/Makefile2:252: recipe for target 'offline_compiler/source/CMakeFiles/ocloc_lib.dir/all' failed
make[1]: *** [offline_compiler/source/CMakeFiles/ocloc_lib.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

and here is my source code version:
$ git log
commit d2b218d82d1f1ae32737280d7374e76957ca39ba (HEAD -> master, origin/master, origin/HEAD)
Author: Vinod Tipparaju <vinod.tipparaju@intel.com>
Date:   Thu Sep 3 17:45:25 2020 +0530

    Return ZE_RESULT_OUT_OF_MEMORY when alloc on device fails during cmdlist create
    
    Change-Id: Ia03e7ac190598c56de044d3ad8216087b8da94f2
    Signed-off-by: Vinod Tipparaju <vinod.tipparaju@intel.com>

guoyejun, 2020-09-18:
anyone can help? thanks.

JacekDanecki, 2020-09-18:
As I said earlier, you can't mix packages from github with packages from launchpad, so please remove packages from github and install igc and gmmlib packages from ~ocl-dev ppa. Currently there are intel-gmmlib 20.2.5 and igc 1.0.4882 on launchpad, but igc packages will be updated soon. 
You need to install level zero packages only when you want to build level zero support in Neo.

guoyejun, 2020-09-18:
thanks i understood now.

I uninstalled deb files from github and apt-get install the two dev packages (intel-gmmlib-dev and libigc-dev) from ~ocl-dev ppa, but still same build issue:
```
[  0%] Building CXX object offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/helpers/hw_info.cpp.o
In file included from /work/gfx/compute-runtime/shared/source/gen12lp/hw_info.h:16:0,
                 from /work/gfx/compute-runtime/shared/source/gen12lp/hw_cmds_base.h:10,
                 from /work/gfx/compute-runtime/shared/source/gen12lp/hw_cmds.h:9,
                 from /work/gfx/compute-runtime/shared/source/helpers/definitions/hw_cmds.h:19,
                 from /work/gfx/compute-runtime/shared/source/helpers/hw_info.cpp:12:
/work/gfx/compute-runtime/shared/source/gen12lp/hw_info_rkl.h:16:17: error: ‘IGFX_ROCKETLAKE’ was not declared in this scope
 struct HwMapper<IGFX_ROCKETLAKE> {
                 ^~~~~~~~~~~~~~~
/work/gfx/compute-runtime/shared/source/gen12lp/hw_info_rkl.h:16:17: note: suggested alternative: ‘IGFX_ICELAKE’
 struct HwMapper<IGFX_ROCKETLAKE> {
                 ^~~~~~~~~~~~~~~
                 IGFX_ICELAKE
/work/gfx/compute-runtime/shared/source/gen12lp/hw_info_rkl.h:16:32: error: template argument 1 is invalid
 struct HwMapper<IGFX_ROCKETLAKE> {
                                ^
offline_compiler/source/CMakeFiles/ocloc_lib.dir/build.make:179: recipe for target 'offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/helpers/hw_info.cpp.o' failed
```

JacekDanecki, 2020-09-18:
Can you provide output from cmake?
I've downloaded gmmlib 20.2.5 from ~ocl-dev ppa: https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl/+files/intel-gmmlib-dev_20.2.5-1~ppa1~bionic1_amd64.deb
and extracted deb package:
```
dpkg-deb -x intel-gmmlib-dev_20.2.5-1~ppa1~bionic1_amd64.deb .
```
and IGFX_ROCKETLAKE is defined in this package:
```
grep -r ROCKET usr 
usr/include/igdgmm/inc/common/igfxfmid.h:    IGFX_ROCKETLAKE,
```

The same is for gmmlib 20.2.4: https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl/+files/intel-gmmlib-dev_20.2.4-1~ppa1~bionic1_amd64.deb

guoyejun, 2020-09-21:
thanks @JacekDanecki i found the root cause, i once tried to build/install gmmlib from source code early this year, so there's two igfxfmid.h in my system, and neo finds my old version of this file.

closed 2020-09-22 by guoyejun