intel/compute-runtime #356: fedora packages on copr: mismatch of intel-gmmlib and intel-opencl versions

garrett, 2020-09-14:
Thanks for working on OpenCL support on Intel GPUs! I had been happily using it from copr up until this weekend (when I reinstalled my laptop).

When attempting to install `intel-opencl` on Fedora 33 beta (but also similar results for F32), I get the following:

```
$ dnf install intel-opencl
Last metadata expiration check: 0:05:41 ago on Mon 14 Sep 2020 01:25:56 PM CEST.
Error: 
 Problem: conflicting requests
  - nothing provides intel-gmmlib = 20.2.2 needed by intel-opencl-20.28.17293-2.fc33.x86_64
(try to add '--skip-broken' to skip uninstallable packages)
```

It seems copr's latest build of `intel-gmmlib` is from 2020-Apr-09 whereas `intel-opencl` is from 2020-Jul-22.

Triggering a rebuild of both gmmlib and opencl (and possibly other packages) would probably fix the issue.

The last time I see both gmmlib and opencl packages together is at https://sourceforge.net/projects/intel-compute-runtime/files/20.25.17111/fedora-33/

garrett, 2020-09-14:
Aha, in issue https://github.com/intel/compute-runtime/issues/196, intel-opencl has been built on Fedora's packaged intel-gmmlib. However, Fedora is shipping intel-gmmlib-20.2.5-2.fc33.x86_64 whereas the package expects 20.2.2 exactly (not 20.2.2 or higher).

JacekDanecki, 2020-09-16:
You can use gmmlib 20.2.2 for Fedora 33 from koji: https://koji.fedoraproject.org/koji/buildinfo?buildID=1528694
Starting with Neo 20.29.17408, IGC introduced new components in build, so my specs for copr stopped to work.
Restoring copr builds is in my "todo" queue.

garrett, 2020-09-17:
> You can use gmmlib 20.2.2 for Fedora 33 from koji

Thanks for the work-around! I'm attaching the RPMs here (in zip files, so GitHub accepts them in this comment), in case they're removed from koji.

- [intel-gmmlib-20.2.2-1.fc33.src.rpm.zip](https://github.com/intel/compute-runtime/files/5237478/intel-gmmlib-20.2.2-1.fc33.src.rpm.zip)
- [intel-gmmlib-i686.zip](https://github.com/intel/compute-runtime/files/5237479/intel-gmmlib-i686.zip)
- [intel-gmmlib-x86_64.zip](https://github.com/intel/compute-runtime/files/5237480/intel-gmmlib-x86_64.zip)

> Restoring copr builds is in my "todo" queue.

Thanks! Looking forward to this.

(Hopefully it can get into Fedora officially at some point in the near future too.)

JacekDanecki, 2020-11-09:
Packages for:
* Neo 20.44.18297
* IGC 1.0.5353 
* L0 1.0.16
were built on copr for fedora 33.
Neo is using gmmlib from fedora (updates-testing):
```
dnf install intel-opencl
Last metadata expiration check: 0:00:17 ago on Mon Nov  9 22:40:25 2020.
Dependencies resolved.
==============================================================================================================================================================================================================================================================
 Package                                               Architecture                                Version                                                    Repository                                                                                 Size
==============================================================================================================================================================================================================================================================
Installing:
 intel-opencl                                          x86_64                                      20.44.18297-1.fc33                                         copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                      1.4 M
Installing dependencies:
 intel-gmmlib                                          x86_64                                      20.3.2-1.fc33                                              updates-testing                                                                           129 k
 intel-igc-core                                        x86_64                                      1.0.5353-1.fc33                                            copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                      142 M
 intel-igc-opencl                                      x86_64                                      1.0.5353-1.fc33                                            copr:copr.fedorainfracloud.org:jdanecki:intel-opencl                                       33 M

Transaction Summary
==============================================================================================================================================================================================================================================================
Install  4 Packages

Total download size: 177 M
Installed size: 680 M
```
Packages for Fedora 32 will be created soon.

shlomif, 2020-11-10:
On Mon, 09 Nov 2020 14:49:28 -0800
Jacek Danecki <notifications@github.com> wrote:

> Packages for:
> * Neo 20.44.18297
> * IGC 1.0.5353 
> * L0 1.0.16
> were built on copr for fedora 33.
> Neo is using gmmlib from fedora (updates-testing):
> ```
> dnf install intel-opencl
> Last metadata expiration check: 0:00:17 ago on Mon Nov  9 22:40:25 2020.
> Dependencies resolved.

Thanks, Jacek!

I'll try taking a look.

-- 

Shlomi Fish       https://www.shlomifish.org/
https://youtu.be/KxGRhd_iWuE - Never Give Up!!

Logic sucks. Morality sucks. Reality sucks. Deal with it!

Please reply to list if it's a mailing list post - https://shlom.in/reply .

shlomif, 2020-11-10:
@JacekDanecki : hi! Tested now on fedora 33 x86-64 and it seems to install and work fine. For the record, my code is here: https://github.com/shlomif/fc-solve/blob/feature-branch--find-deal-idx--opencl-port-to-c/fc-solve/source/board_gen/find-freecell-deal-index-julia-opencl.py . Thanks!

JacekDanecki, 2020-11-10:
Packages for Fedora 32 were prepared on copr, closing this issue. Feel free to open new issue, when you find problems.

closed 2020-11-10 by JacekDanecki