intel/compute-runtime #357: can EU execute two kernels in parallel

guoyejun, 2020-09-15:
Hi, 

suppose I have a GPU with 32 EUs, and each EU supports 7 threads, and SIMD8 is used, so there are 32*7*8 = 1792 work items can be executed in parallel in the GPU.

Now I have two independent kernels, kernel A and kernel B, they are submitted to GPU/ring buffer at almost the same time, the time difference is very small compared to the kernel work load (for example, kernel A needs 2 seconds and kernel B needs 3 seconds, while the time difference might be less than 1 ms).  And kernel A has 512 work items, and kernel B has 256 work items (I know the work item number is so small, but just to make my question clear, this is just for an example). Both kernels do not use work-group relative features such as barrier, __local variable.

My question is, are both kernel A and kernel B executed on the EU in parallel?  Or, the second kernel has to wait for the first kernel to finish the execution on EU (even EU is not full utilized since the EU's full capability is 1792 work items)?

btw, i don't try to understand it from user's perspective, just want to understand it from hardware/EU's perspective.

bfliflet, 2020-09-15:
The GPU is capable of executing kernels in parallel.  There are some cliffs that you can fall off degenerating into serialized execution.  It depends on the kernels and how you're invoking them,  I hope this helps!

guoyejun, 2020-09-16:
thanks.

Regarding "how you're invoking them", do you mean how I use the OpenCL API?  or how the opencl driver handles it, and so it means that it is driver implementation dependent?

Regarding "It depends on the kernels", could you give some examples on how the OpenCL kernels impact the behavior?

thanks.

bfliflet, 2020-09-16:
For example, if you're using an in-order command queue, the driver is required to ensure completion of the previous kernel before launching the next.  If you're using an out-of-order command queue but enqueue a barrier or clFinish, the driver will also issue GPU commands that prevent concurrency.  These are inherently HW agnostic behaviors.  As for HW specific issues on currently available hardware, OpenCL events and profiling prevent concurrency.  In this case, the events (with or without profiling) behave like a barrier is enqueued.  In certain situations where the driver runs out of internal heap memory, it may be forced to issue commands that may inhibit concurrency.  Memory related issues can happen randomly but generally infrequently.

The best way to achieve concurrent kernels on existing Intel GPU hardware via OpenCL is to: 1) use out of order queues without events and 2) enqueue barriers to identify the boundaries of concurrency.  Everything between the barriers in this model will execute concurrently in most cases.  Level Zero has additional modes to explicitly leverage concurrent kernels.  Future hardware and driver optimizations will provide more opportunities for concurrency with popular usage models.  Stay tuned!

Do you have a usage model that _requires_ concurrency or just looking for increased performance through concurrency?

bfliflet, 2020-09-16:
As to certain kernel specific behaviors, the use of kernel barriers or shared local memory can lock up valuable HW resources where the next kernel may need to wait until the previous completes as it needs those limited resources.

guoyejun, 2020-09-16:
thanks, i just thinking of it to increased performance through concurrency.

happened to find https://github.com/intel/compute-samples/tree/master/compute_samples/applications/commands_aggregation which describes how to achieve concurrent kernel execution submitted to multiple independent command queues.  So the feature requires some effort within graphics driver and also requires hardware's support.

bfliflet, 2020-09-16:
The compute runtime will implicitly aggregate multiple queues today in a fashion similar to what's described in the sample.  However, it does this in a just-in-time behavior looking at each enqueue to determine if it needs to stall for completion prior to issuing the next kernel.  The order of submissions from the application influences heavily.  Using the example picture at [https://github.com/intel/compute-samples/blob/master/compute_samples/applications/commands_aggregation/docs/images/aggregation-schema.png](url), if the application submits Cmd1, Cmd2, Cmd3, ... from the various queues in that order, the driver will internally create the picture on the right.  If, however, the application enqueues Cmd1, Cmd4, Cmd6, Cmd10, Cmd2, Cmd5, Cmd7, Cmd3, ... the driver will still aggregate but the performance will be lower.  

I believe the point behind the sample is to demonstrate how to do this explicitly at the application level and demonstrate the opportunities.  The rules I've described above will help delineate boundaries of concurrency (dashes in the picture on the right).  This process is effectively interleaving different task graphs of execution and performing a _breadth first traversal_.  It's an optimization problem where minimizing the # of dashed lines is the goal.  Happy scheduling!

bfliflet, 2020-09-16:
> if there are some independent kernels enqueued to the out-of-order queue, can these independent kernels (from the same queue) be executed in parallel?
The front-end of our GPUs are inherently _in-order-start_ / _out-of-order-completion_ queues.  This means that independent kernels from the same queue can be executed concurrently as long as something upstream doesn't prevent a downstream kernel from starting.  

Ex, KernelA, KernelB (dependent on A), KernelC into an out-of-order-queue.
Even though KernelC isn't dependent on anything, it won't start until the dependency between A & B in resolved.  An improvement to this might be...   KernelA, KernelC, KernelB(dependent on A)
In this case, KernelA and KernelC will be executed concurrently assuming available HW resources.  It is possible this could actually be slower despite KernelA and KernelC executing concurrently.  It depends on the longest path through the task execution graphs.

guoyejun, 2020-09-17:
thanks @bfliflet 

Below is my understanding what the low level implementation is.  Suppose we now have the following analytic result via CSR:
[image]

And the compute runtime will do:
* for task level 0:
flush cmd 1 to KMD
flush cmd 2 to KMD
flush cmd 3 to KMD

and KMD will append/insert cmd 1, cmd 2 and cmd3 to ring buffer, and trigger the GPU to work. And the GPU will read cmd1/2/3 in order and execute cmd 1/2/3 out-of-order (concurrently if possible).

* for task level 1:
wait until previous cmds (cmd1/2/3) are finished, and then
flush cmd 4 to KMD
flush cmd 5 to KMD

* for task level 2:
wait until previous cmds (cmd4/5) are finished, and then
flush cmd 6 to KMD
flush cmd 7 to KMD

* for task level 3:
wait until previous cmds (cmd6/7) are finished, and then
flush cmd 8 to KMD
flush cmd 9 to KMD
flush cmd 10 to KMD

Is my understanding right? thanks.

bfliflet, 2020-09-17:
At a high level you're correct.  To optimize things a bit more than you represent, the UMD will generate one longer sequence of commands and attempt to minimize the number of submissions to KMD.  _When_ the driver submits the sequence or chain to KMD can depend on the type of queue and whether Windows or Linux.  With out of order queues on Linux, the UMD generally attempts to submit a single, longer chain of commands.  The KMD will add a single command to the ring buffer to execute the entire chain of commands.  To be clear, the 'waits' are processed by the GPU not by the host.  Happy scheduling!

guoyejun, 2020-09-18:
thanks, and so another implementation looks like:

for task level 0:
fill command buffer (batch buffer A) with cmd1/2/3
flush batch buffer A to KMD

and KMD adds a command in ring buffer to point to batch buffer A, and triggers the GPU to work. And the GPU reads batch buffer A and executes cmd 1/2/3 out-of-order (concurrently if possible).

for task level 1:
fill command buffer (batch buffer B) with cmd4/5
flush batch buffer B to KMD

and KMD adds a command in ring buffer to point to batch buffer B, and triggers the GPU to work. And the GPU reads batch buffer B, and waits for the finish of batch buffer A, and then executes cmd 4/5 out-of-order (concurrently if possible).

my question is how the GPU knows what to 'waits'?  I think the dependency logic is maintained in user mode level, how the GPU know it, thanks.

bfliflet, 2020-09-18:
At the UMD level, dependencies are explicit in the form of enqueuing a barrier or with events.  In currently available hardware, a command is inserted to wait for all execution units (EUs) to complete.  In the future, consumer kernels will have a command inserted to wait on the completion of specific producer kernels much like how the API expresses these dependencies.

guoyejun, 2020-09-21:
btw, if CSR (or similar functionality) sits in KMD, and it puts commands from different processes together and then do gpu trigger once, will this improve the performance (because of the concurrent kernels from different processes)? thanks.

bfliflet, 2020-09-21:
It is theoretically possible to combine kernels from different applications to be run concurrently but some challenges need to be overcome.  The biggest challenges are address space isolation and state management across applications.  All it takes is for a single application to issue a draining instruction and the concurrency is prevented.  I personally have explored this as an optimization but have decided against it for the time being.

guoyejun, 2020-09-22:
really thanks!

closed 2020-09-22 by guoyejun