intel/compute-runtime #363: CPU over-loaded caused by busy wait in waitForCompletionWithTimeout

feijiang1, 2020-09-23:
Is running into some performance bottlenecks when offloading the ASR compute to the GPU via OpenCL. And tracked it down to OpenCL driver implementation. Found a lot of CPU time is being spent on sched_yield when waiting for the GPU to finish execution and to copy the buffers back for consumption.

From the call stack, it looks like the OpenCL driver spends a lot of time busy-waiting during GPU execution. On looking further into the code for this particular open sourced driver, see some more evidence of this. In command_stream_receiver.cpp in the function waitForCompletionWithTimeout, see the following code.

while (*getTagAddress() < taskCountToWait && timeDiff <= timeoutMicroseconds) {
    std::this_thread::yield();
    CpuIntrinsics::pause();
...

As we do not have other higher or similar priority threads waiting for the CPU at that time, we see this busy wait artificially increasing the CPU usage. This behavior of std::this_thread::yield is according to spec and is not a bug. We also see a similar busy wait when transferring data from the CPU to the GPU.

Tried to modify the driver and tested it out by adding a 100us sleep instead of the yield and the CPU usage improved quite a bit. For the particular use case that I was testing, the average CPU load decreased to 61% from 95% with no observable impact to the latency. For a much simpler use case with only a single thread executing computations on the GPU and waiting for the results in a loop, the CPU load decreased from 100% to 30%. There are a couple of issues with this solution though.

•	100 microseconds is quite a small sleep and the context switching penalties may be significant if the order of magnitude of time taken for GPU execution is considerably higher (few 10s of milliseconds), as the thread has to wake up and go back to sleep immediately multiple times.
•	It is not easy to increase the sleep value as this is really model dependent. Moreover the function waitForCompletionWithTimeout is also common to just data transfer, and they are considerably quicker operations.

The ideal and the most general solution would be to use mutex and signaling.

bfliflet, 2020-09-23:
> It is not easy to increase the sleep value as this is really model dependent. Moreover the function waitForCompletionWithTimeout is also common to just data transfer, and they are considerably quicker operations.

> The ideal and the most general solution would be to use mutex and signaling.

This is the crux of the problem known as '_completion latency_'.  Polling provides the best performance whereas event-driven (mutex + signaling) provides the lowest power.  Existing event-driven mechanisms, such as the one you mention, introduce high enough latency that it negatively impacts latency critical workloads. This ultimately boils down to power vs latency.  The driver attempts to combine a blend of the two to give the broadest benefit to as many workloads as possible.  We're at where we're at because of many, many hours of performance testing.  There is an obvious bias of performance over power which is likely what you're  seeing.  We'd be happy to discuss alternative 'completion' algorithms.  These heuristic approaches get more complex as they're dealing with a disparity between the producer, aka host, and the consumer, aka gpu which can vary wildly based on CPU speed, # of cores, GPU frequency, # of compute units, etc.  Behind the scenes we're driving for improvements to the system architecture to provide lower latency, event-driven mechanisms in future hardware.

Applications can tweak this behavior by tracking progress and sleeping themselves if the host gets too far beyond the GPU.  This actually works better because applications know their content whereas the driver has to guess what's coming up next.

MichalMrozek, 2020-09-23:
We have a power saving mode that is controlled with throttle hints
https://www.khronos.org/registry/OpenCL/sdk/2.2/docs/man/html/cl_khr_throttle_hints.html

If you want waits to be power effective, create command queue with following property :
CL_QUEUE_THROTTLE_LOW_KHR (optimized/lowest energy consumption)

guoyejun, 2020-09-23:
i was just about to ask a similar question about it, i found this issue in clWaitForEvents when running openvino with GPU path, see https://github.com/openvinotoolkit/openvino/issues/2239 for detail.

For me, the current implementation is good for latency, and bad for throughput. My original issue is that I do video analytic with FFmpeg->OpenVINO->cldnn->OpenCL. We cares the throughput more here (how many ffmpeg instances can run together with acceptable FPS).

Regarding "Applications can tweak this behavior by tracking progress and sleeping themselves", shall the OpenCL add a new parameter for  
cl_int clWaitForEvents(
    cl_uint num_events,
    const cl_event* event_list, **int blocking**);

And so the application can sleep itself like:
```
if (clWaitForEvents(... blocking = 0)  == NOT_COMPLETE) {
  sleep(1);
}
```

As you mentioned that "event-driven mechanisms, impacts latency critical workloads", regarding "provide lower latency, event-driven mechanisms in future hardware", just curious how it could be achieved to not impact the latency critical workloads?  I just think the time consumed for the event mechanism is usually fixed by the system (OS and hardware).

bfliflet, 2020-09-23:
[clGetEventInfo](https://www.khronos.org/registry/OpenCL/sdk/2.2/docs/man/html/clGetEventInfo.html) + CL_​EVENT_​COMMAND_​EXECUTION_​STATUS with a small wrapper is what I think you're looking for.

Event driven completion research is ongoing.

feijiang1, 2020-09-25:
Thanks a lot for your prompt respond. And has validate the fix, as setting queue properties to CL_QUEUE_THROTTLE_LOW_KHR did lower the CPU usage to levels similar to what was seen with the sleep added in the driver.
Then I will close this bug.

closed 2020-09-25 by feijiang1