intel/compute-runtime #364: intel-opencl fails to compile darktable blendop.cl

cryptomilk, 2020-10-08:
I'm using darktable [1] master [2] and it worked just fine. Then intel-opencl (intel-igc-opencl-1.0.4241-1.fc32.x86_64, intel-opencl-clang-10.0.12-1.fc32.x86_64) got and darktable started segfaulting. When doing an strace it looks like it tries to compile the blendop.cl kernel.

```
Thread 1 (Thread 0x7f6f7a2fc140 (LWP 219082)):
#0  0x00007f6f7e964f0f in wait4 () at /lib64/libc.so.6
#1  0x00007f6f7ebc9690 in _dt_sigsegv_handler () at /usr/bin/../lib64/darktable/libdarktable.so
#2  0x00007f6f7e8d3a70 in <signal handler called> () at /lib64/libc.so.6
#3  0x00007f6f7ce95ba3 in rustc_demangle::v0::Printer::print_path () at /lib64/librsvg-2.so.2
#4  0x00007ffcfd58f0e0 in  ()
#5  0x00007ffcfd58f050 in  ()
#6  0x00007ffcfd58f0e0 in  ()
#7  0x00005556f7211c70 in  ()
#8  0x00007f6f3b0134f7 in clang::ASTDeclReader::VisitFunctionDecl(clang::FunctionDecl*) () at /lib64/libclang-cpp.so.10
#9  0x00007f6f3b01a7f6 in clang::ASTDeclReader::Visit(clang::Decl*) () at /lib64/libclang-cpp.so.10
#10 0x00007f6f3b01ac2b in clang::ASTReader::ReadDeclRecord(unsigned int) () at /lib64/libclang-cpp.so.10
#11 0x00007f6f3afb7a91 in clang::ASTReader::GetDecl(unsigned int) () at /lib64/libclang-cpp.so.10
#12 0x00007f6f3afda87e in clang::ASTReader::ReadASTBlock(clang::serialization::ModuleFile&, unsigned int) () at /lib64/libclang-cpp.so.10
#13 0x00007f6f3afe4de3 in clang::ASTReader::ReadAST(llvm::StringRef, clang::serialization::ModuleKind, clang::SourceLocation, unsigned int, llvm::SmallVectorImpl<clang::ASTReader::ImportedSubmodule>*) () at /lib64/libclang-cpp.so.10
#14 0x00007f6f3b145d50 in clang::CompilerInstance::loadModuleFile(llvm::StringRef) () at /lib64/libclang-cpp.so.10
#15 0x00007f6f3b18649c in clang::FrontendAction::BeginSourceFile(clang::CompilerInstance&, clang::FrontendInputFile const&) () at /lib64/libclang-cpp.so.10
#16 0x00007f6f3b13d269 in clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) () at /lib64/libclang-cpp.so.10
#17 0x00007f6f3b1f712c in clang::ExecuteCompilerInvocation(clang::CompilerInstance*) () at /lib64/libclang-cpp.so.10
#18 0x00007f6f2c35b6b2 in Compile () at /lib64/libopencl-clang.so.10
#19 0x00007f6f64224137 in TC::CClangTranslationBlock::TranslateClang(TC::TranslateClangArgs const*, TC::STB_TranslateOutputArgs*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, char const*) () at /lib64/libigdfcl.so.1
#20 0x00007f6f64225a5e in TC::CClangTranslationBlock::Translate(TC::STB_TranslateInputArgs const*, TC::STB_TranslateOutputArgs*) () at /lib64/libigdfcl.so.1
#21 0x00007f6f64229969 in IGC::FclOclTranslationCtx<0ul>::Impl::Translate(unsigned long, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, unsigned int) () at /lib64/libigdfcl.so.1
#22 0x00007f6f64561a7a in NEO::CompilerInterface::build(NEO::Device const&, NEO::TranslationInput const&, NEO::TranslationOutput&) () at /usr/lib64/intel-opencl/libigdrcl.so
#23 0x00007f6f643a9bef in NEO::Program::build(unsigned int, _cl_device_id* const*, char const*, void (*)(_cl_program*, void*), void*, bool) () at /usr/lib64/intel-opencl/libigdrcl.so
#24 0x00007f6f6434b7c8 in clBuildProgram () at /usr/lib64/intel-opencl/libigdrcl.so
#25 0x00007f6f6c1f9472 in clBuildProgram () at /lib64/libOpenCL.so
#26 0x00007f6f7ebb007d in dt_opencl_build_program () at /usr/bin/../lib64/darktable/libdarktable.so
#27 0x00007f6f7ebb14f1 in dt_opencl_device_init () at /usr/bin/../lib64/darktable/libdarktable.so
#28 0x00007f6f7ebb5083 in dt_opencl_init () at /usr/bin/../lib64/darktable/libdarktable.so
#29 0x00007f6f7eb3f5e2 in dt_init () at /usr/bin/../lib64/darktable/libdarktable.so
#30 0x00005556f561b0cc in main ()
```

It works just fine if I disable OpenCL.

[1] https://darktable.org
[2] https://github.com/darktable-org/darktable/

alalek, 2020-10-08:
> /lib64/libopencl-clang.so.10

AFAIK, OpenCL runtime should use own Clang compiler (`libopencl-clang.so`).

Next investigation steps:
- `ls -l /usr/lib64/intel-opencl/`
- run application with `LD_DEBUG=files`

JacekDanecki, 2020-10-19:
I suppose you're using IGC packages from copr, or did you compile IGC under Fedora 32?
What Neo version are you using?
What IGC version have you installed earlier, which worked?

cryptomilk, 2020-10-19:
Yes, I'm using the copr repo:

```
krikkit:~ # rpm -q intel-igc-core intel-igc-opencl intel-opencl
intel-igc-core-1.0.4241-1.fc32.x86_64
intel-igc-opencl-1.0.4241-1.fc32.x86_64
intel-opencl-20.28.17293-1.fc32.x86_64
```

Everything worked just fine at the beginning of the year. I don't know exactly when it broke.

AdamCetnerowski, 2020-11-20:
We are debugging this, but I don't have a trend date.

cryptomilk, 2020-11-20:
I've just tested with Fedora 33 and it works there.

ArchangeGabriel, 2020-11-20:
Indeed, there were dozen of similar reports on the Darktable bugtracker, but it has been solved for everyone now that distros are building the stack cleanly against LLVM 11.

JacekDanecki, 2020-11-30:
Neo packages on copr were updated to 20.47.18513 for Fedora 32-34. IGC packages were compiled with llvm 10.
I've tested darktable with Neo 20.47.18513 on Fedora 32, and it worked correctly.
Closing the issue, as it's working now. Feel free to create new issue when you find any problems.

closed 2020-11-30 by JacekDanecki