intel/compute-runtime #368: how to dump GPU ASM for the kernels

guoyejun, 2020-10-22:
Hi,

we can dump the kernel source code by setting NEOReadDebugKeys and DumpKernels, according to https://github.com/intel/compute-runtime/blob/master/shared/source/debug_settings/debug_variables_base.inl, I verified it with cmake option CMAKE_BUILD_TYPE=Debug, but I could not find a method to dump the hardware asm for that kernel, is there a way to dump it? thanks.

actually, i want to check how much of my workload (based on cnn model, the OpenCL kernels are written by another middle-ware, i do not touch the kernel source code) is optimized with MAD instructions, so I can predict the best possible FPS of my workload by comparing to the hardware's FLOPs/cycle (mad peak).  If mad is used mostly for the kernel, i'll use the peak FLOPs directly, if not, i might have to divide peak FLOPs by 2?

thanks

alalek, 2020-10-22:
You may want to check this thread: #75

guoyejun, 2020-10-22:
thanks, will try it.

just curious, why we don't integrate it within the driver, just like DumpKernelsISABin and DumpKernelsISAText, it will be much much more convenient, thanks.

guoyejun, 2020-11-03:
tried successfully with that tool, thanks.

It would be pretty much convenient and nature to support within this driver ...

closed 2020-11-03 by guoyejun