intel/compute-runtime #369: make fails unless ULT is skipped

js333031, 2020-10-24:
I am getting the linker error below unless I use `cmake -D SKIP_ALL_ULT=ON ../`

I'm trying to compile branch 19.41.14441 for use with openvino 2021.1 on Arch Linux.

How can I resolve this issue? 

```
[ 89%] Linking CXX executable ../bin/igdrcl_tests
/usr/bin/ld: CMakeFiles/igdrcl_tests.dir/compiler_interface/compiler_interface_tests.cpp.o:(.data.rel.ro._ZTV21MockCompilerDeviceCtxIN3NEO19MockFclOclDeviceCtxENS0_24MockFclOclTranslationCtxEE[_ZTV21MockCompilerDeviceCtxIN3NEO19MockFclOclDeviceCtxENS0_24MockFclOclTranslationCtxEE]+0x80): undefined reference to `IGC::FclOclDeviceCtx<3ul>::CreateTranslationCtxImpl(unsigned long, unsigned long, unsigned long, CIF::Builtins::Buffer<1ul>*)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o): in function `IGC::GTSystemInfo<3ul>::~GTSystemInfo()':
mock_compilers.cpp:(.text._ZN3IGC12GTSystemInfoILm3EED2Ev[_ZN3IGC12GTSystemInfoILm3EED5Ev]+0xf): undefined reference to `vtable for IGC::GTSystemInfo<3ul>'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o): in function `IGC::FclOclDeviceCtx<3ul>::~FclOclDeviceCtx()':
mock_compilers.cpp:(.text._ZN3IGC15FclOclDeviceCtxILm3EED2Ev[_ZN3IGC15FclOclDeviceCtxILm3EED5Ev]+0xf): undefined reference to `vtable for IGC::FclOclDeviceCtx<3ul>'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o): in function `IGC::GTSystemInfo<3ul>::GTSystemInfo()':
mock_compilers.cpp:(.text._ZN3IGC12GTSystemInfoILm3EEC2Ev[_ZN3IGC12GTSystemInfoILm3EEC5Ev]+0x1b): undefined reference to `vtable for IGC::GTSystemInfo<3ul>'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o): in function `IGC::FclOclDeviceCtx<3ul>::FclOclDeviceCtx()':
mock_compilers.cpp:(.text._ZN3IGC15FclOclDeviceCtxILm3EEC2Ev[_ZN3IGC15FclOclDeviceCtxILm3EEC5Ev]+0x1b): undefined reference to `vtable for IGC::FclOclDeviceCtx<3ul>'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO19MockFclOclDeviceCtxE[_ZTVN3NEO19MockFclOclDeviceCtxE]+0x80): undefined reference to `IGC::FclOclDeviceCtx<3ul>::CreateTranslationCtxImpl(unsigned long, unsigned long, unsigned long, CIF::Builtins::Buffer<1ul>*)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO7MockCIFIN3IGC15FclOclDeviceCtxILm3EEEEE[_ZTVN3NEO7MockCIFIN3IGC15FclOclDeviceCtxILm3EEEEE]+0x80): undefined reference to `IGC::FclOclDeviceCtx<3ul>::CreateTranslationCtxImpl(unsigned long, unsigned long, unsigned long, CIF::Builtins::Buffer<1ul>*)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO16MockGTSystemInfoE[_ZTVN3NEO16MockGTSystemInfoE]+0x1c8): undefined reference to `IGC::GTSystemInfo<3ul>::GetMaxDualSubSlicesSupported() const'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO16MockGTSystemInfoE[_ZTVN3NEO16MockGTSystemInfoE]+0x1d0): undefined reference to `IGC::GTSystemInfo<3ul>::SetMaxDualSubSlicesSupported(unsigned int)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO16MockGTSystemInfoE[_ZTVN3NEO16MockGTSystemInfoE]+0x1d8): undefined reference to `IGC::GTSystemInfo<3ul>::GetDualSubSliceCount() const'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO16MockGTSystemInfoE[_ZTVN3NEO16MockGTSystemInfoE]+0x1e0): undefined reference to `IGC::GTSystemInfo<3ul>::SetDualSubSliceCount(unsigned int)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE[_ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE]+0x1c8): undefined reference to `IGC::GTSystemInfo<3ul>::GetMaxDualSubSlicesSupported() const'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE[_ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE]+0x1d0): undefined reference to `IGC::GTSystemInfo<3ul>::SetMaxDualSubSlicesSupported(unsigned int)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE[_ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE]+0x1d8): undefined reference to `IGC::GTSystemInfo<3ul>::GetDualSubSliceCount() const'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE[_ZTVN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE]+0x1e0): undefined reference to `IGC::GTSystemInfo<3ul>::SetDualSubSliceCount(unsigned int)'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTIN3NEO7MockCIFIN3IGC15FclOclDeviceCtxILm3EEEEE[_ZTIN3NEO7MockCIFIN3IGC15FclOclDeviceCtxILm3EEEEE]+0x10): undefined reference to `typeinfo for IGC::FclOclDeviceCtx<3ul>'
/usr/bin/ld: ../lib/libigdrcl_mocks.a(mock_compilers.cpp.o):(.data.rel.ro._ZTIN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE[_ZTIN3NEO7MockCIFIN3IGC12GTSystemInfoILm3EEEEE]+0x10): undefined reference to `typeinfo for IGC::GTSystemInfo<3ul>'
collect2: error: ld returned 1 exit status
make[2]: *** [unit_tests/CMakeFiles/igdrcl_tests.dir/build.make:7541: bin/igdrcl_tests] Error 1
make[1]: *** [CMakeFiles/Makefile2:7073: unit_tests/CMakeFiles/igdrcl_tests.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
[user@archlinux build]$
```

JablonskiMateusz, 2020-10-27:
`SKIP_ALL_ULT` was removed some time ago, please retry with `SKIP_UNIT_TESTS`.

js333031, 2020-10-29:
Using SKIP_UNIT_TESTS (SKIP_ALL_ULT) would avoid the linker error. Intention of the question was if this linker error should be encountered.

JablonskiMateusz, 2020-10-29:
Sorry, I misread the major version, I thought about 20 not 19.

For that version SKIP_ALL_ULT was correct flag but AFAIR it disabled ULT runners, the ULT-related targets like _igdrcl_tests_ were being built.

Based on the log, it looks like you use more recent IGC headers than used for release [19.41.14441](https://github.com/intel/compute-runtime/releases/tag/19.41.14441). As _mock_compilers_ needs to override the entire IGC interface, in order to resolve the issue you need to downgrade your IGC version or add mock implementations of your current IGC interface.

js333031, 2020-10-29:
What version of IGC headers should be used?

JablonskiMateusz, 2020-10-29:
This release was built with https://github.com/intel/intel-graphics-compiler/commit/b3ed5e584dac45d560e6f68e5f6e84af8a661e6e

JablonskiMateusz, 2020-11-23:
@js333031 could you confirm that it worked for you?

AdamCetnerowski, 2020-12-03:
Please re-open if issue was not resolved

closed 2020-12-03 by AdamCetnerowski