intel/compute-runtime #371: Build failures with -Werror (int-in-bool-context) in unit_tests

ArchangeGabriel, 2020-10-29:
The latest release FTBFS in unit_tests:
```
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp: In instantiation of ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matchBody() [with PRODUCT_FAMILY productFamily = IGFX_ROCKETLAKE; GFXCORE_FAMILY gfxCoreFamily = IGFX_GEN12LP_CORE; FamilyType = NEO::TGLLPFamily]’:
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matched() [with PRODUCT_FAMILY productFamily = IGFX_ROCKETLAKE]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 10]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from here
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:195:102: error: enum constant in boolean context [-Werror=int-in-bool-context]
  195 |         if ((cmd->getPostSyncOperation(), MI_FLUSH_DW::POST_SYNC_OPERATION_WRITE_TIMESTAMP_REGISTER) &&
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
  196 |             (cmd->getDestinationAddress(), timestampAddress)) {
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                          
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp: In instantiation of ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matchBody() [with PRODUCT_FAMILY productFamily = IGFX_TIGERLAKE_LP; GFXCORE_FAMILY gfxCoreFamily = IGFX_GEN12LP_CORE; FamilyType = NEO::TGLLPFamily]’:
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matched() [with PRODUCT_FAMILY productFamily = IGFX_TIGERLAKE_LP]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 9]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 10]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from here
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:195:102: error: enum constant in boolean context [-Werror=int-in-bool-context]
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp: In instantiation of ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matchBody() [with PRODUCT_FAMILY productFamily = IGFX_JASPERLAKE; GFXCORE_FAMILY gfxCoreFamily = IGFX_GEN11_CORE; FamilyType = NEO::ICLFamily]’:
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matched() [with PRODUCT_FAMILY productFamily = IGFX_JASPERLAKE]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   recursively required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 9]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 10]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from here
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:195:102: error: enum constant in boolean context [-Werror=int-in-bool-context]
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp: In instantiation of ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matchBody() [with PRODUCT_FAMILY productFamily = IGFX_LAKEFIELD; GFXCORE_FAMILY gfxCoreFamily = IGFX_GEN11_CORE; FamilyType = NEO::ICLFamily]’:
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matched() [with PRODUCT_FAMILY productFamily = IGFX_LAKEFIELD]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   recursively required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 9]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 10]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from here
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:195:102: error: enum constant in boolean context [-Werror=int-in-bool-context]
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp: In instantiation of ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matchBody() [with PRODUCT_FAMILY productFamily = IGFX_ICELAKE_LP; GFXCORE_FAMILY gfxCoreFamily = IGFX_GEN11_CORE; FamilyType = NEO::ICLFamily]’:
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from ‘void L0::ult::CommandListCreate_givenCopyOnlyCommandListWhenAppendWriteGlobalTimestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::matched() [with PRODUCT_FAMILY productFamily = IGFX_ICELAKE_LP]’
[... cut ...] mestampCalledThenMiFlushDWWithTimestampEncoded_Platforms_Test::checkForMatch(PRODUCT_FAMILY) [with unsigned int matcherOrdinal = 10]’
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:169:1:   required from here
/build/intel-compute-runtime/src/compute-runtime-20.42.18209/level_zero/core/test/unit_tests/sources/cmdlist/test_cmdlist_2.cpp:195:102: error: enum constant in boolean context [-Werror=int-in-bool-context]
```

JacekDanecki, 2020-11-03:
It should be fixed with commit: https://github.com/intel/compute-runtime/commit/f9e0c04b542ba6fd41952df7282a6984cf75d77a

ArchangeGabriel, 2020-11-14:
I can confirm, thanks!

closed 2020-11-14 by ArchangeGabriel