intel/compute-runtime #373: SIGSEGV on copr version in Fedora 32

jhidding, 2020-11-03:
Running anything OpenCL related currently results in a segmentation fault, including `clinfo`.

```
$ rpm -q intel-igc-core intel-opencl clinfo
intel-igc-core-1.0.4241-1.fc32.x86_64
intel-opencl-20.28.17293-1.fc32.x86_64
clinfo-2.2.18.04.06-4.fc32.x86_64
```

The error seems located in `clang-libs`.

```
$ gdb clinfo
... run
...
Thread 1 "clinfo" received signal SIGSEGV, Segmentation fault.
0x00007ffff63399bf in clang::serialization::BasicReaderBase<clang::ASTRecordReader>::readDeclarationName() () from /lib64/libclang-cpp.so.10
```

JablonskiMateusz, 2020-11-03:
Could you provide file log.txt created by command:
`strace -f -s 255 -o log.txt clinfo`
?

JacekDanecki, 2020-11-03:
I've just installed intel-opencl packages on updated Fedora 32, and clinfo works, can you compare installed packages on your setup with below packages
```
dnf install intel-opencl
Last metadata expiration check: 0:00:34 ago on Tue Nov  3 18:39:43 2020.
Dependencies resolved.
==============================================================================================================================
 Package                 Arch        Version                  Repository                                                 Size
==============================================================================================================================
Installing:
 intel-opencl            x86_64      20.28.17293-1.fc32       copr:copr.fedorainfracloud.org:jdanecki:intel-opencl      1.2 M
Installing dependencies:
 intel-gmmlib            x86_64      20.2.2-1.fc32            updates                                                   138 k
 intel-igc-core          x86_64      1.0.4241-1.fc32          copr:copr.fedorainfracloud.org:jdanecki:intel-opencl      6.6 M
 intel-igc-opencl        x86_64      1.0.4241-1.fc32          copr:copr.fedorainfracloud.org:jdanecki:intel-opencl       73 k
 intel-opencl-clang      x86_64      10.0.12-1.fc32           copr:copr.fedorainfracloud.org:jdanecki:intel-opencl      9.3 M

Transaction Summary
==============================================================================================================================
Install  5 Packages

Total download size: 17 M
Installed size: 56 M

$ rpm -q clinfo
clinfo-2.2.18.04.06-4.fc32.x86_64

$ lsb_release -a
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: Fedora
Description:    Fedora release 32 (Thirty Two)
Release:        32
Codename:       ThirtyTwo

$ clinfo -l
Platform #0: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) Gen9 HD Graphics NEO

$ rpm -qf /lib64/libclang-cpp.so.10
clang-libs-10.0.1-3.fc32.x86_64

$ rpm -qa | grep -E "llvm|clang"
clang-10.0.1-3.fc32.x86_64
clang-libs-10.0.1-3.fc32.x86_64
clang-tools-extra-10.0.1-3.fc32.x86_64
llvm-libs-10.0.1-4.fc32.x86_64
spirv-llvm-translator-10.0.14-1.fc32.x86_64
llvm-10.0.1-4.fc32.x86_64
intel-opencl-clang-10.0.12-1.fc32.x86_64
```

jhidding, 2020-11-03:
```
$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 32 (Thirty Two)
Release:	32
Codename:	ThirtyTwo
```

```
$ rpm -qa | grep -E "llvm|clang"
clang-libs-10.0.1-3.fc32.x86_64
clang-tools-extra-10.0.1-3.fc32.x86_64
llvm-libs-10.0.1-4.fc32.i686
clang-10.0.1-3.fc32.x86_64
llvm-libs-10.0.1-4.fc32.x86_64
intel-opencl-clang-10.0.12-1.fc32.x86_64
spirv-llvm-translator-10.0.12-1.fc32.x86_64
```

```
$ clinfo -l
Platform #0: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) Gen9 HD Graphics NEO
Platform #1: Clover
```

Full output of `clinfo`:

```
$ clinfo
Number of platforms                               2
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.1 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_fp64 cl_khr_subgroups cl_khr_il_program cl_intel_spirv_device_side_avc_motion_estimation cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory_preview cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_device_side_avc_motion_estimation cl_intel_advanced_motion_estimation cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_intel_media_block_io cl_khr_3d_image_writes cl_intel_va_api_media_sharing 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             INTEL

  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 20.1.10
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO 
  Driver Version                                  20.28.17293
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               24
  Max clock frequency                             1150MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
fish: 'clinfo' terminated by signal SIGSEGV (Address boundary error)
```

jhidding, 2020-11-04:
Output of `strace -f -s 255 -o log.txt clinfo` attached:
[log.txt](https://github.com/intel/compute-runtime/files/5486259/log.txt)

JacekDanecki, 2020-11-04:
Can you remove mesa-libOpenCL package and run clinfo again? We've observed incompatibility between mesa opencl and IGC on Debian based distributions.
I've reproduced this issue on my Fedora 32 system after mesa-libOpenCL installation. On Debian based distributions as a workaround we compiled IGC with llvm/clang sources instead of using system binaries. On Copr for Fedora 32 packages were compiled with system llvm/clang so far, but it'll be changed with newer IGC.

jhidding, 2020-11-04:
I've tried uninstalling the mesa-libOpenCL package but nothing changes; maybe rebooting would help? To clarify, I don't just get the segfault running `clinfo`, also this simple `pyopencl` example crashes with the same segfault.

```python
import numpy as np
import pyopencl as cl

a_np = np.random.rand(50000).astype(np.float32)
b_np = np.random.rand(50000).astype(np.float32)

ctx = cl.create_some_context()
queue = cl.CommandQueue(ctx)

mf = cl.mem_flags
a_g = cl.Buffer(ctx, mf.READ_ONLY | mf.COPY_HOST_PTR, hostbuf=a_np)
b_g = cl.Buffer(ctx, mf.READ_ONLY | mf.COPY_HOST_PTR, hostbuf=b_np)

prg = cl.Program(ctx, """
__kernel void sum(
    __global const float *a_g, __global const float *b_g, __global float *res_g)
{
  int gid = get_global_id(0);
  res_g[gid] = a_g[gid] + b_g[gid];
}
""").build()

res_g = cl.Buffer(ctx, mf.WRITE_ONLY, a_np.nbytes)
prg.sum(queue, a_np.shape, None, a_g, b_g, res_g)

res_np = np.empty_like(a_np)
cl.enqueue_copy(queue, res_np, res_g)

# Check on CPU with Numpy:
print(res_np - (a_np + b_np))
print(np.linalg.norm(res_np - (a_np + b_np)))
assert np.allclose(res_np, a_np + b_np)
```

JacekDanecki, 2020-11-04:
When you downgrade clang-libs
```
dnf install clang-libs-10.0.0-1.fc32
```
clinfo and your test are  working again.

jhidding, 2020-11-04:
Indeed! Thank you.

JacekDanecki, 2020-11-04:
Closing as fix was provided.

closed 2020-11-04 by JacekDanecki