intel/compute-runtime #375: Intel GPU not listed in clinfo

mimi89999, 2020-11-09:
Hello,
I have an `i5-7400` and I would like to use OpenCL on it.
On that machine I have Ubuntu 20.10 with the `intel-opencl-icd` package installed. The problem is that `clinfo` only lists the Mesa and POCL platforms:
```
$ sudo clinfo
Number of platforms                               2
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 20.2.1
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Portable Computing Language
  Platform Vendor                                 The pocl project
  Platform Version                                OpenCL 1.2 pocl 1.5, None+Asserts, LLVM 9.0.1, RELOC, SLEEF, DISTRO, POCL_DEBUG
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             POCL

  Platform Name                                   Clover
Number of devices                                 2
  Device Name                                     Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-20-generic, LLVM 11.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 20.2.1
  Driver Version                                  20.2.1
  Device OpenCL C Version                         OpenCL C 1.1 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Max compute units                               36
  Max clock frequency                             1366MHz
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              64
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 2 / 2       
    half                                                 0 / 0        (n/a)
    float                                                4 / 4       
    double                                               2 / 2        (cl_khr_fp64)
  Half-precision Floating-point support           (n/a)
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Address bits                                    64, Little-Endian
  Global memory size                              8589934592 (8GiB)
  Error Correction support                        No
  Max memory allocation                           6871947673 (6.4GiB)
  Unified memory for Host and Device              No
  Minimum alignment for any data type             128 bytes
[... 167 lines cut ...]
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Clover
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [MESA]
  clCreateContext(NULL, ...) [default]            Success [MESA]
  clCreateContext(NULL, ...) [other]              Success [POCL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Clover
    Device Name                                   Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-20-generic, LLVM 11.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (2)
    Platform Name                                 Clover
    Device Name                                   Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-20-generic, LLVM 11.0.0)
    Device Name                                   Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-20-generic, LLVM 11.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (2)
    Platform Name                                 Clover
    Device Name                                   Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-20-generic, LLVM 11.0.0)
    Device Name                                   Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-20-generic, LLVM 11.0.0)

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2
```

JacekDanecki, 2020-11-09:
Can you provide output from commands:
```
lspci -vnn
cat /proc/cmdline
dmesg | grep -E "drm|i915"
ls -lR /dev/dri/
```
and file log.txt created by command:
```
strace -f -s 255 -o log.txt clinfo
```

mimi89999, 2020-11-09:
```
$ lspci -vnn
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [8086:591f] (rev 05)
	Subsystem: Gigabyte Technology Co., Ltd Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [1458:5000]
	Flags: bus master, fast devsel, latency 0
	Capabilities: <access denied>
	Kernel driver in use: skl_uncore

00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 05) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 121
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 0000e000-0000efff [size=4K]
	Memory behind bridge: efe00000-efefffff [size=1M]
	Prefetchable memory behind bridge: 00000000d0000000-00000000e01fffff [size=258M]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:14.0 USB controller [0c03]: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller [8086:a12f] (rev 31) (prog-if 30 [XHCI])
	Subsystem: Gigabyte Technology Co., Ltd 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller [1458:5007]
	Flags: bus master, medium devsel, latency 0, IRQ 129
	Memory at eff30000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci

00:16.0 Communication controller [0780]: Intel Corporation 100 Series/C230 Series Chipset Family MEI Controller #1 [8086:a13a] (rev 31)
	Subsystem: Gigabyte Technology Co., Ltd 100 Series/C230 Series Chipset Family MEI Controller [1458:1c3a]
	Flags: bus master, fast devsel, latency 0, IRQ 130
	Memory at eff4d000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: mei_me
	Kernel modules: mei_me

00:17.0 SATA controller [0106]: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode] [8086:a102] (rev 31) (prog-if 01 [AHCI 1.0])
	Subsystem: Gigabyte Technology Co., Ltd Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode] [1458:b005]
	Flags: bus master, 66MHz, medium devsel, latency 0, IRQ 128
	Memory at eff48000 (32-bit, non-prefetchable) [size=8K]
	Memory at eff4c000 (32-bit, non-prefetchable) [size=256]
	I/O ports at f050 [size=8]
	I/O ports at f040 [size=4]
	I/O ports at f020 [size=32]
	Memory at eff4b000 (32-bit, non-prefetchable) [size=2K]
	Capabilities: <access denied>
	Kernel driver in use: ahci
	Kernel modules: ahci

00:1b.0 PCI bridge [0604]: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #19 [8086:a169] (rev f1) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 122
	Bus: primary=00, secondary=02, subordinate=03, sec-latency=32
	I/O behind bridge: [disabled]
	Memory behind bridge: [disabled]
	Prefetchable memory behind bridge: [disabled]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:1b.3 PCI bridge [0604]: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #20 [8086:a16a] (rev f1) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 123
	Bus: primary=00, secondary=04, subordinate=04, sec-latency=0
	I/O behind bridge: [disabled]
	Memory behind bridge: [disabled]
	Prefetchable memory behind bridge: [disabled]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:1c.0 PCI bridge [0604]: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #3 [8086:a112] (rev f1) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 124
	Bus: primary=00, secondary=05, subordinate=05, sec-latency=0
	I/O behind bridge: [disabled]
	Memory behind bridge: [disabled]
	Prefetchable memory behind bridge: [disabled]
[... 225 lines cut ...]
[    8.585232] kernel: [drm] vram apper at 0xB0000000
[    8.585232] kernel: [drm] size 14745600
[    8.585233] kernel: [drm] fb depth is 24
[    8.585233] kernel: [drm]    pitch is 10240
[    8.585300] kernel: amdgpu 0000:06:00.0: fb1: amdgpudrmfb frame buffer device
[    8.589242] kernel: [drm] Initialized amdgpu 3.38.0 20150101 for 0000:06:00.0 on minor 1
```
```
$ ls -lR /dev/dri/
/dev/dri/:
razem 0
drwxr-xr-x  2 root root        120 paź  7 17:01 by-path
crw-rw----+ 1 root video  226,   0 paź 27 15:26 card0
crw-rw----+ 1 root video  226,   1 lis  9 12:38 card1
crw-rw----+ 1 root render 226, 128 paź 12 23:42 renderD128
crw-rw----+ 1 root render 226, 129 paź 12 23:42 renderD129

/dev/dri/by-path:
razem 0
lrwxrwxrwx 1 root root  8 paź 27 15:26 pci-0000:01:00.0-card -> ../card0
lrwxrwxrwx 1 root root 13 paź 12 23:42 pci-0000:01:00.0-render -> ../renderD128
lrwxrwxrwx 1 root root  8 lis  9 12:38 pci-0000:06:00.0-card -> ../card1
lrwxrwxrwx 1 root root 13 paź 12 23:42 pci-0000:06:00.0-render -> ../renderD129
```
[log.txt](https://github.com/intel/compute-runtime/files/5510155/log.txt)

JacekDanecki, 2020-11-09:
I don't see Intel GPU on PCI, have you disabled it in BIOS/EFI? Only AMD GPU devices are listed on PCI.

mimi89999, 2020-11-09:
I enabled the GPU in UEFI and the device is now visible. Sorry.

closed 2020-11-09 by mimi89999