intel/compute-runtime #376: Unable to use address sanitizer in a program that links against compute-runtime

Bidski, 2020-11-09:
I am running an application that links against the compute-runtime that I am trying to run an address sanitizer on. At runtime I get the following error

```
You are trying to dlopen a libigfxdbgxchg64.so shared library with RTLD_DEEPBIND flag which is incompatibe with sanitizer runtime (see https://github.com/google/sanitizers/issues/611 for details). If you want to run libigfxdbgxchg64.so library under sanitizers please remove RTLD_DEEPBIND from dlopen flags.  
```

I noticed in https://github.com/intel/compute-runtime/blob/master/shared/source/os_interface/linux/os_library_linux.cpp#L37-L42 that you are checking for a `SANITIZER_BUILD` compile definition and disabling `RTLD_DEEPBIND` if it is defined. However, this only works if the compute-runtime was compiled with this flag defined. Is it possible to move this check into a header file so that when people wish to use address sanitizers in their applications they don't need to recompile the compute-runtime as well?

Bidski, 2020-11-16:
With further investigation I have found that this is still broken in a strange way.

When running cmake with `-DUSE_ASAN=ON` to build the compute runtime **_any_** program that either links against or loads (via `libOpenCL.so`) the compute runtime **_must_** either link against `libasan.so` itself or **_must_** have `libasan.so` preloaded for it (using `LD_PRELOAD` environment variable). This includes programs like `clinfo` which will fail to detect the Intel GPU platform if `libasan.so` is not preloaded for it.

This behaviour is different from the behaviour of every other library that I am working with. With all other libraries I am able to enable address sanitization in my own program without needing to recompile any of the other libraries.

Is there some way to either remove `RTLD_DEEPBIND` from the compute runtime? The only options I can immediately think of that would allow `RTLD_DEEPBIND` to be used in the compute runtime library is either an environment variable or a configuration file.

HoppeMateusz, 2021-02-25:
Hello,

Issue with DEEPBIND should be resolved by: 66f4c75aeaf9d62f63360a588ca9c852ef465f2b  

In order to disable passing DEEPBIND flag to dlopen() in NEO, you need to set two environment variables:
NEOReadDebugKeys=1 DisableDeepBind=1

Please let us know if such configuration resolves the issue.

Mateusz

closed 2021-05-06 by AdamCetnerowski

kallaballa, 2023-11-11:
It works,. I can run now tsan (and other sanitizers!).

HoppeMateusz, 2025-03-13:
> `export NEOReadDebugKeys=1` `export DisableDeepBind=1` `dpcpp host_vs_device_code.cpp -g -fsanitize=thread` `./a.out` `==6411==You are trying to dlopen a libur_adapter_level_zero.so.0 shared library with RTLD_DEEPBIND flag which is incompatible with sanitizer runtime (see https://github.com/google/sanitizers/issues/611 for details). If you want to run libur_adapter_level_zero.so.0 library under sanitizers please remove RTLD_DEEPBIND from dlopen flags.`

Are you using recent Level zero loader ? 

this problem could be related to old loader version -  there was change in loader to remove deepbind:
https://github.com/oneapi-src/level-zero/commit/f2a5dcc7806b89ac992bef07fb17e1a8ac102612

MajidAbdelilah, 2025-03-14:
The issue has been resolved. Thanks for your help