intel/compute-runtime #378: Segfault in IGC (LLVM)

maleadt, 2020-11-19:
Using compute-runtime v20.44.18297, build against gmmlib v20.3.2 and igc v1.0.5353, I'm running into a segfault trying to compile SPIRV code produced by Khronos' SPIRV to LLVM translator (from the LLVM 11 branch):

```llvm
target triple = "spir64-unknown-unknown"

define internal fastcc { float, [1 x float] } @child() {
top:
  %0 = insertvalue { float, [1 x float] } undef, float undef, 1, 0
  ret { float, [1 x float] } %0
}

define spir_kernel void @kernel() {
topi:
  %0 = call fastcc { float, [1 x float] } @child()
  ret void
}
```

```
llvm-as main.ll
llvm-spirv -o main.spv main.bc
ocloc -device kbl -spirv_input -file main.spv
```

```
Compilation from IR - skipping loading of FCL
[0]: libocloc.so(_ZN16SafetyGuardLinux9sigActionEiP7siginfoPv+0x30) [0x7fac28052650]
[1]: /usr/lib/libpthread.so.0(+0x140f0) [0x7fac27f860f0]
[2]: libigc.so.1(_ZNK4llvm5Value10getContextEv+0) [0x7fac26067850]
[3]: libigc.so.1(_ZN4llvm9StoreInstC1EPNS_5ValueES2_bNS_10MaybeAlignENS_14AtomicOrderingEhPNS_11InstructionE+0x35) [0x7fac25ff2725]
[4]: libigc.so.1(_ZN4llvm9StoreInstC2EPNS_5ValueES2_bNS_10MaybeAlignEPNS_11InstructionE+0x13) [0x7fac25ff27e3]
[5]: libigc.so.1(+0x4ec092) [0x7fac24bbc092]
[6]: libigc.so.1(+0x4ebffa) [0x7fac24bbbffa]
[7]: libigc.so.1(+0x4ec285) [0x7fac24bbc285]
[8]: libigc.so.1(+0x4ec3b6) [0x7fac24bbc3b6]
[9]: libigc.so.1(_ZN4llvm13FPPassManager13runOnFunctionERNS_8FunctionE+0x28e) [0x7fac26012d9e]
[10]: libigc.so.1(_ZN4llvm13FPPassManager11runOnModuleERNS_6ModuleE+0x31) [0x7fac260135e1]
[11]: libigc.so.1(_ZN4llvm6legacy15PassManagerImpl3runERNS_6ModuleE+0x396) [0x7fac260139c6]
[12]: libigc.so.1(+0x57971f) [0x7fac24c4971f]
[13]: libigc.so.1(+0x4d5ee3) [0x7fac24ba5ee3]
[14]: libigc.so.1(+0x4da6e1) [0x7fac24baa6e1]
[15]: libigc.so.1(+0x586201) [0x7fac24c56201]
[16]: libigc.so.1(+0x588693) [0x7fac24c58693]
[17]: libocloc.so(_ZN3NEO15OfflineCompiler15buildSourceCodeEv+0x146) [0x7fac2802a8a6]
[18]: libocloc.so(_ZN3NEO15OfflineCompiler5buildEv+0x45) [0x7fac2802ebb5]
[19]: libocloc.so(_ZN16SafetyGuardLinux4callIiN3NEO15OfflineCompilerEMS2_FivEEET_PT0_T1_S5_+0x49) [0x7fac28052709]
[20]: libocloc.so(_Z20buildWithSafetyGuardPN3NEO15OfflineCompilerE+0x7b) [0x7fac280525eb]
[21]: libocloc.so(oclocInvoke+0xe28) [0x7fac2801ad68]
[22]: loc(main+0x23) [0x401073]
[23]: /usr/lib/libc.so.6(__libc_start_main+0xf2) [0x7fac27a91152]
[24]: loc() [0x4010a5]
```

Interestingly, the segfault doesn't occur if I disassemble and reassemble the SPIRV file using the SPIRV Tools, so something looks off with the produced file. Attaching the original file (which makes `ocloc` crash) here for that reason: [main.spv.zip](https://github.com/intel/compute-runtime/files/5565918/main.spv.zip)

pszymich, 2020-12-02:
Hello Tim,

In IGC (compiler used by the runtime) we currently don't support neither compiling LLVM 11 IR nor building/linking IGC with LLVM 11. LLVM 11 upgrade is still work-in-progress, albeit in an advanced stage.
For more information and LLVM compatibility updates please visit our [build_ubuntu.md](https://github.com/intel/intel-graphics-compiler/blob/master/documentation/build_ubuntu.md#llvmclang-version-information) documentation file in [IGC repository](https://github.com/intel/intel-graphics-compiler).

If you wish you may try to build IGC with LLVM 11 (instructions in the file linked above) but we can't assure conformance/performance on such a build.

I've tried compiling the shader you posted with IGC linked to LLVM 11 and it compiled fully with no errors,
Although the final LLVM IR kernel is empty, your kernels are mostly dead-code, so it's to be expected:

```llvm
; Function Attrs: alwaysinline convergent
define spir_kernel void @kernel(<8 x i32> %r0, <8 x i32> %payloadHeader) #0 {
topi:
    ret void
}
```

If you have any following/similar questions regarding our compiler, please visit [IGC repository](https://github.com/intel/intel-graphics-compiler) and open an issue there.

maleadt, 2020-12-02:
I'm not linking IGC to LLVM 11, I'm using the recommended hashes as listed on the GitHub release. I only mentioned I'm using the LLVM to SPIR-V translator linked against LLVM 11, because my front-end uses LLVM 11. The rest of the Intel toolchain uses LLVM 10.

JacekDanecki, 2021-01-08:
As this issue looks like problem on IGC side, I'm transferring it to IGC repository: https://github.com/intel/intel-graphics-compiler