intel/compute-runtime #379: OpenCL image gives different result than buffer

boxerab, 2020-11-20:
I have a test program here:

https://github.com/GrokImageCompression/latke

that performs the exact same image processing algorithm (debayer) using either image or buffer.

Using latest version of Neo driver,  with buffer, I get a correct result (3 channel colour image).
With opencl image, I get the wrong result (mono image: all three channels are equal).

Here is the ocl buffer result:

[image]

and here is the ocl image result:

[image]

AdamCetnerowski, 2020-11-26:
Please provide build / release driver version where it worked and where it is broken

boxerab, 2020-11-26:
Thanks, how can I query the driver version?

AdamCetnerowski, 2020-11-26:
Are you using the posted weekly releases? That number is good enough for us.

boxerab, 2020-11-29:
Yes, I am using the weekly release. I don't remember the exact version when it was working.

By the way, I just tried the exact same code with my RX 480 card, and I get the correct result.

AdamCetnerowski, 2021-01-04:
Unfortunately we can't find a working build, so this isn't a regression. This will go into our regular debug queue. We will update the issue if we have more information or a fix.

boxerab, 2021-01-04:
@AdamCetnerowski thanks - to clarify, you can't find a working build of the latke project ? I can help , with more details.

AdamCetnerowski, 2021-01-04:
No, we got the workload to work. We can't find a driver version where the results would be correct.

boxerab, 2021-01-04:
ahhh I see. So, you can repro, great.