intel/compute-runtime #384: Warning message when running clinfo.

CaoZhongZ, 2020-11-23:
I've noticed when I type clinfo there was a warning say:

	NOTE:	your OpenCL library only supports OpenCL 2.2,
		but some installed platforms support OpenCL 3.0.
		Programs using 3.0 features may crash
		or behave unexpectedly

Should it be expected?

JacekDanecki, 2020-11-26:
You need to update ocl-icd package to release 2.2.13, see: https://github.com/OCL-dev/ocl-icd/releases/tag/v2.2.13

ArchangeGabriel, 2020-11-26:
I think their ocl-icd is up-to-date. The issue is rather that distro (like us, Arch Linux) build the Neo stack against system LLVM and don’t integrate the OpenCL 3.0 patches from here: https://github.com/intel/opencl-clang/blob/ocl-open-110/patches/clang/0001-OpenCL-3.0-support.patch. We (Arch) discussed integrating it, but it was decided too big to backport. So we will likely not have OpenCL 3.0 before LLVM 12.

ArchangeGabriel, 2020-12-05:
Actually that does not seem quite exact: it was a bug in clinfo, which did not handle OpenCL 3.0. It has been fixed in a newer release, but I’m not sure whether OpenCL 3.0 really works without the aforementioned patch.

JacekDanecki, 2021-01-04:
@CaoZhongZ Have you tried with newer ocl-icd?

CaoZhongZ, 2021-01-10:
Problem still exists however does not affect other functionality. So I'm going to close it.

closed 2021-01-10 by CaoZhongZ