intel/compute-runtime #385: clarify runtime/compiler support for Comet Lake mobile CPUs having UHD graphics

diablodale, 2020-11-24:
Hello. I'm an OpenCV and OpenCL developer. In your README you don't list "gen10" or "comet lake" as being supported.
Is this true?
Or perhaps the support list should be more granular as some UHD models are used across generations.

I ask because the laptop CPUs we are about to purchase and I will develop on (and for production use) have the Intel Core i7-10875H processor which was launched Q2 of this year 2020. Intel Ark writes it is a current processor and has "Intel® UHD Graphics for 10th Gen Intel® Processors" with device id 0x9BC4.

While its generally known that desktop gen10 processors didn't materialize with graphics enabled. This is opposite in mobile processors where the gen10 mobile processors all have graphics enabled and are widely sold.

Google searches suggest the specific model in that mobile processor is the UHD 630. I see elsewhere that UHD 630 was also used in the gen9 processors.
Mesa code writes that device id is "Intel(R) UHD Graphics (Comet Lake 3x8 GT2)"
I do see two other issues here which mention the UHD 630; suggesting some level of functionality.

So I am unclear if this runtime, and the needed compiler, supports the graphics compute in Intel Core i7-10875H processor.

AdamCetnerowski, 2020-11-24:
The listed Gens refer to the GPU, not CPU. Comet Lake is indeed supported. To check if a device is supported, you can find the Device ID at [Ark](ark.intel.com) (yours is 0x9BC4) and look in the supported [device list](https://github.com/intel/compute-runtime/blob/master/opencl/source/dll/linux/devices/devices_base.inl) (line 191 in this case)

In the unlikely event that the device id is not listed, please file an issue and we'll look into.

diablodale, 2020-11-24:
Doh! 🤦‍♀️ I failed to check in this repo's code. My mistake. Sorry.

closed 2020-11-24 by diablodale