intel/compute-runtime #387: Fedora package problem, file in conflict from intel-igc-opencl

metal3d, 2020-12-04:
Hello, Trying to install fedora package from copr:

```
$ sudo dnf install intel-opencl
...
...

[SKIPPED] intel-igc-core-1.0.5585-1.fc32.x86_64.rpm: Already downloaded                                                                                                                                             
[SKIPPED] intel-igc-opencl-1.0.5585-1.fc32.x86_64.rpm: Already downloaded                                                                                                                                           
[SKIPPED] intel-opencl-20.47.18513-1.fc32.x86_64.rpm: Already downloaded                                                                                                                                            
Running transaction check
Transaction check succeeded.
Running transaction test
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: Transaction test error:
  file /usr/lib64/libopencl-clang.so.10 from install of intel-igc-opencl-1.0.5585-1.fc32.x86_64 conflicts with file from package intel-opencl-clang-10.0.12-1.fc32.x86_64
```

lwesiers, 2020-12-18:
hello there, 

Could you please remove the old packet for our compiler "intel-opencl-clang-10.0.12-1.fc32.x86_64" ?

AdamCetnerowski, 2021-01-21:
Closing as this has been answered. Please re-open if this still occurs.

closed 2021-01-21 by AdamCetnerowski

JacekDanecki, 2021-01-21:
IGC specs was changed for Fedora 32 to add conflict with intel-opencl-clang: https://copr.fedorainfracloud.org/coprs/jdanecki/intel-opencl/build/1891301/