intel/compute-runtime #390: clinfo does not detect GPU

jitendra42, 2020-12-11:
`clinfo` detects CPU and one other device but it does not detect intel GPU.

```
(py36) root@933ec02aa577:/home/user# clinfo -l
Platform #0: Intel(R) OpenCL
 `-- Device #0: Genuine Intel(R) CPU $0000%@
Platform #1: Intel(R) FPGA Emulation Platform for OpenCL(TM)
 `-- Device #0: Intel(R) FPGA Emulation Device
(py36) root@933ec02aa577:/home/user#
```
I have added user to `video` group, installed drivers mentioned in below link but still not able to detect Intel GPU https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-focal.html

jandres742, 2020-12-11:
Looks like problems in configuration in your system, as the OpenCL library from this repo is only for GPUs. Please check the contents of `/etc/OpenCL/vendors`. If you have more than one `icd` file, please leave only `intel.icd` and confirm it has a path to libigdrcl.so. Confirm that that file in that path exists. For the other `icd` files, move them like to `/tmp` and rename them, so the system cannot find them. This is just to test.

jitendra42, 2020-12-11:
moved other `icd`  files then verified that file in `intel.icd` exists
```
(py36) user@2d3317e196c8:~/icd_files$ cd /etc/OpenCL/vendors/
(py36) user@2d3317e196c8:/etc/OpenCL/vendors$ ls
intel.icd
(py36) user@2d3317e196c8:/etc/OpenCL/vendors$ cat intel.icd
/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
(py36) user@2d3317e196c8:/etc/OpenCL/vendors$ ls -l /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
-rw-r--r-- 1 root root 5836920 Nov 24 17:20 /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
```

Now it is showing CPU only

```(py36) user@2d3317e196c8:~$ clinfo -l
Platform #0: Intel(R) OpenCL
 `-- Device #0: Genuine Intel(R) CPU $0000%@```

jandres742, 2020-12-11:
Please execute with `strace`, and check what happens when it's trying to find libigdrcl.so, so see if it is finding it or not.

`strace -o output.log clinfo`

Are you using oneAPI? If so, try to execute clinfo in a new shell w/o setting the oneAPI `setvars` script.

jitendra42, 2020-12-11:
yes I am using oneAPI basekit. Ran `clinfo` as you said and attached strace log
```
(py36) user@2d3317e196c8:~$ sudo su
root@2d3317e196c8:/home/user# strace -o output.log clinfo
Number of platforms                               0
```
[output.log](https://github.com/intel/compute-runtime/files/5681963/output.log)

jandres742, 2020-12-11:
Do you confirm your system has a GPU? strace doesnt show that, You should have a line like:

`openat(AT_FDCWD, "/sys/bus/pci/devices/0000:00:02.0/drm", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 5`

but you dont have it. Execute also with `sudo`.

jitendra42, 2020-12-11:
Sorry, forgot to mention that I am running all this inside docker container. On "host" machine I am able to see GPU using clinfo  and in strace log I can see the log like you metioned.
`openat(AT_FDCWD, "/sys/bus/pci/devices/0000:9a:00.0/drm", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 5`

however inside container I am having this issue.
This is how I am running docker container
`docker run --device /dev/dri:/dev/dri  -it -e http_proxy=$http_proxy -e https_proxy=$https_proxy <image-name>  bash`

alalek, 2020-12-12:
> oneAPI

You may want to check environment variables in oneAPI environement.
I have this from `export` output (it overrides `/etc/OpenCL/vendors`): 

```
declare -x OCL_ICD_FILENAMES="libintelocl_emu.so:libalteracl.so:/opt/intel/inteloneapi/compiler/latest/linux/lib/x64/libintelocl.so"
```

jandres742, 2021-01-04:
@jitendra42 I think we solved this already by direct messaging. Feel free to reopen or reach out if there are further concerns.

closed 2021-01-04 by jandres742

jandres742, 2021-08-30:
@theUpsider not detecting the GPU could be for a number of reasons. Firstly, we would need to check:

- is the i915 loaded? you can check if there's dmesg error and if this command `ls /sys/kernel/debug/dri/*/i915*` returns something. 
- are the libraries loading correctly? strace output should display some warnings or errors, or success messages.