intel/compute-runtime #391: WSL/WSL2 and Intel HD Graphics

vrpascuzzi, 2020-12-13:
**System**
  * [Razer Blade Studio 2020](https://www.razer.com/gaming-laptops/Razer-Blade-15/RZ09-0330QEM3-R3U1?gclid=Cj0KCQiA8dH-BRD_ARIsAC24umbOeLOkTF6HH2S9nCNj2TyrOx_ej7jHpwN1041pCNXQKo9e-_EMR4kaAgQFEALw_wcB)
  * Edition:	Windows 10 Pro Insider Preview
Version:	2004
Installed on:	‎12/‎11/‎2020
OS build:	21277.1000
Experience:	Windows Feature Experience Pack 220.32303.0.2003
WSL kernel:	5.4.72-microsoft-standard-WSL2

**Problem**
Intel HD Graphics setup/configuration in WSL/WSL2. Related to #300.

**Process**
[Installed NVIDIA CUDA Quadro for WSL drivers](https://developer.nvidia.com/cuda/wsl/download) for Windows 10 and followed [instructions for CUDA on WSL](https://docs.nvidia.com/cuda/wsl-user-guide/index.html#abstract) to configure an Ubuntu distro. Built and ran a simple CUDA app -- everything works fine. Instructions from NVIDIA are very clear.
Intel is a different story. [Installed Intel beta drivers](https://downloadcenter.intel.com/download/29526) and followed [Intel's GPGPU setup instructions for Ubuntu 20.04](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-focal.html). Built a simple app with [oneAPI](https://www.oneapi.com/) went smoothly but ran into a runtime error when selecting Intel iGPU:
```
$ ./test_sycl
terminate called after throwing an instance of 'cl::sycl::runtime_error'
  what():  No device of requested type available. Please check https://software.intel.com/en-us/articles/intel-oneapi-dpcpp-compiler-system-requirements-beta -1 (CL_DEVICE_NOT_FOUND)
Aborted
```
Selecting `default`/`CPU` device things are fine:
```
$ ./test_sycl
Running on: Intel(R) Core(TM) i7-10875H CPU @ 2.30GHz
<...>
```
I get the same result when using WSL and WSL2.

Anyone able to get Intel HD Graphics (iGPU) working in WSL/WSL2?

alalek, 2020-12-13:
duplicate of #300

vrpascuzzi, 2020-12-13:
> duplicate of #300

In what sense? #300 was from April 2020, and there wasn't yet support for Intel HD Graphics in WSL -- it is now available. However, I am unable to find any instructions to get things working.

My first hunch is that I may need to build intel/compute-runtime and intel/intel-graphics-compiler myself. Please advise.

vrpascuzzi, 2020-12-13:
@alalek,
Do you prefer that next time I re-open a closed issue with follow-ups rather than open a new issue if the problem is similar?

AdamCetnerowski, 2020-12-13:
@vrpascuzzi - we prefer a new issue. Possibly referencing the previous one. 

Thanks for asking and thank you for the report.

vrpascuzzi, 2020-12-13:
Thanks, @AdamCetnerowski!
Indeed, the previous one was referenced in the original post.

bfliflet, 2020-12-14:
WSL/WSL2 are not currently supported in the compute runtime.  We are actively exploring support and will update this issue accordingly.

vrpascuzzi, 2020-12-14:
Thanks, @bfliflet.

Can you fill me in on what purpose the [Intel Graphics BETA WSL Windows 10 DCH Drivers](https://downloadcenter.intel.com/download/29526) serve? Really I was just hoping for instructions to follow to have the iGPU work under WSL/WSL2.

bfliflet, 2020-12-14:
@vrpascuzzi, thanks for the interest.  The initial support in that driver is for 

"D3D12 on Linux support: Enables running DirectML training via TensorFlow in a Linux terminal on a Windows 10 host."

The communication flow for the compute runtime will piggy-back on top of this previous work; however, the compute-runtime path for WSL/WSL2 relies on a parallel user-mode driver / kernel mode driver interface flow that is still being evaluated by ourselves and relevant parties.

vrpascuzzi, 2020-12-14:
I appreciate the reply, @bfliflet.

Looking forward to the added support. Please let me know if I can help testing.

jchodor, 2021-11-10:
WSL2 support was recently added. Please see here : https://github.com/intel/compute-runtime/blob/master/WSL.md for instructions.

closed 2021-11-10 by jchodor

fedecompa, 2024-10-01:
I have a similar problem. I installed all the packages from Intel OneAPI link. On Windows 11 all works fine and the VideoProcessor Intel(R) UHD Graphics can be used from OpenVINO libraries. On the contrary on WSL2 with the Ubuntu 22.04 distro I see no GPU:

  clinfo
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  
   sycl-ls
   [opencl:cpu][opencl:0] Intel(R) OpenCL, Intel(R) Core(TM) i7-10850H CPU @ 2.70GHz OpenCL 3.0 (Build 0) [2024.18.7.0.11_160000]
   
   
   https://www.intel.com/content/www/us/en/docs/oneapi/installation-guide-linux/2023-0/configure-wsl-2-for-gpu-workflows.html