intel/compute-runtime #392: Intel HD Graphics: Crash when rendering default cube using blender cycles

jarrellmark, 2020-12-22:
Hi,

Blender is crashing while rendering a default cube with the compute-runtime NEO opencl driver on an Intel HD Graphics BroadWell GPU.

I found some related issues, mainly https://github.com/intel/compute-runtime/issues/178 and https://github.com/intel/intel-graphics-compiler/issues/123. Let me know if I should be commenting in one of those threads instead. It looks like it is working for some situations based on @JacekDanecki 's [comment](https://github.com/intel/compute-runtime/issues/178#issuecomment-712093261). Thanks for all the work so far on this.

**Here's my specs:**

CPU: Intel(R) CORE i5 BroadWell
GPU: Intel(R) HD Graphics 5500 BroadWell
OS: Ubuntu 20.04 focal

**Here's what I did to create the issue:**

Added intel-opencl [PPA](https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl).

$ sudo add-apt-repository ppa:intel-opencl/intel-opencl
$ sudo apt-get update

Installed various dependencies using apt:

$ sudo apt install libigc libigc-dev
$ sudo apt install libigdgmm-dev>=20.2
$ sudo apt install libigdgmm11>=20.2

Compiled compute-runtime version as mentioned [here](https://github.com/intel/intel-graphics-compiler/issues/123#issuecomment-584657917). I got version 20.28.17293 from GitHub because that's the version in Launchpad, so it compiled with the PPA dependencies.

I followed all the instructions in [BUILD.md](https://github.com/intel/compute-runtime/blob/20.28.17293/BUILD.md) with the exception of adding `-DCMAKE_BUILD_TYPE=Debug` to the cmake line.

Downloaded Blender source code from GitHub: commit 0144b7094899798625f98cb08d90db2f05f329cb, date Sat Dec 19 14:48:36 2020 -0500.

Made a small change in the source code to make sure the NEO device would show up. Compiled in debug mode.

Ran Blender in Qt Creator with the environment variable EnableStatelessToStatefulBufferOffsetOpt=0 as mentioned [here](https://github.com/intel/intel-graphics-compiler/issues/123#issuecomment-584657917).

Went to Edit -> Preferences -> System -> OpenCL and made sure the NEO device was the only one checked.

Click on Render Properties. Change Render Engine to Cycles. Change Device to GPU Compute.

Click Render -> Render Image. This is when the segmentation fault happens.

The debug-enabled function is:

function: NEO::Linux::OsLibrary::OsLibrary
file: os_library_linux.cpp

code snippet:

```c++
OsLibrary::OsLibrary(const std::string &name) {
    if (name.empty()) {
        this->handle = dlopen(0, RTLD_LAZY);
    } else {
#ifdef SANITIZER_BUILD
        constexpr auto dlopenFlag = RTLD_LAZY;
#else
        constexpr auto dlopenFlag = RTLD_LAZY | RTLD_DEEPBIND;
#endif
        this->handle = dlopen(name.c_str(), dlopenFlag);
    }
}
```

Exact line is the third from bottom: `this->handle = dlopen(name.c_str(), dlopenFlag);`

Here's the output from Blender:

```
22:51:44: Debugging starts
ndof: spacenavd not found
Cycles: compiling OpenCL program base...
Cycles: compiling OpenCL program background...
Segmentation fault (core dumped)
Cycles: compiling OpenCL program base...
Segmentation fault (core dumped)
```

Let me know if you need any more details.

Thanks and happy holidays!

JacekDanecki, 2021-01-08:
My latest tests with Blender I've executed in October on Fedora 31 with IGC and Blender compiled with llvm 9. They were run on SKL device (Intel(R) Graphics Gen9 [0x591b]). 
I don't remember what Neo release I was using for tests, so I'll prepare new setup with current Neo.

JacekDanecki, 2021-01-14:
I've recompiled Blender
```
$ git log -q --oneline -1
5edba9b42f6 Fix T81688: BPY_thread_save crashes with Python 3.9
```
under Fedora 34, and executed it with [Neo release 21.01.18793](https://github.com/intel/compute-runtime/releases/tag/21.01.18793), and default cube was rendered correctly on Skylake GPU.
I've *not* set EnableStatelessToStatefulBufferOffsetOpt=0 and did *not* change Blender code.
I've executed Blender using command:
```
CYCLES_OPENCL_SPLIT_KERNEL_TEST=1 CYCLES_OPENCL_TEST=all ./blender -E CYCLES --debug-cycles
```
Can you try with Neo release 21.01.18793 from releases page: https://github.com/intel/compute-runtime/releases.
You can also try with updated Neo packages from ppa:~intel-opencl for Ubuntu 18.04. 
I've not updated packages on launchpad for Ubuntu 20.04 yet, because there is issue in IGC compilation for release 1.0.5964 (https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl/+sourcepub/11981742/+listing-archive-extra)

On my side I'll try with Blender commit you've used, and with newer code.

JacekDanecki, 2021-01-15:
I've compiled Blender with commit id provided in this issue, and also the latest blender available
```
 git log -q --oneline -1
cab6d5e94a6 (HEAD -> master, origin/master, origin/HEAD) Merge branch 'blender-v2.92-release'
```
and both versions worked as expected on SKL.

jarrellmark, 2021-01-15:
Thanks @JacekDanecki ! I'll try again this weekend with Neo 21.01.18793.

jarrellmark, 2021-01-22:
Unfortunately I'm still getting an error on Ubuntu 20.04.

I installed Neo 21.01.18793 using the deb files. `clinfo` shows the correct version.

```sh
$ clinfo

...
Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Graphics [0x1616]
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 3.0 NEO 
  Driver Version                                  21.01.18793
  Device OpenCL C Version                         OpenCL C 3.0 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
...
```

I also created a new Blender installation using cab6d5e94a6. Didn't modify any code this time:

```
$ git log -q --oneline -1
cab6d5e94a6 (HEAD) Merge branch 'blender-v2.92-release'
```

The only thing I could think of is I have newer Mesa drivers installed which I did as a part of the Proton installation for Steam. Everything else on my system is pretty much vanilla Ubuntu 20.04.

Installation for Proton was done like this:

```
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt dist-upgrade
sudo apt install mesa-vulkan-drivers mesa-vulkan-drivers:i386
```

Instructions here: https://github.com/ValveSoftware/Proton/wiki/Requirements

Here's Blender's output:

```
CYCLES_OPENCL_SPLIT_KERNEL_TEST=1 CYCLES_OPENCL_TEST=all ./blender -E CYCLES --debug-cycles
I0121 23:07:11.923118 300958 blender_python.cpp:194] Debug flags initialized to:
CPU flags:
  AVX2       : True
  AVX        : True
  SSE4.1     : True
  SSE3       : True
  SSE2       : True
  BVH layout : EMBREE
  Split      : False
CUDA flags:
  Adaptive Compile : False
OptiX flags:
  CUDA streams : 1
OpenCL flags:
  Device type    : ALL
  Debug          : False
  Memory limit   : 0
I0121 23:07:23.953624 300958 device_opencl.cpp:48] CLEW initialization succeeded.
I0121 23:07:23.958070 300958 opencl_util.cpp:957] Enumerating devices for platform Intel(R) OpenCL HD Graphics.
I0121 23:07:23.958122 300958 opencl_util.cpp:995] Adding new device Intel(R) Graphics [0x1616].
I0121 23:07:38.161764 301013 device_opencl_impl.cpp:640] Creating new Cycles device for OpenCL platform Intel(R) OpenCL HD Graphics, device Intel(R) Graphics [0x1616].
I0121 23:07:38.163306 301013 blender_sync.cpp:254] Total time spent synchronizing data: 0.000906944
I0121 23:07:38.164772 301015 scene.cpp:540] Requested features:
Experimental features: On
Max nodes group: 0
Nodes features: 0
Use Hair: False
Use Object Motion: False
Use Camera Motion: False
Use Baking: False
Use Subsurface: False
Use Volume: False
Use Branched Integrator: False
Use Patch Evaluation: False
Use Transparent Shadows: False
Use Principled BSDF: True
Use Denoising: False
Use Displacement: False
Use Background Light: True
I0121 23:07:38.164875 301015 device_opencl_impl.cpp:760] Loading kernels for platform Intel(R) OpenCL HD Graphics, device Intel(R) Graphics [0x1616].
I0121 23:07:38.165014 301015 opencl_util.cpp:298] OpenCL program base not found in cache.
I0121 23:07:38.190831 301015 opencl_util.cpp:298] OpenCL program base not found on disk.
I0121 23:07:38.190964 301015 opencl_util.cpp:298] OpenCL program background not found in cache.
I0121 23:07:38.191079 300978 opencl_util.cpp:298] OpenCL program base not found in cache.
Cycles: compiling OpenCL program base...
I0121 23:07:38.268992 300978 opencl_util.cpp:298] Build flags: 
I0121 23:07:38.290711 301015 opencl_util.cpp:298] OpenCL program background not found on disk.
I0121 23:07:38.290967 300979 opencl_util.cpp:298] OpenCL program background not found in cache.
Cycles: compiling OpenCL program background...
I0121 23:07:38.503335 300979 opencl_util.cpp:298] Build flags: -D__KERNEL_EXPERIMENTAL__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__
Segmentation fault (core dumped)
I0121 23:07:39.410238 300978 opencl_util.cpp:298] Separate-process building of /home/mark/.cache/cycles/kernels/cycles_kernel_base_9CCD6B9180E01A2A1CB9C9EFC8887D12_BD47D827337A14C375FD8F11388F4A73.clbin failed, will fall back to regular building.
Cycles: compiling OpenCL program base...
I0121 23:07:39.450955 300978 opencl_util.cpp:298] Build flags: 
I0121 23:07:39.451062 300978 opencl_util.cpp:325] Build options passed to clBuildProgram: '-cl-no-signed-zeros -cl-mad-enable -cl-std=CL2.0 -D__KERNEL_CL_KHR_FP16__ -DWITH_NANOVDB '.
Writing: /tmp/blender.crash.txt
Segmentation fault (core dumped)
```

Thanks! I'd be happy to provide any other info.

JacekDanecki, 2021-01-26:
Can you run offline compiler on kernel_background.cl kernel? In directory build_linux/bin/2.93/scripts/addons/cycles/source
```
ocloc -file kernel/kernels/opencl/kernel_background.cl -device bdw
```
When compilation finishes correctly you should see message:
```
Build succeeded.
```
and 3 binaries created:
```
kernel_background_Gen8core.bin
kernel_background_Gen8core.gen
kernel_background_Gen8core.spv
```

jarrellmark, 2021-01-27:
Hi @JacekDanecki ,

Yes, I ran that command and saw the `Build succeeded` message along with the 3 files.

Thanks

JacekDanecki, 2021-01-27:
What about commands:
```
ocloc -file kernel/kernels/opencl/kernel_background.cl -device bdw -options "-D__KERNEL_EXPERIMENTAL__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ 
-D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__"

ocloc -file kernel/kernels/opencl/kernel_base.cl -device bdw -options "-cl-no-signed-zeros -cl-mad-enable -cl-std=CL2.0 -D__KERNEL_CL_KHR_FP16__"
```
As I can see in your log you've -D__KERNEL_EXPERIMENTAL__ flag added during kernel_background.cl compilation. I can't add  it even I set "Experimental" in "Rendring/Scene/Feature Set"

Could you set:
```
export IGC_ShaderDumpEnableAll=1
```
before running blender and share files created in directory /tmp/IntelIGC

jarrellmark, 2021-01-28:
Looks like it might be related to the build flags.

First I ran this:

```
ocloc -file kernel/kernels/opencl/kernel_background.cl -device bdw -options "-D__KERNEL_EXPERIMENTAL__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ 
-D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__"
```

Got the following output:

```
Unrecognized build options: 
-D__NO_SUBSURFACE__
Build failed with error code: -11
Command was: ocloc -file kernel/kernels/opencl/kernel_background.cl -device bdw -options -D__KERNEL_EXPERIMENTAL__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ 
-D__NO_SUBSURFACE__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__
```

Next I ran the same command but without `-D__NO_SUBSURFACE__`:

```
ocloc -file kernel/kernels/opencl/kernel_background.cl -device bdw -options "-D__KERNEL_EXPERIMENTAL__ -D__NODES_MAX_GROUP__=0 -D__NODES_FEATURES__=0 -D__NO_HAIR__ -D__NO_OBJECT_MOTION__ -D__NO_CAMERA_MOTION__ -D__NO_BAKING__ -D__NO_VOLUME__ -D__NO_BRANCHED_PATH__ -D__NO_PATCH_EVAL__ -D__NO_TRANSPARENT__ -D__NO_SHADOW_TRICKS__ -D__NO_DENOISING__ -D__NO_SHADER_RAYTRACE__"
```

The output was:

```
Build succeeded.
```

The build also produced the 3 files: `kernel_background_Gen8core.bin`, `kernel_background_Gen8core.gen`, and `kernel_background_Gen8core.spv`.

I also ran this command:

```
ocloc -file kernel/kernels/opencl/kernel_base.cl -device bdw -options "-cl-no-signed-zeros -cl-mad-enable -cl-std=CL2.0 -D__KERNEL_CL_KHR_FP16__"
```

Output was:

```
Build succeeded.
```

And three files were produced: `kernel_base_Gen8core.bin`, `kernel_base_Gen8core.gen`, and `kernel_base_Gen8core.spv`.

Next I ran the following:

```
$ export IGC_ShaderDumpEnableAll=1
$ echo IGC_ShaderDumpEnableAll
1
$ CYCLES_OPENCL_SPLIT_KERNEL_TEST=1 CYCLES_OPENCL_TEST=all ./blender -E CYCLES --debug-cycles
```

Then I made sure the only OpenCL device set was `Intel Graphics [0x1616]`, and also changed the Cycles Device to GPU Compute. Rendering the default cube produced the same error I got last time as expected. The output mentioned `-D__NO_SUBSURFACE__`.

However, there was no file called `/tmp/IntelIGC`. Maybe Blender crashed before it could write it out. I also ran: `find / -name "IntelIGC*" 2>/dev/null` which didn't return anything.

JacekDanecki, 2021-01-28:
You've got "Unrecognized build options", because probably you copied command and some character was added between -D__NO_VOLUME__ and -D__NO_SUBSURFACE__. Can you use test1.sh script to run ocloc?
[test1.sh.gz](https://github.com/intel/compute-runtime/files/5887252/test1.sh.gz)

To verify whether IGC can dump files, can you run:
```
IGC_ShaderDumpEnableAll=1 clinfo
```
and then see whether /tmp/IntelIGC directory is created?

jarrellmark, 2021-01-29:
Thanks @JacekDanecki ,

Turns out a newline or something was introduced when I copied & pasted.

I ran the script `test1.sh` and got the following output:

```
Build succeeded.
```

It took a few minutes to compile. When blender crashes, it happens right after I press render.

The ocloc command in `test1.sh` created a folder called /tmp/IntelIGC. I zipped it up, but it's too big to share on GitHub. Here's a dropbox link: https://www.dropbox.com/s/r7ph8pgzpw6dmjr/IntelIGC.zip?dl=0

JacekDanecki, 2021-01-29:
Can you run blender under gdb:
```
$ export CYCLES_OPENCL_SPLIT_KERNEL_TEST=1
$ export CYCLES_OPENCL_TEST=all 
$ gdb ./blender
(gdb) run -E CYCLES --debug-cycles
```
and provide output from below gdb commands after segfault:
```
bt
info threads
info shared
info files
info proc
``` 
Do you have mesa-opencl-icd installed on your system? If so, can you disable or remove it?
What is content of /etc/OpenCL/vendors directory?
Can you provide full output from clinfo?

jarrellmark, 2021-02-04:
Hi @JacekDanecki ,

I ran the gdb commands and attached their outputs as files.

I don't have `mesa-opencl-icd` installed.

```
$ apt search mesa-opencl
Sorting... Done
Full Text Search... Done
mesa-opencl-icd/focal 20.3.4~kisak1~f amd64
  free implementation of the OpenCL API -- ICD runtime
```

The icd packages I do have installed are:

```
$ apt search icd

ocl-icd-dev/focal,now 2.2.11-1ubuntu1 amd64 [installed]
  Development files to build a ICD Loader

ocl-icd-libopencl1/focal,now 2.2.11-1ubuntu1 amd64 [installed,automatic]
  Generic OpenCL ICD Loader

ocl-icd-opencl-dev/focal,now 2.2.11-1ubuntu1 amd64 [installed]
  OpenCL development files
```

My `/etc/OpenCL/vendors` directory looks like this:

```
$ ls /etc/OpenCL/vendors
intel-beignet-x86_64-linux-gnu.icd  intel.icd
```

I had beignet installed at one point, but I uninstalled it a while ago.

`clinfo` output is also attached.

Thanks!

[info_threads.txt](https://github.com/intel/compute-runtime/files/5923695/info_threads.txt)
[info_shared.txt](https://github.com/intel/compute-runtime/files/5923696/info_shared.txt)
[info_proc.txt](https://github.com/intel/compute-runtime/files/5923697/info_proc.txt)
[info_files.txt](https://github.com/intel/compute-runtime/files/5923698/info_files.txt)
[clinfo.txt](https://github.com/intel/compute-runtime/files/5923699/clinfo.txt)
[bt.txt](https://github.com/intel/compute-runtime/files/5923700/bt.txt)

JacekDanecki, 2021-02-11:
I've recompiled Blender on Ubuntu 20.04 and executed it on BDW machine. Default cube was rendered correctly.
Before I built I'd used below command to install dependencies:
```
./install_deps.sh --skip-osl --skip-alembic --skip-embree --skip-openvdb --skip-ocio --skip-oiio --skip-osd --skip-opencollada --skip-ffmpeg  --skip-openxr --skip-oidn --skip-llvm --skip-tbb --skip-boost  --skip-numpy --skip-python
```
To compile Blender I used command:
```
make -j8 BUILD_CMAKE_ARGS="-U *SNDFILE* -U PYTHON* -U *BOOST* -U *Boost* -U *TBB* -U *OPENCOLORIO* -U *OPENEXR* -U *OPENIMAGEIO* -U *LLVM* -U *CYCLES* -U *OPENSUBDIV* -U *OPENVDB*  -U *BLOSC* -U *COLLADA* -U *FFMPEG* -U *ALEMBIC* -U *USD* -U *EMBREE* -U *OPENIMAGEDENOISE* -U *OPENXR* -D WITH_CODEC_SNDFILE=ON -D PYTHON_VERSION=3.8 -D WITH_CYCLES_OSL=OFF -D WITH_LLVM=OFF -D WITH_USD=ON -D USD_ROOT_DIR=/opt/lib/usd -D WITH_XR_OPENXR=ON -D XR_OPENXR_SDK_ROOT_DIR=/opt/lib/xr-openxr-sdk"
```
I've changed parameter PYTHON_VERSION from default 3.7 to 3.8.

How did you compile Blender?

fubutv, 2021-03-02:
Thanks @JacekDanecki 

This could be related to the way I installed the dependencies. I used the binary libraries from here https://svn.blender.org/svnroot/bf-blender/trunk/lib/linux_centos7_x86_64 described here https://wiki.blender.org/wiki/Building_Blender/Linux/Ubuntu. The site said they should work for distributions besides CentOS, but maybe they don't work with this configuration.

I got an error when trying install_deps.sh. I'll try again later and will provide more details when I have some more time to debug.

jarrellmark, 2021-03-02:
Thanks @JacekDanecki

This could be related to the way I installed the dependencies. I used the binary libraries from here https://svn.blender.org/svnroot/bf-blender/trunk/lib/linux_centos7_x86_64 described here https://wiki.blender.org/wiki/Building_Blender/Linux/Ubuntu. The site said they should work for distributions besides CentOS, but maybe they don't work with this configuration.

I got an error when trying install_deps.sh. I'll try again later and will provide more details when I have some more time to debug.

JacekDanecki, 2021-03-02:
I've recompiled Blender with llvm 9.0.1 sources, and with llvm 11 from ppa:kisak/kisak-mesa repository.
I also tested compilation with default cmake parameters:
```
make -j8 BUILD_CMAKE_ARGS="-D PYTHON_VERSION=3.8"
```
In all above configurations, Blender worked correctly.
Can you share list of installed packages, file build_linux/CMakeCache.txt, and output from commads:
```
ls -l /opt/lib
ls -l ~/src/blender-deps
```
Have you cleared ~/.cache/cycles/kernels directory before Blender execution?
Some libraries from linux_centos7_x86_64 I've used on Fedora for some tests without issues.

jarrellmark, 2021-03-08:
Hi @JacekDanecki ,

It's working now, thanks for all the help! It's really great to see the increased render times.

The default cube doesn't look as good as the one rendered with the CPU, but I guess maybe that's something on the Blender side of things. Here's what each render looked like.

GPU: 1 min 36 seconds render time

[image]

CPU: 10 mins 59 seconds render time

[image]

The only thing I changed was Render Properties (right-hand menu) -> Performance -> Tiles -> Tiles X and Tiles Y from 64 px to 512 px for the GPU render. Changed it back to 64x64 for the CPU render.

Here's what was needed to get it working:

To install Blender, I followed the instructions here with a few modifications: https://wiki.blender.org/wiki/Building_Blender/Linux/Ubuntu

Install Packages

```
sudo apt-get update
sudo apt-get install build-essential git subversion cmake libx11-dev libxxf86vm-dev libxcursor-dev libxi-dev libxrandr-dev libxinerama-dev libglew-dev
```

Also had to install Embree to fix a compilation error.

```
sudo apt install libembree-dev
```

Download Sources

```
mkdir ~/blender-git
cd ~/blender-git
git clone https://git.blender.org/blender.git
```

Checkout branch we used earlier (probably not needed)

```
git checkout cab6d5e94a6
```

Build Libraries

```
cd ~/blender-git
./blender/build_files/build_environment/install_deps.sh
```

That failed so then I did:

```
./blender/build_files/build_environment/install_deps.sh --skip-osl --skip-embree
```

This commands output gave the following make command which I saved for later (didn't run yet):

```
make -j4 BUILD_CMAKE_ARGS="-U *SNDFILE* -U PYTHON* -U *BOOST* -U *Boost* -U *TBB* -U *OPENCOLORIO* -U *OPENEXR* -U *OPENIMAGEIO* -U *LLVM* -U *CYCLES* -U *OPENSUBDIV* -U *OPENVDB*  -U *BLOSC* -U *COLLADA* -U *FFMPEG* -U *ALEMBIC* -U *USD* -U *EMBREE* -U *OPENIMAGEDENOISE* -U *OPENXR* -D WITH_CODEC_SNDFILE=ON -D PYTHON_VERSION=3.8 -D WITH_OPENCOLORIO=ON -D WITH_CYCLES_OSL=OFF -D WITH_LLVM=OFF -D WITH_OPENSUBDIV=ON -D OPENSUBDIV_ROOT_DIR=/opt/lib/osd -D WITH_OPENVDB=ON -D WITH_OPENVDB_BLOSC=ON -D OPENVDB_ROOT_DIR=/opt/lib/openvdb -D BLOSC_ROOT_DIR=/opt/lib/blosc -D WITH_ALEMBIC=ON -D ALEMBIC_ROOT_DIR=/opt/lib/alembic -D WITH_USD=ON -D USD_ROOT_DIR=/opt/lib/usd -D WITH_CODEC_FFMPEG=ON -D FFMPEG_LIBRARIES='avformat;avcodec;avutil;avdevice;swscale;swresample;lzma;rt;theora;theoradec;theoraenc;vorbis;vorbisenc;vorbisfile;ogg;x264;openjp2' -D WITH_XR_OPENXR=ON -D XR_OPENXR_SDK_ROOT_DIR=/opt/lib/xr-openxr-sdk"
```

Update and Build

```
cd ~/blender-git/blender
make update

# Used the make command from above
make -j4 BUILD_CMAKE_ARGS="-U *SNDFILE* -U PYTHON* -U *BOOST* -U *Boost* -U *TBB* -U *OPENCOLORIO* -U *OPENEXR* -U *OPENIMAGEIO* -U *LLVM* -U *CYCLES* -U *OPENSUBDIV* -U *OPENVDB*  -U *BLOSC* -U *COLLADA* -U *FFMPEG* -U *ALEMBIC* -U *USD* -U *EMBREE* -U *OPENIMAGEDENOISE* -U *OPENXR* -D WITH_CODEC_SNDFILE=ON -D PYTHON_VERSION=3.8 -D WITH_OPENCOLORIO=ON -D WITH_CYCLES_OSL=OFF -D WITH_LLVM=OFF -D WITH_OPENSUBDIV=ON -D OPENSUBDIV_ROOT_DIR=/opt/lib/osd -D WITH_OPENVDB=ON -D WITH_OPENVDB_BLOSC=ON -D OPENVDB_ROOT_DIR=/opt/lib/openvdb -D BLOSC_ROOT_DIR=/opt/lib/blosc -D WITH_ALEMBIC=ON -D ALEMBIC_ROOT_DIR=/opt/lib/alembic -D WITH_USD=ON -D USD_ROOT_DIR=/opt/lib/usd -D WITH_CODEC_FFMPEG=ON -D FFMPEG_LIBRARIES='avformat;avcodec;avutil;avdevice;swscale;swresample;lzma;rt;theora;theoradec;theoraenc;vorbis;vorbisenc;vorbisfile;ogg;x264;openjp2' -D WITH_XR_OPENXR=ON -D XR_OPENXR_SDK_ROOT_DIR=/opt/lib/xr-openxr-sdk"
```

Run blender

```
cd ~/blender-git/build_linux/bin
CYCLES_OPENCL_SPLIT_KERNEL_TEST=1 CYCLES_OPENCL_TEST=all ./blender -E CYCLES --debug-cycles
```

JacekDanecki, 2021-03-08:
Closing and issue, as Blender is working now.

closed 2021-03-08 by JacekDanecki