intel/compute-runtime #395: zeCommandListAppendSignalEvent stuck in "NOT_READY"

al42and, 2021-01-04:
Hello,

Trying to submit an event to a pristine CommandList with `zeCommandListAppendSignalEvent` results in the event that is permanently not ready, and trying to synchronize results in timeout (or infinite loop if no timeout set).

Submitting to CommandListImmediate works as expected (the event is immediately triggered, and waiting on it takes no time).

Same behavior for `zeCommandListAppendBarrier` (with no prerequisite events).

Tested with the latest compute-runtime release (20.52.18783). Also tried compiling [master branch](https://github.com/oneapi-src/level-zero/commit/a88be3290b2022032dec03f0d8a281b0a91c83bb) of oneAPI level zero loader, to no avail.

The behavior is identical on two systems tested: Gen9 (i5-6600, Ubuntu 20.04) and Gen12 (i7-1165G7, Ubuntu 20.10 + kernel 5.10). The hardware logs in the repo below are for the latter system. 

Simple repro code is available here: https://github.com/al42and/zello_hang, based on the `zello_world` sample. The issue also appears when trying to wait on an [enqueued barrier](https://github.com/intel/llvm/blob/sycl/sycl/doc/extensions/EnqueueBarrier/enqueue_barrier.asciidoc) in SYCL code (see `sycl_hang.cpp` in the same repo).

```
$ make zello_hang 
g++ zello_hang.cpp -g -o zello_hang -lze_loader

$ gdb -q ./zello_hang
Reading symbols from ./zello_hang...
(gdb) r
Starting program: /home/aland/sycl_tests/zello_hang 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6ef7640 (LWP 83496)]
Driver initialized.
Found ZE_DEVICE_TYPE_GPU device...
Driver version: 16795999
API version: 1.0
Enqueueing event
Checking event status
Status: ZE_RESULT_NOT_READY
Waiting on the event  # < Hanging there for more than a minute
^C
Thread 1 "zello_hang" received signal SIGINT, Interrupt.
0x00007ffff7bef23b in sched_yield () at ../sysdeps/unix/syscall-template.S:120
120     ../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt
#0  0x00007ffff7bef23b in sched_yield () at ../sysdeps/unix/syscall-template.S:120
#1  0x00007ffff733197a in L0::EventImp::hostSynchronize(unsigned long) () from /usr/local/lib/libze_intel_gpu.so.1
#2  0x00007ffff7f15b6e in zeEventHostSynchronize (hEvent=0x5555558fcbe0, timeout=18446744073709551615) at /home/aland/sycl_workspace/llvm/build/tools/sycl/plugins/level_zero/level_zero/level_zero_loader/source/lib/ze_libapi.cpp:2277
#3  0x000055555555d7a8 in main (argc=1, argv=0x7fffffffc3e8) at zello_hang.cpp:113
```

jandres742, 2021-01-04:
@al42and Thanks. From your code, you have this:

```cpp
zeCommandListAppendSignalEvent(command_list, event);
std::cout << "Checking event status" << std::endl;
status = zeEventHostSynchronize(event, 0);
std::cout << "Status: " << to_string(status) << std::endl;
std::cout << "Waiting on the event" << std::endl;
zeEventHostSynchronize(event, UINT64_MAX);
```

if the concern is that the `hostSynchronize` shown in this snippet is not working as expected with regular list, as opposed to an immediate list, it's because you haven't executed list.

`zeCommandListAppendSignalEvent` only appends the command to the list. If you have an immediate list, then the command is submitted right away, but in a regular list, you need to call `zeCommandQueueExecuteCommandLists` for the appended commands to be submitted to the device for execution. As you have it right now, the `appendSignalEvent` is never executed.

al42and, 2021-01-06:
Thank you very much for pointing out the omission! That solves the issue with the event status.

However, now running [the updated code](https://github.com/al42and/zello_hang/blob/888c9c366b8c0d4b8642710df511a3d249cd91e7/zello_hang.cpp) triggers GPU reset on my Gen12 system (Gen9 works fine), and then NEO runtime hangs in the exit handlers:

DMESG: 
```
Asynchronous wait on fence 0000:00:02.0:kwin_x11[2037]:a0aa timed out (hint:intel_atomic_commit_ready [i915])
i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:85dffffb, in zello_hang [118738]
i915 0000:00:02.0: [drm] Resetting rcs0 for stopped heartbeat on rcs0
i915 0000:00:02.0: [drm] zello_hang[118738] context reset due to GPU hang
```

GDB trace:
```
[[gdb output skipped]]
Driver initialized.
Found ZE_DEVICE_TYPE_GPU device...
Driver version: 16795999
API version: 1.0
Enqueueing event
Checking event status
Status: ZE_RESULT_NOT_READY
Executing the command list
Waiting on the event
Done!
^C      # < Stuck here
Thread 1 "zello_hang" received signal SIGINT, Interrupt.
0x00007ffff7504980 in NEO::CommandStreamReceiver::waitForTaskCountAndCleanAllocationList(unsigned int, unsigned int) () from /usr/local/lib/libze_intel_gpu.so.1
(gdb) bt
#0  0x00007ffff7504980 in NEO::CommandStreamReceiver::waitForTaskCountAndCleanAllocationList(unsigned int, unsigned int) () from /usr/local/lib/libze_intel_gpu.so.1
#1  0x00007ffff7504be1 in NEO::CommandStreamReceiver::cleanupResources() () from /usr/local/lib/libze_intel_gpu.so.1
#2  0x00007ffff7504e54 in NEO::CommandStreamReceiver::~CommandStreamReceiver() () from /usr/local/lib/libze_intel_gpu.so.1
#3  0x00007ffff75acfaa in NEO::DrmCommandStreamReceiver<NEO::TGLLPFamily>::~DrmCommandStreamReceiver() () from /usr/local/lib/libze_intel_gpu.so.1
#4  0x00007ffff752aa3c in NEO::Device::~Device() () from /usr/local/lib/libze_intel_gpu.so.1
#5  0x00007ffff759ba59 in NEO::RootDevice::~RootDevice() () from /usr/local/lib/libze_intel_gpu.so.1
#6  0x00007ffff732b2c1 in L0::DeviceImp::releaseResources() () from /usr/local/lib/libze_intel_gpu.so.1
#7  0x00007ffff732b3b7 in L0::DeviceImp::~DeviceImp() () from /usr/local/lib/libze_intel_gpu.so.1
#8  0x00007ffff732b439 in L0::DeviceImp::~DeviceImp() () from /usr/local/lib/libze_intel_gpu.so.1
#9  0x00007ffff732d337 in L0::DriverHandleImp::~DriverHandleImp() () from /usr/local/lib/libze_intel_gpu.so.1
#10 0x00007ffff732d539 in L0::DriverHandleImp::~DriverHandleImp() () from /usr/local/lib/libze_intel_gpu.so.1
#11 0x00007ffff72ed43c in driverHandleDestructor() () from /usr/local/lib/libze_intel_gpu.so.1
#12 0x00007ffff7fe01a3 in ?? () from /lib64/ld-linux-x86-64.so.2
#13 0x00007ffff7b36a57 in __run_exit_handlers (status=0, listp=0x7ffff7cd5738 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true) at exit.c:108
#14 0x00007ffff7b36c00 in __GI_exit (status=<optimized out>) at exit.c:139
#15 0x00007ffff7b1acb9 in __libc_start_main (main=0x55555555e04f <main(int, char**)>, argc=1, argv=0x7fffffffc408, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffc3f8)
    at ../csu/libc-start.c:348
#16 0x000055555555796e in _start ()
```

bfliflet, 2021-01-06:
If you're intending to create an immediate command list as @jandres742 hinted at please use [zeCommandListCreateImmediate](https://spec.oneapi.com/level-zero/latest/core/api.html#zecommandlistcreateimmediate).  If you're not intending to use an immediate command list, you need to call [zeCommandListClose](https://spec.oneapi.com/level-zero/latest/core/api.html#zecommandlistclose) prior to executing the command list on the command queue.

Regular command lists are designed for batching up work prior to submission and may have better throughput.  Immediate command lists are designed for immediate submission and are designed for reduced latency.  Your sample has patterns of both but should be aligned in one direction or another.

I hope this helps!

al42and, 2021-01-06:
@bfliflet Oh, got it, thanks! I'm not primary working with LevelZero, but rather trying to nail down the issue with using it as a SYCL backend, so I apologize for rookie mistakes :)

closed 2021-01-06 by al42and