intel/compute-runtime #396: Slightly misleading --help output of ocloc

al42and, 2021-01-10:
When looking at valid device types in `ocloc compile --help`, they are printed with capital "G": "Gen9", "Gen12LP", etc.

However, the tool only accepts lower-case "g": "gen9", "gen12LP", etc.

```
# Latest release installed
$ dpkg -l | grep ocloc
ii  intel-ocloc                                                             21.01.18793                                 amd64        Intel(R) Graphics Compute Runtime

# Let's check for which devices we can compile:
$ ocloc compile --help | grep device_type
Usage: ocloc [compile] -file <filename> -device <device_type> [-output <filename>] [-out_dir <output_dir>] [-options <options>] [-32|-64] [-internal_options <options>] [-llvm_text|-llvm_input|-spirv_input] [-options_name] [-q] [-cpp_file] [-output_no_suffix] [--help]
  -device <device_type>         Target device.
                                <device_type> can be: bdw, skl, kbl, cfl, bxt, glk, icllp, lkf, ehl, tgllp, rkl, adls, Gen8, Gen9, Gen11, Gen12LP - can be single or multiple target devices.

# Okay, so Gen12LP is supported, right?
$ ocloc -output test.out -file test.spv -spirv_input -device Gen12LP
Error: Cannot get HW Info for device Gen12LP.
Command was: ocloc -output test.out -file test.spv -spirv_input -device Gen12LP

# Works fine with gen12LP. Same with other Gen's.
```

JablonskiMateusz, 2021-01-12:
Hi, could you retry using ocloc built with https://github.com/intel/compute-runtime/commit/5a5ad64f5d0a99be6ba4e38872c459fc4b316826 ?

al42and, 2021-01-22:
Sorry for the delayed response.

Tested the 21.03.18857 release, the issue is indeed resolved. Both "gen" and "Gen" work fine. Thank you!

closed 2021-01-22 by al42and