intel/compute-runtime #397: compiling Level Zero on a non-IntelGPU system

jeffhammond, 2021-01-11:
I would like to attempt to compile Level Zero for something other than Intel GPUs.  I do not expect this work, but I'd like to see how far I can get.  Unfortunately, right now, CMake prevents me from doing anything interesting, since it expects IGC to be present everywhere.

```
ubuntu@ubuntu:~/intel-computer-runtime-gpu/build$ uname -a
Linux ubuntu 5.4.0-1026-raspi #29-Ubuntu SMP PREEMPT Mon Dec 14 17:01:16 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
```

```
ubuntu@ubuntu:~/intel-computer-runtime-gpu/build$ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- branch dir list: /
-- CMAKE_BUILD_TYPE not specified, using Release
-- Release build configuration
-- Computed OpenCL version major is: 21
-- Computed OpenCL version minor is: 02
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- Source Level Debugger headers dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/source_level_debugger
-- Aub Stream Headers dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/aub_stream/headers
-- Metrics Library dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/metrics_library
-- Metrics Discovery dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/metrics_discovery
-- i915 includes dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/uapi
-- Khronos OpenCL headers dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/opencl_headers
-- Khronos OpenGL headers dir: /home/ubuntu/intel-computer-runtime-gpu/third_party/opengl_headers
-- Third party dir: /home/ubuntu/third_party
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'igc-opencl'
--   No package 'igc-opencl' found
CMake Error at CMakeLists.txt:418 (message):
  Intel Graphics Compiler not found!

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/intel-computer-runtime-gpu/build/CMakeFiles/CMakeOutput.log".
```

```
ubuntu@ubuntu:~/intel-computer-runtime-gpu/level_zero/build$ cmake ..
-- L0: Selected platforms are not supported.
-- Skipping level zero
CMake Warning (dev) at CMakeLists.txt:540 (set):
  Cannot set "BUILD_WITH_L0": current scope has no parent.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.16)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/intel-computer-runtime-gpu/level_zero/build
```

jandres742, 2021-01-11:
> CMake prevents me from doing anything interesting, since it expects IGC to be present everywhere.

That is because when building level-zero (or opencl), we also compile the built-in kernels use for operations like append copy, and that compilation requires IGC. 

What happens when you install IGC? Theoretically, build should complete, since nothing is run in the GPU as part of the build (only unit tests are run, but those use just mock objects). So as long as you have the deps satisfied, you should be able to build it.

jeffhammond, 2021-01-11:
I will try to build IGC but know that an LLVM build on a Raspberry Pi 4 takes approximately one day, so this is a pretty huge pain just to work around the CMake checks.

jeffhammond, 2021-01-11:
I am having similar problems when trying to build IGC.

jandres742, 2021-01-11:
@jeffhammond could you use just the IGC deb packages? 
https://github.com/intel/intel-graphics-compiler/releases
https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.5964/intel-igc-core_1.0.5964_amd64.deb
https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.5964/intel-igc-opencl_1.0.5964_amd64.deb

jeffhammond, 2021-01-11:
As shown above, I am doing this on a Raspberry Pi, which is an AArch64 platform.  I am aware that this is not supported, but I am trying to establish what needs to be done to change that, besides the uninteresting answer that CMake disallows Level Zero from being portable.

jandres742, 2021-01-12:
@JacekDanecki might know something here.

PiotrRozenfeld, 2021-02-15:
By default, we precompile builtin kernels at build time, which creates IGC dependency. However, NEO also supports creating copy kernels at runtime (if precompiled versions are absent) – cmake may be too strict here. We currently have no plans to change that behavior, but contributions are always welcome.

eero-t, 2021-06-30:
> I would like to attempt to compile Level Zero for something other than Intel GPUs.

Level zero is another project: https://github.com/oneapi-src/level-zero

This project provides the L0 Intel _backend_, but L0 applications do not need backend to build, because backend is loaded dynamically at run-time by the level-zero loader.  Or did I misunderstand what is requested?

closed 2022-10-13 by AdamCetnerowski