intel/compute-runtime #398: Conformance and performance of multi-threads ocl processing

chris-pn-dev, 2021-01-15:
Hi,
    I have questions about the conformance and performance of belowing 2 user case:
1. There are N (N >= 2) host-side threads, which setup ocl and do processing indepently. (From clGetPlatformIDs, clGetDeviceIDs, clCreateContext, clCreateCommandQueue to clEnqueueNDRangeKernel...).
2. Same number host threads (N threads). Setup ocl (including clGetPlatformIDs, clGetDeviceIDs, clCreateContext, clCreateCommandQueue) before creating threads. It means host threads will share clContext and clCommandQueue.
    My questions are:
      - Is there any conformance issue in case 1?
      - Does case 1 will have performance loss (consume more cpu or gpu) compared with case2?

MichalMrozek, 2021-01-18:
In OpenCL only clSetKernlArg call is not thread safe, all other API's are by definition, so you shouldn't be running into any conformance problems if you handle that corectly. ( if you do not re-use Kernels between threads you are basically free of issues).

clGetPlatformIDs and clGetDeviceIDs will get serialized, there can be only one instance in the driver of those, so there is no point in doing them in multi threaded fashion.

Driver currently creates OS context per every execution engine device has, on GEN9 there is one execution engine. So no matter how many command queues you will create, they will effectively route down to one execution engine. Since there is one global state being programmed for execution engine, this gets serialized when clEnqueueNDRangeKernel calls are coming from different threads. So doing enqueues from multiple threads will not be done in parallel fashion.

chris-pn-dev, 2021-01-21:
Hi Michal,
    Thanks for your reply.
    Actually my question is about buffer sharing between media and ocl. Becasue of our bussiness logic, we should init ocl in thread 1 for data processing. Then in sometime we will create thread 2 for media usage and we want to share media buffer with ocl for following processing by ocl extension. So thread 2 need to call clGetDeviceIDsFromVA_APIMediaAdapterINTEL and clCreateContext.
    Per ocl spec and your reply, only clSetKernlArg call is not thread safe. So I think it's ok (will have no conformance or performance loss) if I use ocl in 2 threads independently (setup ocl and do processing).

MichalMrozek, 2021-01-21:
Yeah, you may use OpenCL objects in arbitrary amount of threads, there are no restrictions.

chris-pn-dev, 2021-01-25:
Thanks Michal. I think this issue can be closed now. I Will re-open it when I have related question. Thanks!

closed 2021-01-25 by chris-pn-dev