intel/compute-runtime #400: clGetEventProfilingInfo return host timer

TApplencourt, 2021-01-25:
Hi,

As specificed by the specification, `clGetEventProfilingInfo` (https:://www.khronos.org/registry/OpenCL/sdk/1.0/docs/man/xhtml/clGetEventProfilingInfo.html)  should return `device` timers. 

Compute runtime (at least `2021.01.04`) on IrisGen9 seem to return `host` timers. 

Please find a reproducer bellow:

```
$ cat test.cpp
#define CL_HPP_TARGET_OPENCL_VERSION 210 // Need for clGetDeviceAndHostTimer
#include <CL/opencl.hpp>
#include <iostream>
#include <vector>
#include <cassert>

/*
 * Verify that clGetEventProfilingInfo return DeviceTimer
 */
int main(int argc, char* argv[]) {

    /* Select first GPU */
    cl::Device device;

    std::vector<cl::Platform> platforms;
    cl::Platform::get(&platforms);
    for (auto& plat: platforms) {
        std::vector<cl::Device> devices;
        plat.getDevices(CL_DEVICE_TYPE_GPU, &devices);
        if (devices.size() >= 1) {
            device =  devices[0];
            break;
        }
    }
   std::cout << "  -- Device: " << device.getInfo<CL_DEVICE_NAME>() << std::endl;
   cl::Context context(device);

        /*Queue with profiling */
        cl::CommandQueue queue(context, device, CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE|CL_QUEUE_PROFILING_ENABLE);

        /* Dummy kernel with profile*/
        std::string kernelstring{R"CLC(
            __kernel void hello_world(const int id) {
                printf("Hello world from id %d \n", id);
            }
        )CLC"};

        cl::Program program(context, kernelstring);
        program.build();

        cl::Kernel kernel(program, "hello_world");
        auto kernel_func = cl::KernelFunctor<int>(kernel);

        cl::Event event = kernel_func(cl::EnqueueArgs(queue, cl::NDRange(1), cl::NDRange(1)), 0);;
        cl::WaitForEvents(std::vector<cl::Event>{event});

        /*Meat is here */
        /* Proiling Info Spec (https://www.khronos.org/registry/OpenCL/sdk/1.0/docs/man/xhtml/clGetEventProfilingInfo.html)
         * required the timer to be device timer.
         * But they appear to be host timer */
        auto [device_timer, host_timer] = device.getDeviceAndHostTimer();

        cl_ulong event_timer;
        // We use  `CL_PROFILING_COMMAND_QUEUED` but the behavior is the same regardless of the query
        event.getProfilingInfo(CL_PROFILING_COMMAND_QUEUED, &event_timer);

        std::cout << "Host timer: " << host_timer << std::endl;
        std::cout << " Device Timer: " << device_timer << std::endl;
        std::cout << "EventProfilingInfo timer (should be close to device timer " << event_timer <<  " )"<< std::endl;

        /* Assert that the event_timer is closert to the device_timer than the host_timer */
        assert ( std::abs( (cl_long) (event_timer - device_timer) ) <  std::abs( (cl_long) (event_timer - host_timer) ));
}
$  g++ -lOpenCL test.cpp -std=c++17
$ ./a.out
  -- Device: Intel(R) Graphics [0x193a]
Hello world from id 0
Host timer: 2862068602769952
 Device Timer: 4507954724492
EventProfilingInfo timer (should be close to device timer 2862068601422238 )
a.out: test.cpp:62: int main(int, char**): Assertion `std::abs( (cl_long) (event_timer - device_timer) ) < std::abs( (cl_long) (event_timer - host_timer) )' failed.
Aborted (core dumped)
```

Don't hesitate if you need more information,
Regards

Kerilk, 2021-01-25:
To clarify a bit maybe, our issue here is more the discrepancy between what `clGetDeviceAndHostTimer` and `clGetEventProfilingInfo` return. Especially, according to `clGetDeviceAndHostTimer` documentation:
https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/clGetDeviceAndHostTimer.html

> The timestamp returned from clGetEventProfilingInfo for an event on a device and a device timestamp queried from the same device will always be in the same timebase.

TApplencourt, 2021-02-01:
Opened another ticket. Close this issue to avoid duplicating traciking.

closed 2021-02-01 by TApplencourt