intel/compute-runtime #404: Add Comet Lake Support

kcxserver, 2021-02-17:
Hi Guys! 

I learned that I need the following package for the Plex HDR to SDR transcode feature in Ubuntu:

`sudo apt install ocl-icd-libopencl1 beignet-opencl-icd`
https://support.plex.tv/articles/hdr-to-sdr-tone-mapping/

Then I was told that the two Beignet & OpenCL packages did not support Comet Lake CPUs. 
I bought an i3 10100 for the Plex Server and cannot use the Tonemapping feature due to the lack of support. 

There is already a request for the Beignet package:
https://github.com/intel/beignet/pull/20

Many greetings !

PiotrRozenfeld, 2021-02-17:
If your device is i3 10100:
https://ark.intel.com/content/www/us/en/ark/products/199283/intel-core-i3-10100-processor-6m-cache-up-to-4-30-ghz.html

then it is already supported:
https://github.com/intel/compute-runtime/blob/master/opencl/source/dll/linux/devices/devices_base.inl#L190

kcxserver, 2021-02-17:
> then it is already supported:
OK thank you for the info. Then the Plex team didn’t answer quite correctly? Or is only the Beignet package the problem?

closed 2021-02-17 by kcxserver