intel/compute-runtime #406: Wrong version string in 21.08 release

mmmm1998, 2021-03-01:
It looks like the version string for 21.08 haven't been updated - the clinfo still reports the 21.07 version.
And code have this value - confirmation below (last line is important)
```
nikita@workmachine:~/workspace/neo/compute_runtime [[21.08.19096]]
[14:55:16]$ grep "NEO_OCL_DRIVER_VERSION" -r ..
../driver_version.h.in:11:#cmakedefine NEO_OCL_DRIVER_VERSION ${NEO_OCL_DRIVER_VERSION}
../version.cmake:8:  if(NOT DEFINED NEO_OCL_DRIVER_VERSION)
../version.cmake:52:    set(NEO_OCL_DRIVER_VERSION "${NEO_OCL_VERSION_MAJOR}.${NEO_OCL_VERSION_MINOR}.${NEO_VERSION_BUILD}")
../version.cmake:66:  set(NEO_OCL_DRIVER_VERSION "${NEO_OCL_VERSION_MAJOR}.${NEO_OCL_VERSION_MINOR}.${NEO_VERSION_BUILD}")
../opencl/source/cl_device/cl_device_caps.cpp:35:static std::string driverVersion = TOSTR(NEO_OCL_DRIVER_VERSION);
../opencl/source/cl_device/cl_device_caps.cpp:74:    driverVersion = TOSTR(NEO_OCL_DRIVER_VERSION);
../opencl/source/helpers/neo_driver_version.h:23:constexpr const char *driverVersion = TOSTR(NEO_OCL_DRIVER_VERSION);
../opencl/test/unit_test/device/device_caps_tests.cpp:1257:    const std::string expectedVersion = TOSTR(NEO_OCL_DRIVER_VERSION);
../compute_runtime/driver_version.h:11:#define NEO_OCL_DRIVER_VERSION 21.07.0
```

JacekDanecki, 2021-03-02:
Minor part in version is calculated from commit in https://github.com/intel/compute-runtime/blob/master/version.cmake#L39 as ISO week number.
As workaround you can set cmake parameter -DNEO_OCL_VERSION_MINOR, for example:
```
cmake  -DNEO_OCL_VERSION_MINOR=8 
```

JacekDanecki, 2021-03-23:
Fixed by c78f929f718ac89981da173eb92e27cc24725c86

closed 2021-03-23 by JacekDanecki