intel/compute-runtime #407: -DBUILD_WITH_L0=1 fails to with level-zero >= 1.2

jbeich, 2021-03-05:
[level-zero@v1.2.3](https://github.com/oneapi-src/level-zero/releases/tag/v1.2.3) said "Minimum required version 21.09.19150". From [error log](https://github.com/intel/compute-runtime/files/6091174/intel-compute-runtime-21.09.19150.log):
```c++
In file included from level_zero/core/source/gen11/icllp/cmdqueue_icllp.cpp:11:
In file included from level_zero/core/source/cmdqueue/cmdqueue_hw.inl:31:
In file included from level_zero/core/source/driver/driver_handle_imp.h:12:
level_zero/api/extensions/public/ze_exp_ext.h:22:14: error: redefinition of '_ze_module_program_exp_version_t'
typedef enum _ze_module_program_exp_version_t {
             ^
/usr/include/level_zero/ze_api.h:5280:14: note: previous definition is here
typedef enum _ze_module_program_exp_version_t
             ^
In file included from level_zero/core/source/gen11/icllp/cmdqueue_icllp.cpp:11:
In file included from level_zero/core/source/cmdqueue/cmdqueue_hw.inl:31:
In file included from level_zero/core/source/driver/driver_handle_imp.h:12:
level_zero/api/extensions/public/ze_exp_ext.h:38:16: error: redefinition of '_ze_module_program_exp_desc_t'
typedef struct _ze_module_program_exp_desc_t {
               ^
/usr/include/level_zero/ze_api.h:5297:16: note: previous definition is here
typedef struct _ze_module_program_exp_desc_t
               ^
In file included from level_zero/core/source/gen11/icllp/cmdqueue_icllp.cpp:11:
In file included from level_zero/core/source/cmdqueue/cmdqueue_hw.inl:31:
In file included from level_zero/core/source/driver/driver_handle_imp.h:12:
level_zero/api/extensions/public/ze_exp_ext.h:63:14: error: redefinition of '_ze_global_offset_exp_version_t'
typedef enum _ze_global_offset_exp_version_t {
             ^
/usr/include/level_zero/ze_api.h:6013:14: note: previous definition is here
typedef enum _ze_global_offset_exp_version_t
             ^
3 errors generated.
```

JacekDanecki, 2021-03-05:
Neo release 21.09.19150 was tested with https://github.com/oneapi-src/level-zero/releases/tag/v1.1.0, see Neo release notes: https://github.com/intel/compute-runtime/releases/tag/21.09.19150
L0 loader update in commit: https://github.com/intel/compute-runtime/commit/76666662c479d8fa1301e99d1c6c7fa9735d2f1b was reverted in https://github.com/intel/compute-runtime/commit/cac1b5465b768d3b21c03b2357fd7adf4349775d

jbeich, 2021-03-19:
21.11.19310 builds fine.

closed 2021-03-19 by jbeich