intel/compute-runtime #408: DEBUG OPENCL code

menggege321, 2021-03-08:
Hello, how can I debug the kernel code of Intel OpenCL with vs2019 on Windows 10? I've tried the Intel OpenCL SDK on the official website, but I can't debug it. I'm looking forward to your reply, thank you!!!

alalek, 2021-03-08:
BTW, you can always try to "debug" using `printf()` (refer to OpenCL specifications), but ensure first that this preserves the kernel results (printf may hide synchronization bugs).

HoppeMateusz, 2021-03-11:
Hello

there is Intel Distribution of GDB for kernel debugging on GPU.

Please take a look at "Get Started" section:
https://software.intel.com/content/www/us/en/develop/documentation/get-started-with-debugging-dpcpp-windows/top.html

and tutorial with step by step instructions here:
https://software.intel.com/content/www/us/en/develop/documentation/debugging-dpcpp-windows/top/debug-a-program-on-a-gpu.html

You can download Intel Distribution of GDB as part of Intel oneAPI Base Toolkit:
https://software.intel.com/content/www/us/en/develop/tools/oneapi/all-toolkits.html

Mateusz

HoppeMateusz, 2021-03-16:
@menggege321  Did you manage to use debugger with instructions above ?
If you face any problems, please let us know.

menggege321, 2021-03-17:
Thank you for your reply. I mainly want to debug the OpenCL code of Intel GPU in vs2019 on win10. 
According to your last reply, I went to the corresponding website to see the guide, but the guide said "for debugging on a GPU using Intel? Distribution for GDB *, you must complete the remote setup of Microsoft Visual Studio * remote debugger. Debugging and running an application on the same machine is not supported yet ", according to the guide, I need to have two machines at the same time to enter GPU debugging, but my purpose is to debug Intel GPU code on the same machine, I don't know if my understanding is wrong, I hope you can help guide me, thank you.

HoppeMateusz, 2021-03-18:
Hello,

Yes, you need to use two machines to be able to debug GPU on Windows:
- target machine (remote) 
- host machine

Debugging requires disabling preemption that may lead to GUI freeze:
> The installer sets the following registry keys to 0:
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\GraphicsDrivers\Scheduler\EnablePreemption
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\GraphicsDrivers\TdrLevel
These settings disable Timeout Detection and Recovery (TDR) and preemption requests from the GPU scheduler.

With Linux OS, it is possible to run and debug application locally, though ssh connection is recommended.

Mateusz

AdamCetnerowski, 2021-04-15:
This looks resolved. If you have any more questions, please open a new issue.

closed 2021-05-06 by AdamCetnerowski