intel/compute-runtime #409: crash in `NEO::DrmAllocation::makeBOsResident` or in `checkAllocationsForOverlapping` when using more than one opencl block in gnuradio gr-clenabled

stef, 2021-03-08:
not sure if this is a gr-clenabled or a intel opencl icd issue, thus a crosspost notifying you of this report: https://github.com/ghostop14/gr-clenabled/pull/13#issuecomment-792879219

happy to supply more information on request. also happy to copy paste the issue itself here in case that is more convenient to process.

HoppeMateusz, 2021-03-19:
Hi,

problems with `checkAllocationsForOverlapping()` or with `this->fragmentsStorage.fragmentStorageData[0].residency` are related and may originate from overlapping memory regions used: a. to create multiple buffers / b. as host_ptr to read/write calls.

To get a better idea what might be happening please check if following is the case:

1. Is flag CL_USE_HOST_PTR  used to create buffers?
Are such buffers mapped (clEnqueueMapBuffer) and returned pointers used to create other buffers? Or passed as a ptr to EnqueueReadWriteBuffer/Image() calls? 

This is forbidden by spec.

2. Is pointer returned from Map used in an overlapping manner? 
Offset pointer used to create buffers :
auto ptr = clEnqueueMapBuffer(buffer, mapSize);
auto offsetPtr = ptr + 20100;
clCreateBuffer(CL_MEM_USE_HOST_PTR, size, offsetPtr);

3. API log from CLIntercept would be helpful, or igdrcl.log after setting debug variables with env: NEOReadDebugKeys=1 LogApiCalls=1
It should be possible to see if the same or offset host_ptr is used to multiple calls

4. You may also try below debug keys to check if they resolve the problem
NEOReadDebugKeys=1 EnableHostPtrTracking=0

Mateusz

stef, 2021-03-19:
1. no `CL_USE_HOST_PTR` is not found anywhere in the sources of gr-clenabled.
2. i did not see any call to `clEnqueueMapBuffer` nor `clCreateBuffer` in gr-clenabled sources.
3. i set these variables (and googled where i could find this igdrcl.log) but no result. i also rerun the whole test with `strace -f` but there was no call to `getenv` anywhere in the strace output.
4. see 3. and lack of calls to `getenv` in straced test.

HoppeMateusz, 2021-03-22:
1. / 2. Ok, so only images are used ( clCreateImage / clEnqueueRead/Write/MapImage )?
Is clEnqueueMapImage used? is returned pointer used to Read/Write calls ?

3. Do you see libigdrcl.so to be opened in strace log?
it is strange that getenv is not there - do you use recent driver, getenv should be called just after opening libigdrcl.so ? What version of compute runtime do you use?

stef, 2021-03-22:
> / 2. Ok, so only images are used ( clCreateImage / clEnqueueRead/Write/MapImage )? Is clEnqueueMapImage used? is returned pointer used to Read/Write calls ?

sorry if i am annoying, but that's because i'm a noob and have no clue about cl - but i'm super excited :)

i checked, and neither `clCreateImage` nor `clEnqueue` strings are found in the code. buffers are created with c++ new operator calling this signature: `cl::Buffer (const Context &context, cl_mem_flags flags, size_type size)```

this is the specific file that i am testing with: https://github.com/ghostop14/gr-clenabled/blob/master/lib/clMathOp_impl.cc - in case it helps to look at this.

> Do you see libigdrcl.so to be opened in strace log?

```
18934 openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so", O_RDONLY|O_CLOEXEC) = 4
18934 read(4, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260Z\2\0\0\0\0\0"..., 832) = 832
```

looks ok to me.

> it is strange that getenv is not there - do you use recent driver, getenv should be called just after opening libigdrcl.so ? What version of compute runtime do you use?

i use `20.44.18297-1`  - the latest version - from debian unstable.

HoppeMateusz, 2021-03-24:
Thanks for checks, 

this project is using C++ wrappers, it is not possible to tell from the code if overlapping memory regions are used or not.

Driver version you are using should support NEOReadDebugKeys - strace won;t show getenv calls, you need to use ltrace:
ltrace ./your_app

Can you try collecting log with:
NEOReadDebugKeys=1 PrintDebugSettings=1 LogApiCalls=1  ./your_app  ? It will help root cause the problem.

Log - igdrcl.log file - should be produced in current working directory from where you execute application. Please make sure you have correct rights. 
You should also see modified, non-default debug keys in the output.

stef, 2021-03-24:
> Thanks for checks,

thanks for your continued engagement!

> this project is using C++ wrappers, it is not possible to tell from the code if overlapping memory regions are used or not.

the c++ class being created seems to be standard though: https://github.khronos.org/OpenCL-CLHPP/classcl_1_1_buffer.html - and we do not pass any extra parameters to it. why is it not possible to tell from the code if overlapping regions are used, is this not possible in the khronos c++ api?

> 
> Driver version you are using should support NEOReadDebugKeys - strace won;t show getenv calls, you need to use ltrace:
> ltrace ./your_app

oh, of course, getenv is no kernel syscall. indeed with ltrace i see getenv calls. sadly my testcase is multithreaded, and ltrace refuses to trace the other threads for some reason/

> Can you try collecting log with:
> NEOReadDebugKeys=1 PrintDebugSettings=1 LogApiCalls=1 ./your_app ? It will help root cause the problem.

yes. i actually had to drop the python frontend and implemented the testcase in c++ https://gist.github.com/stef/e7818dc85b48c06d98a333e01f3d526f

debug settings were indeed dumped, i put them up in a gist here: https://gist.github.com/stef/d6a6f14a945387c5fedddec7ea23564b

> Log - igdrcl.log file - should be produced in current working directory from where you execute application. Please make sure you have correct rights.

i have the correct rights, yet no logfile at all is generated. i have this slight suspicion that something is missing, is there anything i need to install for this clintercept to work, or should it be supported out of the box by the debian intel opencl icd package?

> You should also see modified, non-default debug keys in the output.

indeed. have a look a the the gist i linked above.

HoppeMateusz, 2021-03-25:
Hi,

It is not possible to deduce overlapping from the code as pointers are passed around and i do not know where input_items/ output_items passed to enqueueReadBuffer/WriteBuffer comes from.

Unfortunately, internal logging with LogApiCalls is only available with Debug build of compute runtime, so to collect API trace, clIntercept must be used. It requires special setup, please take a look at :
https://github.com/intel/opencl-intercept-layer

ghostop14, 2021-03-25:
Hi folks, everything from above looks fine.  There are no overlapping or host pointers used in the code anywhere.  And thread locks are used to ensure there's no race conditions.

stef, 2021-03-25:
i patched up clMathOp_impl.c (linked above), and added this code after each `new cl::Buffer` operation:
```c++
std::cout << "clMathOp aBuffer ptr: " << aBuffer <<
             " host ptr: " << aBuffer->getInfo<CL_MEM_HOST_PTR>() <<
             " size: " << aBuffer->getInfo<CL_MEM_SIZE>() <<
             " type: " << aBuffer->getInfo<CL_MEM_TYPE>() <<
             " flags: " << aBuffer->getInfo<CL_MEM_FLAGS>() <<
             std::endl;
```

when running my test case now (in which i also enabled debug for the two clMathOp instances), the output is the following:

```
OpenCL: using Intel(R) OpenCL HD Graphics
clMathOp aBuffer ptr: 0x5570fe3bc160 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp bBuffer ptr: 0x5570feb8b370 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp cBuffer ptr: 0x5570fef54bb0 host ptr: 0 size: 65536 type: 4336 flags: 1
OpenCL INFO: Math Op - too many items for constant memory.  Building kernel with __global params...
OpenCL: using Intel(R) OpenCL HD Graphics
clMathOp aBuffer ptr: 0x5570febc9870 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp bBuffer ptr: 0x5570fe860c30 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp cBuffer ptr: 0x5570fe840680 host ptr: 0 size: 65536 type: 4336 flags: 1
OpenCL INFO: Math Op - too many items for constant memory.  Building kernel with __global params...
```

i guess the host ptr being 0 is not very useful and that would be the info that would be most important. is there any other way to get the address of the buffer than what i'm noobing around here?

ghostop14, 2021-03-25:
A host ptr is a different kind of pointer (and not what's used here), so I'm not surprised it's returning zero.  Everything else around constant memory checks looks correct to me.

ghostop14, 2021-03-25:
If you run clinfo from the command-line (you may need to apt install it), and post the output here, it may contain some clues.

stef, 2021-03-25:
> A host ptr is a different kind of pointer (and not what's used here), so I'm not surprised it's returning zero. Everything else around constant memory checks looks correct to me.

sure. that's clear, the question is, is there any way to figure out if the buffers are somehow overlapping?

stef, 2021-03-25:
> If you run clinfo from the command-line (you may need to apt install it), and post the output here, it may contain some clues.

there you go sir:  https://gist.github.com/stef/ecb32caaceba2c90891c64f065dee554

HoppeMateusz, 2021-03-29:
Hello,
adding logs around createBuffer in single file  is not enough - it is important to capture all pointers used for buffers and memory transfers. Is clMathOp_impl.c the only file using OpenCL ? If not, clIntercept with every API call logged ( with inputs and outputs ) could help find overlapping regions.

please also try debug keys:

NEOReadDebugKeys=1 EnableHostPtrTracking=0  PrintDebugSettings=1   ./app

and make sure non default keys are printed to confirm that debug keys were correctly applied.

stef, 2021-04-02:
i built clintercept, and ran my app like this:
```
~/qwer/opencl-intercept-layer/build$ NEOReadDebugKeys=1 EnableHostPtrTracking=0 PrintDebugSettings=1 LogApiCalls=1 ./cliloader/cliloader ../../racecond/build/racecond
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
CLIntercept (64-bit) is loading...
CLIntercept file location: ~/qwer/opencl-intercept-layer/build/cliloader/../intercept/libOpenCL.so
CLIntercept URL: https://github.com/intel/opencl-intercept-layer
CLIntercept git description: v3.0.0-23-g25557bb
CLIntercept git refspec: refs/heads/master
CLIntercept git hash: 25557bbc018308923975efeeb481e51521ea61df
CLIntercept optional features:
    cliloader(supported)
    cliprof(supported)
    kernel overrides(supported)
    ITT tracing(NOT supported)
    MDAPI(supported)
    clock(steady_clock)
CLIntercept environment variable prefix: CLI_
CLIntercept config file: clintercept.conf
Trying to load dispatch from: ./real_libOpenCL.so
Couldn't load library: ./real_libOpenCL.so
Trying to load dispatch from: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
... success!
Control ReportToStderr is set to non-default value: true
Timer Started!
... loading complete.
Non-default value of debug variable: PrintDebugSettings = 1
Non-default value of debug variable: LogApiCalls = 1
Non-default value of debug variable: EnableHostPtrTracking = 0
OpenCL: using Intel(R) OpenCL HD Graphics
clMathOp aBuffer ptr: 0x5611922d3920 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp bBuffer ptr: 0x5611928ebaa0 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp cBuffer ptr: 0x561192ece8c0 host ptr: 0 size: 65536 type: 4336 flags: 1
OpenCL INFO: Math Op - too many items for constant memory.  Building kernel with __global params...
OpenCL: using Intel(R) OpenCL HD Graphics
clMathOp aBuffer ptr: 0x561192db7fe0 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp bBuffer ptr: 0x56119276e3d0 host ptr: 0 size: 65536 type: 4336 flags: 4
clMathOp cBuffer ptr: 0x56119276b460 host ptr: 0 size: 65536 type: 4336 flags: 1
OpenCL INFO: Math Op - too many items for constant memory.  Building kernel with __global params...
Abort was called at 122 line in file:
/build/intel-compute-runtime-7vSeZ9/intel-compute-runtime-20.44.18297/opencl/source/os_interface/linux/drm_command_stream.inl
```

however `find ../.. -name igdrcl.log` still does not find anything. no API log file.

i checked, my app is linking libOpenCL directly, ldd confirms it:
```
libOpenCL.so.1 => /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
```
thus i beileve i do not have to install clintercept globally, right? that would only be necessary if libOpenCL would be loaded dynamically with dlopen(), which it seems it is not.

HoppeMateusz, 2021-04-02:
igdrcl.log would only be available with Debug NEO build - in Release logs are not dumped unfortunately.

This is why clIntercept logs are needed.

From above log abort happens in a different place - most probably Exec. Do you see any errors / GPU hang in dmesg ? Exec usually fails when there is hang or crash on GPU.

stef, 2021-04-02:
> igdrcl.log would only be available with Debug NEO build - in Release logs are not dumped unfortunately.

ah i see, you were asking for this earlier, i thought it can be enabled with the LogAPICalls=1. good to know. 

> This is why clIntercept logs are needed.

i did run it through clintercept. but i don't know were to look for the logs.

> From above log abort happens in a different place - most probably Exec. Do you see any errors / GPU hang in dmesg ? Exec usually fails when there is hang or crash on GPU.

no messages whatsoever in dmesg.

pwilma, 2021-04-07:
There are clIntercept logs printed to output what suggests that config file for clIntercept is missing. Please create ~/clintercept.conf with at least these variables:
LogToFile = 1
CallLogging = 1

Log file should be stored in ~/CLIntercept_Dump directory.

stef, 2021-04-07:
thanks! indeed that did the trick. i have and present you the logs from my test case: https://gist.github.com/stef/ed741051757b7a0d0fddab22e96987da

pwilma, 2021-04-30:
I analyzed this log but I couldn't find anything suspicious in it. All error codes are CL_SUCCESS and all API functions completed successfully. Did you experience any errors, crashes or hangs during this execution or it was successful run?

Can you provide compilation/execution instructions so we could try to reproduce this problem on our side?

stef, 2021-05-01:
> I analyzed this log but I couldn't find anything suspicious in it. All error codes are CL_SUCCESS and all API functions completed successfully. Did you experience any errors, crashes or hangs during this execution or it was successful run?

depends on what you consider successful, it was certainly successful in reproducing the problem, it fails exactly like i mentioned in https://github.com/intel/compute-runtime/issues/409#issuecomment-812273964, - the last few lines i copy here for your convenience:

```
clMathOp cBuffer ptr: 0x56119276b460 host ptr: 0 size: 65536 type: 4336 flags: 1
OpenCL INFO: Math Op - too many items for constant memory.  Building kernel with __global params...
Abort was called at 122 line in file:
/build/intel-compute-runtime-7vSeZ9/intel-compute-runtime-20.44.18297/opencl/source/os_interface/linux/drm_command_stream.inl
```

> Can you provide compilation/execution instructions so we could try to reproduce this problem on our side?

with pleasure, i used both debian and devuan unstable to test this, to install the bare dependencies i hope this is enough:

```
apt install gnuradio-dev ocl-icd-opencl-dev opencl-clhpp-headers opencl-headers intel-opencl-icd
```

make sure that your gnuradio is still 3.8.x, there is already a 3.9 version available since a few months but it did not hit the debian repos yet...

then you need to build `gr-clenabled`:

```
git clone https://github.com/stef/gr-clenabled.git
cd gr-clenabled
git checkout maint-3.8
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr -Wno-dev -DCMAKE_BUILD_TYPE=Debug ..
make
sudo make install
cd ../..
```

then you can clone my testcase:

```
git clone https://gist.github.com/e7818dc85b48c06d98a333e01f3d526f.git testcase
cd testcase
mkdir build
cd build
cmake ..
make
# execute testcase
./racecond
```

stef, 2021-05-01:
let me know if the above does not work for you because i was sloppy in my description how to reproduce, i'll try to give the missing details asap.

KonstantyMisiak, 2021-05-21:
I've built your testcase and I've got reproduction.
After debugging I can see that the reason for the crash is using the same host ptr in non-blocking clEnqueueWriteBuffer calls in different threads and with different command queues.
According to the spec another write with the same host ptr can't be enqueued before previous one finishes.
Please see: https://www.khronos.org/registry/OpenCL/specs/3.0-unified/html/OpenCL_API.html#_reading_writing_and_copying_buffer_objects "The memory pointed to by ptr cannot be reused by the application after the call returns".
So this is violation of the OpenCL spec and it's expected to be undefined behavior.

stef, 2021-05-21:
you are super awesome Konstanty! thank you very much.

closed 2021-05-21 by stef