intel/compute-runtime #411: Error dispatching a single thread SPIRV Kernel?

jjfumero, 2021-03-27:
**Issue migrated from Level Zero:** https://github.com/oneapi-src/level-zero/issues/48 

Executing a single thread kernel (without specifying the thread-id intrinsic in the kernel) gives me wrong results. 

I simplified the kernel as follows:

```opencl
__kernel void initValues(__global long* input, __global long* output) {
       output[0] = input[0];
} 
```

After running the SPIRV kernel I get wrong values in the output. However, if I run the following kernel:

```opencl
__kernel void initValues(__global long* input, __global long* output) {
       uint idx = get_global_id(0);
       output[idx] = input[idx];
} 
```

I get the correct results when running with a single thread.

This is the kernel dispatch

```c++
 uint32_t groupSizeX = 1u;
 uint32_t groupSizeY = 1u;
 uint32_t groupSizeZ = 1u;
 VALIDATECALL(zeKernelSuggestGroupSize(kernel, items, 1U, 1U, &groupSizeX, &groupSizeY, &groupSizeZ));
 VALIDATECALL(zeKernelSetGroupSize(kernel, groupSizeX, groupSizeY, groupSizeZ));

 // Push arguments
 VALIDATECALL(zeKernelSetArgumentValue(kernel, 0, sizeof(inputBuffer), &inputBuffer));
 VALIDATECALL(zeKernelSetArgumentValue(kernel, 1, sizeof(outputBuffer), &outputBuffer));

// Kernel thread-dispatch
ze_group_count_t dispatch;
dispatch.groupCountX = 1u;
dispatch.groupCountY = 1u;
dispatch.groupCountZ = 1u;
VALIDATECALL(zeCommandListAppendLaunchKernel(cmdList, kernel, &dispatch, nullptr, 0, nullptr))
```

Note, I am working on a tool that generates SPIRV, sometimes from single treads kernels (so we do not expose thread-ids). I wonder if I am missing something.  I am not sure if this is an error in Level Zero, but it works in pure OpenCL. 

Full program available here:
https://gist.github.com/jjfumero/a15f24c567953af57c3a02a963577980

Any ideas/pointers are appreciated.

jandres742, 2021-03-27:
@jjfumero Thanks

So these two kernels are not the same:
```cpp
__kernel void initValues(__global long* input, __global long* output) {
       output[0] = input[0];
} 
```
```cpp
__kernel void initValues(__global long* input, __global long* output) {
       uint idx = get_global_id(0);
       output[idx] = input[idx];
}
```

In the first kernel, you are just copying element 0 of `input` into element 0 of `output`. In the second one, you copy all elements in the range [0, `groupSizeX`], since that's the total global size from these lines. `globalSizeX` is most likely equal to `items`
```cpp
VALIDATECALL(zeKernelSuggestGroupSize(kernel, items, 1U, 1U, &groupSizeX, &groupSizeY, &groupSizeZ));
VALIDATECALL(zeKernelSetGroupSize(kernel, groupSizeX, groupSizeY, groupSizeZ));
```

wouldn't be that correct?

jjfumero, 2021-03-27:
Yes, I know that. My point is that the first kernel is launched with a single thread in the kernel dispatch, because I only want the first array-element  (I am doing something more complex actually but this simple case illustrates the issue).  However, the result is not correct. The first version is not correct, it returns `0`. No data is copied. 

By making it "parallel" and only dispatching a single thread-kernel that makes use of the OpenCL global-id, I can make the second version equivalent to the first one and the results are correct. My question is, how can we run single thread kernels (without specifying thread-ids in OpenCL/SPIRV) in Level Zero? Does it make sense? Let me know if you need more details.

jjfumero, 2021-04-08:
Hi @jandres742 , @AdamCetnerowski . I wonder if there is any update on this issue.

Thank you, 
Juan

jandres742, 2021-04-08:
@jjfumero : I'm not able to reproduce.

I also added a printf to see the actual groupSizes being set, and they are correct: Only one. I dont know if it is maybe you are compiling the SPV differently. Could you share your SPV file? See results below with latest release, https://github.com/intel/compute-runtime/releases/tag/21.13.19438

```sh
$ ./test-411.run
Device   : Intel(R) UHD Graphics 630 [0x3e9b]
Type     : GPU
Vendor ID: 8086
#Queue Groups: 1
groupSizeX 1
groupSizeY 1
groupSizeZ 1
Output: 120012

Zello World Results validation PASSED
$ cat ./test-411.kernel.c
__kernel void initValues(__global long* input, __global long* output) {
    output[0] = input[0];
}
$
```

jjfumero, 2021-04-09:
That's interesting. 

From this OpenCL kernel:

```opencl
__kernel void initValues(__global long* input, __global long* output) {
    output[0] = input[0];
}
```

I am compiling to SPIR-V using the following flags:

```bash
$ clang -cc1 -triple spir opencl-copy.cl -O0 -finclude-default-header -emit-llvm-bc -o opencl-copy.bc
$ llvm-spirv opencl-copy.bc -o opencl-copy.spv
```
Clang from Intel/LLVM

```bash 
$ clang --version
clang version 12.0.0 (/home/juan/manchester/SPIRV/intel-llvm/llvm/clang a05f5de1d558ba7e592c481d2ac69b7dd11de28f)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /home/juan/manchester/SPIRV/intel-llvm/llvm/build/bin
```

And this is the disassembled SPIR-V code I get:

```asm
; SPIR-V
; Version: 1.0
; Generator: Khronos LLVM/SPIR-V Translator; 14
; Bound: 21
; Schema: 0
               OpCapability Addresses
               OpCapability Kernel
               OpCapability Int64
          %1 = OpExtInstImport "OpenCL.std"
               OpMemoryModel Physical32 OpenCL
               OpEntryPoint Kernel %6 "initValues"
         %20 = OpString "kernel_arg_type.initValues.long*,long*,"
               OpSource OpenCL_C 100000
               OpName %input "input"
               OpName %output "output"
               OpName %entry "entry"
               OpName %input_addr "input.addr"
               OpName %output_addr "output.addr"
               OpName %ptridx "ptridx"
               OpName %ptridx1 "ptridx1"
               OpDecorate %input_addr Alignment 4
               OpDecorate %output_addr Alignment 4
      %ulong = OpTypeInt 64 0
       %uint = OpTypeInt 32 0
     %uint_0 = OpConstant %uint 0
       %void = OpTypeVoid
%_ptr_CrossWorkgroup_ulong = OpTypePointer CrossWorkgroup %ulong
          %5 = OpTypeFunction %void %_ptr_CrossWorkgroup_ulong %_ptr_CrossWorkgroup_ulong
%_ptr_Function__ptr_CrossWorkgroup_ulong = OpTypePointer Function %_ptr_CrossWorkgroup_ulong
          %6 = OpFunction %void DontInline %5
      %input = OpFunctionParameter %_ptr_CrossWorkgroup_ulong
     %output = OpFunctionParameter %_ptr_CrossWorkgroup_ulong
      %entry = OpLabel
 %input_addr = OpVariable %_ptr_Function__ptr_CrossWorkgroup_ulong Function
%output_addr = OpVariable %_ptr_Function__ptr_CrossWorkgroup_ulong Function
               OpStore %input_addr %input Aligned 4
               OpStore %output_addr %output Aligned 4
         %13 = OpLoad %_ptr_CrossWorkgroup_ulong %input_addr Aligned 4
     %ptridx = OpInBoundsPtrAccessChain %_ptr_CrossWorkgroup_ulong %13 %uint_0
         %17 = OpLoad %ulong %ptridx Aligned 8
         %18 = OpLoad %_ptr_CrossWorkgroup_ulong %output_addr Aligned 4
    %ptridx1 = OpInBoundsPtrAccessChain %_ptr_CrossWorkgroup_ulong %18 %uint_0
               OpStore %ptridx1 %17 Aligned 8
               OpReturn
               OpFunctionEnd
```

When I run the application:

```bash
./sampleCopy 
Device   : Intel(R) HD Graphics 630 [0x591b]
Type     : GPU
Vendor ID: 8086
#Queue Groups: 1
Output: 0

Zello World Results validation FAILED
```

I also attached the SPIR-V binary file: [opencl-copy.spv.zip](https://github.com/intel/compute-runtime/files/6283718/opencl-copy.spv.zip)

jjfumero, 2021-04-09:
I just updated the Intel OpenCL Drivers to `21.13.19438` and it works! Not sure what the problem was. 

I was using the prev. version: `21.12.19358`

jandres742, 2021-04-10:
Ok, so something was fixed. Please let us know if the problem comes back, in case we inadvertently introduced the bug again.

closed 2021-05-06 by AdamCetnerowski

raikonenfnu, 2022-09-09:
@jandres742 Seems like the problem came back in https://github.com/intel/compute-runtime/issues/561 😅