intel/compute-runtime #415: Rocket Lake support?

mko-io, 2021-04-05:
The recent release RKL contains 2 subslice of Xe GPU and it's every similar to TGL, please consider support it

PiotrRozenfeld, 2021-04-07:
RKL is already supported by compute-runtime - see https://github.com/intel/compute-runtime/blob/master/shared/source/dll/devices/devices_base.inl#L28 
If you purchased a Rocket Lake CPU (e.g. Core i5-11600K or Core i9-11900K), it should be correctly recognized and fully functional on Linux. Windows is out of scope of this github issue.
Rocket Lake will be explicitly added to “Quality expectations” table in the release notes shortly.

mko-io, 2021-04-07:
That's great! I'm not a windows user, so the linux support is enough. Thanks

closed 2021-04-07 by mko-io