intel/compute-runtime #416: Installation methods

jeyblu, 2021-04-06:
The [documentation](https://github.com/intel/compute-runtime) says there are different options for installation. You can use package manager or manually download. It is not clear which one should be used for a developer that needs to develop code using the opencl sdk. Is installing it using the package manager enough? To build Intel oneDNN for [gpu]( https://oneapi-src.github.io/oneDNN/dev_guide_build_options.html), it says it needs the opencl sdk. Is installing the compute-runtime using the package manager sufficient or do you need to use the manual installation method? Can you add more info in the documentation about what gets installed by the package manager and what gets installed if you manually install it? I've installed it on Ubuntu 20.04 using the package manager with "apt-get install intel-opencl-icd". It installed but I couldn't find the path where the sdk is installed. The oneDNN project requires the path to the sdk. Thanks.

AdamCetnerowski, 2021-04-08:
Please take a look at these [instructions](https://dgpu-docs.intel.com/installation-guides/index.html) and see if they address your concerns / questions. It covers recommended installation through package manager.

jeyblu, 2021-04-08:
Thanks, it addresses most of my questions. The "Optional: Install developer packages" section list the packages that need to be installed. Does any of the packages in that section include the Intel opencl sdk which is required for developing gpgpu software for Intel gpu? An example of such gpgpu software is Intel oneDNN.

closed 2022-10-13 by AdamCetnerowski