intel/compute-runtime #417: Information table of supported opencl features like in or better than mesamatrix

stefano2734, 2021-04-12:
See opencl tables in 
https://mesamatrix.net/

AdamCetnerowski, 2021-05-13:
We are looking at ways to do this efficiently, but I don't expect to have a solution soon.

stefano2734, 2021-05-26:
OpenCL 3.0 have many optional features of 1.x and 2.x. Important features are optional.
There are 3 main players with Intel, Nvidia and AMD x86-64.
And 2 dominante platforms with Unix-Linux and Windows.
There are 2 software drivers in Linux with beignet for older software and newer Neo compute  runtime and a Windows Driver.
AMD is with two different concepts by Windows Driver and ROCm.
Nvidia is at same or nearly same  driver State with Linux and Windows, but some important features are not supported.
Mesa 3D and POCL is doing it on linux independent.

So developers see a colorful but mostly grey mosaic of different implementations, possibilities and limitations by software and hardware. CUDA  is here like a solid crystal by hardware and software.

What is here a solution for users on pc Plattform in my small trivial view?
A new standard 1.3: the implemented minimum state by the big three for older hardware like ivy bridge, amd gcn 1. Gen, Nvidia Kepler
A new standard 3.1: 3.0 and the implemented minimum software state by the big three by actual supported hardware 
A new standard 3.2: 3.0 and the implemented state by Intel and AMD with all important  optional 1.x features and some 2.x 
A new standard 3.3: 3.0 with all important optional features  and some to do for Intel, AMD and perhaps Nvidia 
So the developer and user of OpenCL know what level is needed to make software and to buy hardware.

Intel here is in good shape for more consistency and others can do or must do more for same level.
So my issue is here my little idea on way for more consistency, coordination and performance of OpenCL against closed source CUDA.
So I admire your work here and the work of Intels in the last decades and hope for more.

AdamCetnerowski, 2022-11-24:
At this time we do not intend to provide this.

closed 2022-11-24 by AdamCetnerowski