intel/compute-runtime #418: clinfo lists no devices

shlomif, 2021-04-19:
On an Intel NUC with Fedora 33 x86-64, clinfo lists no devices

```
+ inxi -CSG
System:
  Host: fedora Kernel: 5.11.14-200.fc33.x86_64 x86_64 bits: 64 
  Desktop: Xfce 4.14.4 Distro: Fedora release 33 (Thirty Three) 
CPU:
  Info: Quad Core model: Intel Core i5-8259U bits: 64 type: MT MCP cache: 
  L2: 6 MiB 
  Speed: 700 MHz min/max: 400/3800 MHz Core speeds (MHz): 1: 700 2: 700 
  3: 700 4: 700 5: 700 6: 700 7: 741 8: 718 
Graphics:
  Device-1: Intel CoffeeLake-U GT3e [Iris Plus Graphics 655] driver: N/A 
  Display: server: Fedora Project X.org 1.20.11 driver: loaded: vesa 
  unloaded: fbdev,modesetting resolution: 1920x1080~77Hz 
  OpenGL: renderer: llvmpipe (LLVM 11.0.0 256 bits) v: 4.5 Mesa 20.3.5 
+ clinfo
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 20.3.5
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 0

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Clover
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   ÒÕh_
  clCreateContext(NULL, ...) [default]            No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No devices found in platform

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.13
  ICD loader Profile                              OpenCL 3.0
	NOTE:	your OpenCL library declares to support OpenCL 3.0,
		but it seems to support up to OpenCL 2.2 only.
```

JacekDanecki, 2021-04-19:
Could you provide output from commands:
```
lspci -vnn
lsmod
dmesg | grep -E "i915|drm"
cat /proc/cmdline
ls -lR /dev/dri
groups
```

shlomif, 2021-04-19:
> Could you provide output from commands:
> 
> ```
> lspci -vnn
> lsmod
> dmesg | grep -E "i915|drm"
> cat /proc/cmdline
> ls -lR /dev/dri
> groups
> ```

Updated the main comment section with their outputs.

JacekDanecki, 2021-04-19:
You need to remove "nomodeset" parametr  from kernel command line.

shlomif, 2021-04-19:
> You need to remove "nomodeset" parametr from kernel command line.

Thanks! I'll try that soon.

shlomif, 2021-04-19:
> > You need to remove "nomodeset" parametr from kernel command line.
> 
> Thanks! I'll try that soon.

@JacekDanecki : tried it now - still 0 devices on both plasma5 and jwm:

On an Intel NUC with Fedora 33 x86-64, clinfo lists no devices

```
+ inxi -CSG
System:    Host: fedora Kernel: 5.11.14-200.fc33.x86_64 x86_64 bits: 64 
           Desktop: JWM 2.3.7 Distro: Fedora release 33 (Thirty Three) 
CPU:       Info: Quad Core model: Intel Core i5-8259U bits: 64 type: MT MCP cache: 
           L2: 6 MiB 
           Speed: 1600 MHz min/max: 400/3800 MHz Core speeds (MHz): 1: 1600 2: 1591 
           3: 1601 4: 1600 5: 1887 6: 2068 7: 1570 8: 1600 
Graphics:  Device-1: Intel CoffeeLake-U GT3e [Iris Plus Graphics 655] driver: i915 
           v: kernel 
           Display: server: Fedora Project X.org 1.20.11 driver: loaded: modesetting 
           unloaded: fbdev,vesa resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa Intel Iris Plus Graphics 655 (CFL GT3) 
           v: 4.6 Mesa 20.3.5 
+ clinfo
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 20.3.5
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 0

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Clover
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   DG/^
  clCreateContext(NULL, ...) [default]            No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No devices found in platform

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.13
  ICD loader Profile                              OpenCL 3.0
	NOTE:	your OpenCL library declares to support OpenCL 3.0,
		but it seems to support up to OpenCL 2.2 only.
+ lspci -vnn
00:00.0 Host bridge [0600]: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers [8086:3ed0] (rev 08)
	DeviceName:  Intel Dual Band
	Subsystem: Intel Corporation Device [8086:2074]
	Flags: bus master, fast devsel, latency 0
	Capabilities: <access denied>
	Kernel driver in use: skl_uncore

00:02.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-U GT3e [Iris Plus Graphics 655] [8086:3ea5] (rev 01) (prog-if 00 [VGA controller])
	DeviceName:  CPU
	Subsystem: Intel Corporation Device [8086:2074]
	Flags: bus master, fast devsel, latency 0, IRQ 139
	Memory at bf000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 80000000 (64-bit, prefetchable) [size=256M]
	I/O ports at 4000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: i915
[... 691 lines cut ...]
[    2.780025] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
[    2.905373] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    2.985662] fbcon: i915drmfb (fb0) is primary device
[    2.985666] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    8.533012] systemd[1]: Starting Load Kernel Module drm...
[    8.548189] systemd[1]: modprobe@drm.service: Succeeded.
[    8.548314] systemd[1]: Finished Load Kernel Module drm.
[   18.957110] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[   20.865434] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
+ cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.11.14-200.fc33.x86_64 root=UUID=42696b9e-6662-4f45-9539-2bfc6e48543f ro rootflags=subvol=root rhgb quiet
+ ls -lR /dev/dri
/dev/dri:
total 0
drwxr-xr-x. 2 root root         80 Apr 19 21:10 by-path
crw-rw----+ 1 root video  226,   0 Apr 19 21:10 card0
crw-rw-rw-. 1 root render 226, 128 Apr 19 21:10 renderD128

/dev/dri/by-path:
total 0
lrwxrwxrwx. 1 root root  8 Apr 19 21:10 pci-0000:00:02.0-card -> ../card0
lrwxrwxrwx. 1 root root 13 Apr 19 21:10 pci-0000:00:02.0-render -> ../renderD128
+ groups
shlomif wheel
```

shlomif, 2021-04-19:
@JacekDanecki : on the bright side, the speakers now emit audio even with the earphones unplugged. Thanks! ♥

shlomif, 2021-04-19:
> if `sudo clinfo` works, then you need to add your account to `video` or `render` group
> 
> > To allow NEO access to GPU device make sure user has permissions to files /dev/dri/renderD*.
> 
> https://github.com/intel/compute-runtime#readme

Thanks for the lead: `sudo dnf install opencl-intel` made it work as my regular user!

shlomif, 2021-04-19:
@AdamCetnerowski @alalek : my original use case is also working now having ironed out some bugs. Thanks again!

shlomif, 2021-04-20:
> @AdamCetnerowski @alalek : my original use case is also working now having ironed out some bugs. Thanks again!

Hi again!

Per item no. 1 in https://www.joelonsoftware.com/2007/02/19/seven-steps-to-remarkable-customer-service/ we can add some logic to clinfo on fedora/RHEL/etc. to refer future users to the solution. Does this sound good?

JacekDanecki, 2021-05-10:
Closing as an issue was resolved.
For clinfo questions use this project: https://github.com/Oblomov/clinfo

closed 2021-05-10 by JacekDanecki