intel/compute-runtime #420: Is there Level Zero support on Rocket Lake ?

homksei, 2021-05-05:
I have machine with Rocket Lake-S (11th Gen Intel(R) Core(TM) i7-11700K)
I use Ubuntu 20.04.2 with  5.9.0-050900-generic kernel.

Here is a list of packages that I am using:
```
intel-gmmlib_20.4.1_amd64.deb
intel-igc-core_1.0.7041_amd64.deb
intel-igc-media_1.0.7041_amd64.deb
intel-igc-opencl-devel_1.0.7041_amd64.deb
intel-igc-opencl_1.0.7041_amd64.deb
intel-level-zero-gpu_1.1.19610_amd64.deb
intel-ocloc_21.16.19610_amd64.deb
intel-opencl_21.16.19610_amd64.deb
level-zero-devel_1.2.3+u18.04_amd64.deb
level-zero_1.2.3+u18.04_amd64.deb
```

clinfo recognizes GPU: 

```
...
  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Graphics [0x4c8a]
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 3.0 NEO
  Driver Version                                  21.16.19610
  Device OpenCL C Version                         OpenCL C 1.2
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
...
```
but when I try to run something with Level Zero I get:
```
'No device of requested type available. Please check https://software.intel.com/content/www/us/en/develop/articles/intel-oneapi-dpcpp-system-requirements.html -1 (CL_DEVICE_NOT_FOUND)'
```
At the same time everything works with OpenCL.

Also, to check devices, I use the sycl-ls executable from oneAPI 2021.2 (Level Zero is not recognized):
```
ACC : OpenCL 1.2 [ 2020.11.11.0.13_160000 ]
CPU : OpenCL 2.1 (Build 0)[ 2020.11.11.0.13_160000 ]
GPU : OpenCL 3.0 NEO [ 21.16.19610 ]
HOST: 1.2[ 1.2 ]
```

AdamCetnerowski, 2021-05-05:
We expect the support to land this week, which would put it in the 2021.19.xxxxx releaase

closed 2021-05-06 by homksei

generalova-kate, 2021-08-11:
@AdamCetnerowski I don't see any public release(https://github.com/intel/compute-runtime/releases) with support Rocket lake(https://ark.intel.com/content/www/us/en/ark/products/212279/intel-core-i7-11700-processor-16m-cache-up-to-4-90-ghz.html), could you guide me?

AdamCetnerowski, 2021-08-11:
This is an omission in our documentation. RKL support is compiled into the weekly releases and shows good health. The next release will list it.

Thanks for pointing this out!

generalova-kate, 2021-08-11:
@AdamCetnerowski could you share the minimal opencl version with support rocket lake?

AdamCetnerowski, 2021-08-11:
All our platforms are at OpenCL 3.0

generalova-kate, 2021-08-11:
@AdamCetnerowski I am about opencl driver version.