intel/compute-runtime #421: clGetPlatformIDs segfaults

midn1, 2021-05-07:
I had installed the runtime using the `intel-compute-runtime` and `ocl-icd` packages. This minimal example segfaults in the call to clGetPlatformIDs:

	#include<CL/cl.h>

	int main() {
		cl_uint numPlatforms = 0;
		clGetPlatformIDs(0, NULL, &numPlatforms);
	}

This seems to be very specific to my system setup, so I don't know what other information to give. Please ask, and thanks for your patience.

alalek, 2021-05-07:
You can start investigation from `LD_DEBUG=libs ./openclexample` command output. Ensure that there are no libraries from "unexpected" places.

Also check this comment ( https://github.com/intel/compute-runtime/issues/418#issuecomment-822332648 ) and similar from other issues.

midn1, 2021-05-08:
Indeed. Somehow `/opt/intel/opencl/libOpenCL.so` is taking precedence over `/usr/lib/libOpenCL.so`. A hack like `LD_LIBRARY_PATH=/usr/lib ./openclexample` seems to fix the issue.

Thanks.

closed 2021-05-08 by midn1