intel/compute-runtime #422: opencl env based on ubuntu 16.04 issue

jiwei5518, 2021-05-10:
Hello 

         I have successfully installed the neo package as follows from your git webiste(version:18.44.11760 which said proved at ubuntu16.04): 

           intel-gmmlib_18.4.0.326_amd64.deb 
           intel-igc-opencl_18.43.999_amd64.deb
           intel-igc-core_18.43.999_amd64.deb 
           intel-opencl_18.44.11760_amd64.deb 
           ocl-icd-libopencl1_2.2.8-1_amd64.deb (apt-get installed)

           My system is **Ubuntu16.04**, and we cannot upgred to ubuntu18 or more.

           I have added to video group. while I use clinfo, it still got **Number of platforms  0**

           we surly got a intel graphic card HD Graphics 630

          here is the output log of  clinfo 
[output.log](https://github.com/intel/compute-runtime/files/6449912/output.log)

thanks

JacekDanecki, 2021-05-10:
In the log I can see:
```
open("/dev/dri/renderD128", O_RDWR)     = -1 ENOENT (No such file or directory)
```
what means i915 kernel driver was not initialized correctly.
See issue: https://github.com/intel/compute-runtime/issues/418 as it looks similar.

jiwei5518, 2021-05-11:
@JacekDanecki 

       thanks for the feedback.

       Actually, we want to operate it on a system which very similar to ubuntu16 called elux. here is the system info 
       "Linux SZ-T630-RP6-eLux8252-8131-2871-2 5.4.72 #1 SMP Mon Oct 26 10:25:23 CET 2020 x86_64 GNU/Linux"
       we move all the libs  to the system and try again. and I have checked that it seems i915 driver is initialized correctly, it has "/dev/dri/renderDxx", while still got **Number of platforms  0**
        
BTW, I have compared the normal clinfo log(operate on a ubuntu18) and the failed one, it seems that the failed one still not have the step of "open /dev/dri/renderDxx"  

here is the output of failed strace clinfo on elux system
     
[strace.log](https://github.com/intel/compute-runtime/files/6455956/strace.log)

JacekDanecki, 2021-05-13:
Closing as an issue was resolved by installing correct libraries, confirmed by email conversations.

closed 2021-05-13 by JacekDanecki