intel/compute-runtime #424: Ubuntu 18.04 with intel-opencl-icd 21.19.19792-1~ppa1~bionic1: SIGFPE

koitsu, 2021-05-24:
Since updating `intel-opencl-icd` to `21.19.19792-1~ppa1~bionic1`, I've begun to see 100% reproducible SIGFPE on all IGD-related programs (clinfo, dnetc, etc.).  This is on an Intel NUC5i5RYK.  Focusing on `clinfo` just to make my life easier.

```
(00:42:59 jdc@linux2) ~/dnetc521-linux-amd64-opencl $ dpkg -l intel-opencl-icd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                     Version                   Architecture              Description
+++-========================================-=========================-=========================-=====================================================================================
ii  intel-opencl-icd                         21.19.19792-1~ppa1~bionic amd64                     Intel graphics compute runtime for OpenCL
```

Some examples:

```
(00:40:38 jdc@linux2) ~/dnetc521-linux-amd64-opencl $ strace clinfo
...
munmap(0x7f7e87dde000, 31530)           = 0
getdents(3, /* 0 entries */, 32768)     = 0
unlink("igdrcl.log")                    = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/dev/dri/by-path", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 4
fstat(4, {st_mode=S_IFDIR|0755, st_size=80, ...}) = 0
getdents(4, /* 4 entries */, 32768)     = 144
getdents(4, /* 0 entries */, 32768)     = 0
close(4)                                = 0
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:00:02.0-render", O_RDWR) = 4
ioctl(4, DRM_IOCTL_VERSION, 0x7ffc1aa05110) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM or DRM_IOCTL_TEGRA_CLOSE_CHANNEL, 0x7ffc1aa051f0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM or DRM_IOCTL_TEGRA_CLOSE_CHANNEL, 0x7ffc1aa051f0) = 0
ioctl(4, _IOC(_IOC_READ|_IOC_WRITE, 0x64, 0x79, 0x10), 0x7ffc1aa05100) = -1 EINVAL (Invalid argument)
ioctl(4, DRM_IOCTL_I915_GETPARAM or DRM_IOCTL_TEGRA_CLOSE_CHANNEL, 0x7ffc1aa05180) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM or DRM_IOCTL_TEGRA_CLOSE_CHANNEL, 0x7ffc1aa05180) = 0
--- SIGFPE {si_signo=SIGFPE, si_code=FPE_INTDIV, si_addr=0x7f7e86f5416f} ---
+++ killed by SIGFPE +++
Floating point exception
```
Kernel outputs the following for this crash:
```
[  595.485667] traps: clinfo[1473] trap divide error ip:7ff83c11916f sp:7ffcad4dd950 error:0 in libigdrcl.so[7ff83bd69000+47e000]
```
Running normally (i.e. not within strace) but with `LD_DEBUG=libs` shows the crash is in Intel's own library:

```
(00:42:08 jdc@linux2) ~/dnetc521-linux-amd64-opencl $ LD_DEBUG=libs clinfo
      1521:     find library=libOpenCL.so.1 [0]; searching
      1521:      search cache=/etc/ld.so.cache
      1521:       trying file=/usr/lib/x86_64-linux-gnu/libOpenCL.so.1
      1521:
      1521:     find library=libdl.so.2 [0]; searching
      1521:      search cache=/etc/ld.so.cache
      1521:       trying file=/lib/x86_64-linux-gnu/libdl.so.2
      1521:
      1521:     find library=libc.so.6 [0]; searching
      1521:      search cache=/etc/ld.so.cache
      1521:       trying file=/lib/x86_64-linux-gnu/libc.so.6
      1521:
      1521:
      1521:     calling init: /lib/x86_64-linux-gnu/libc.so.6
      1521:
      1521:
      1521:     calling init: /lib/x86_64-linux-gnu/libdl.so.2
      1521:
      1521:
      1521:     calling init: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
      1521:
      1521:
      1521:     initialize program: clinfo
      1521:
      1521:
      1521:     transferring control: clinfo
[... 229 lines cut ...]
[    4.809498] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    4.809504] [drm] Driver supports precise vblank timestamp query.
[    4.811654] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.814291] [drm] Initialized i915 1.6.0 20171023 for 0000:00:02.0 on minor 0
[    4.846113] [drm] Cannot find any crtc or sizes
[    4.862106] [drm] Cannot find any crtc or sizes
[    7.494247] snd_hda_intel 0000:00:03.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
(00:44:56 jdc@linux2) ~ $ cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-4.15.0-143-generic root=UUID=e60ab782-7324-4fb4-bab3-81f216ceb7fe ro maybe-ubiquity ipv6.disable=1
(00:44:59 jdc@linux2) ~ $ ls -lR /dev/dri
/dev/dri:
total 0
drwxr-xr-x 2 root root        80 May 24 00:28 by-path
crw-rw---- 1 root video 226,   0 May 24 00:28 card0
crw-rw---- 1 root video 226, 128 May 24 00:28 renderD128

/dev/dri/by-path:
total 0
lrwxrwxrwx 1 root root  8 May 24 00:28 pci-0000:00:02.0-card -> ../card0
lrwxrwxrwx 1 root root 13 May 24 00:28 pci-0000:00:02.0-render -> ../renderD128
(00:45:02 jdc@linux2) ~ $ groups
jdc adm cdrom sudo dip video plugdev lxd docker
```

Please advise.

koitsu, 2021-05-24:
Kind of smells like https://github.com/intel/compute-runtime/issues/421 but isn't; manifests substantially differently.  I'm left wondering if someone broke some ABI semantics between libOpenCL.so and libigdrcl.so (suspecting the latter, since libOpenCL.so hasn't changed in 4 years):

```
(01:04:55 jdc@linux2) ~ $ gdb /usr/bin/clinfo
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/clinfo...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/clinfo
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGFPE, Arithmetic exception.
0x00007ffff716a16f in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
(gdb) f 0
#0  0x00007ffff716a16f in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
(gdb) disass
No function contains program counter for selected frame.
(gdb) bt full
#0  0x00007ffff716a16f in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
No symbol table info available.
#1  0x00007ffff6e1681d in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
No symbol table info available.
#2  0x00007ffff716e510 in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
No symbol table info available.
#3  0x00007ffff7114de1 in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
No symbol table info available.
#4  0x00007ffff7115683 in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
No symbol table info available.
#5  0x00007ffff6e0e2a7 in ?? () from /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
No symbol table info available.
#6  0x00007ffff7bcca66 in ?? () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
No symbol table info available.
#7  0x00007ffff7bcd0d3 in clGetPlatformIDs () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
No symbol table info available.
#8  0x000055555555a722 in ?? ()
No symbol table info available.
#9  0x00007ffff75f4bf7 in __libc_start_main (main=0x55555555a5d0, argc=1, argv=0x7fffffffe4d8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>,
    stack_end=0x7fffffffe4c8) at ../csu/libc-start.c:310
        self = <optimized out>
        __self = <optimized out>
        result = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {0, -4470602088582343427, 93824992258832, 140737488348368, 0, 0, -7737062915239439107, -7737044503943441155}, mask_was_saved = 0}},
          priv = {pad = {0x0, 0x0, 0x7ffff7de38d3 <_dl_init+259>, 0x7ffff7dd14b0}, data = {prev = 0x0, cleanup = 0x0, canceltype = -136431405}}}
        not_first_call = <optimized out>
#10 0x000055555555ab3a in ?? ()
No symbol table info available.
```

JacekDanecki, 2021-05-24:
Release 21.19.19792 introduced regression on Ubuntu 18.04 with default kernel, see release notes: https://github.com/intel/compute-runtime/releases/tag/21.19.19792
You can update your kernel to linux-image-generic-hwe-18.04 or recompile Neo with fix from commit: https://github.com/intel/compute-runtime/commit/21d93a33d20810dcfc3fbc70bee6aae7a4e57ecc

koitsu, 2021-05-24:
Great, glad to know I wasn't losing my mind and that this was a regression!  :)  I've gone ahead and updated the Ubuntu 18.04 kernel [per Canonical's instructions](https://wiki.ubuntu.com/Kernel/LTSEnablementStack#Ubuntu_18.04_LTS_-_Bionic_Beaver) or more precisely, this:

```
sudo apt-get install --install-recommends linux-generic-hwe-18.04
sudo apt-get autoremove
sudo reboot
```
Now all OpenCL applications are functional.  Thank you very much!

closed 2021-05-24 by koitsu