intel/compute-runtime #425: zeModuleGetGlobalPointer can not get correct size of global variable

jz10, 2021-05-24:
When we invoked zeModuleGetGlobalPointer to retrieve the size of global variable specified by the 2nd argument (which is a string for global variable's name) , zeModuleGetGlobalPointer just returned 0 in the 3rd argument, but gave correct device visible pointer in the 4th argument.

We referred to the description at: https://spec.oneapi.com/level-zero/latest/core/api.html?highlight=globalpointer#zemodulegetglobalpointer . The L0 module was create by adding the compilation flag '-cl-std=CL2.0 -cl-take-global-address'

We tried the definition of global variable as scalar type or array type, neither of their size can be retrieved via zeModuleGetGlobalPointer correctly.

jz10, 2021-05-24:
I was working on the L0 built from this March, maybe I should update to newer version?

jandres742, 2021-05-26:
Yes, let's try with latest release, to confirm.

AdamCetnerowski, 2021-06-17:
Since a fix was provided, I'm closing this. Please re-open if still reoccuring and thank you for the report.

closed 2021-06-17 by AdamCetnerowski